framps/raspiBackup #307: Customize the subject in e-mail notifications

SJMarty, 2021-01-29:
Two questions really...both related to e-mail notifications...

1. Can the subject in the notification e-mails be customized using either the raspiBackup.conf file or some other means that would retain the changes through an update?
2. Is there a list of all of the possible subjects?  I'd like to create some e-mail rules based on certain outcomes.

Thanks.

framps, 2021-01-29:
re 1:

There is no support to customize the subject message. 
Following two subjects are used only:
```
MSG_TITLE_OK=41
MSG_EN[$MSG_TITLE_OK]="%s: Backup finished successfully."
MSG_DE[$MSG_TITLE_OK]="%s: Backup erfolgreich beendet."
MSG_TITLE_ERROR=42
MSG_EN[$MSG_TITLE_ERROR]="%s: Backup failed !!!."
MSG_DE[$MSG_TITLE_ERROR]="%s: Backup nicht erfolgreich !!!."
```

You can customize them with a hack. But I cannot guarantee this code will not be changed. But I see only a small probability for changes in this area. But please note: I don't accept any issues related to the hack :wink: . Write a wrapper bash script, call `raspiBackup.sh --include` to get all variable definitions in your code and then overwrite the two messages.

re 2: See (1) But there is a small probability new subjects will be added in some future release

framps, 2021-02-05:
Any further questions? Otherwise please close the issue.

closed 2021-02-05 by SJMarty