framps/raspiBackup #311: After updating, version number reports the same?

SJMarty, 2021-02-05:
When running raspiBackup -U -S, I get the following output...

```
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
--- RBK0195I: Update raspiBackup.sh to version 0.6.6.
--- RBK0038I: Are you sure? y/N 
y
--- RBK0057I: Downloading file raspiBackup.sh from https://www.linux-tips-and-tricks.de.
--- RBK0072I: /usr/local/bin/raspiBackup.sh updated from version 0.6.6 to version 0.6.6. Previous version saved as /usr/local/bin/raspiBackup.0.6.6.sh. Don't forget to test backup and restore with the new version now.
```
The updated "from" and "to" version are both being reported as v0.6.6.  Should the version number change?  If not, how can I validate that I've updated successfully?

framps, 2021-02-05:
The version number will change whenever a new release is published which includes new features, enhancements and bug fixes. Small bug fixes are added to the current release.

> If not, how can I validate that I've updated successfully?
Option `-S` is the force update option which downloads the current code and replaces the existing code even the version number did not change. But the git commit number has changed. 

To verify you got the updated version just call and check
```
raspiBackup.sh --version 
Version: 0.6.6 CommitSHA: 6dc5832 CommitDate: 2021-02-05 CommitTime: 14:43:04
```

I have to admit the new commit number is missing and should be reported when `-S` is used. That's planned to be added in next release.

framps, 2021-02-05:
Thank you very much for your donation :+1:

closed 2021-05-05 by framps