framps/raspiBackup #312: tar empty

xMadNet, 2021-02-07:
sudo raspiBackup.sh -a : -o : -m detailed >> **backup done in a few seconds !!!! with no errors**
Please Help. I suppose I have nno valid backup

My config
* raspiBackup.sh V0.6.6 (a9c8b83)
* Rpi4 4 GB : model name      : ARMv7 Processor rev 3 (v7l)
* Linux XPi4-32 5.10.10-v7l+ #1397 SMP Mon Jan 25 18:42:27 GMT 2021 armv7l GNU/Linux
* NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0   1.8T  0 disk
└─sda1   8:1    0   1.8T  0 part /Nas
sdb      8:16   0 119.2G  0 disk
├─sdb1   8:17   0   256M  0 part /boot
└─sdb2   8:18   0   119G  0 part /

Resultat
Sun Feb 07 09:28:54 pi@XPi4-32: /backup/XPi4-32/XPi4-32-tgz-backup-20210207-090558 $ ll
total 262232
-rwxrwx--- 1 pi pi     70118 Feb  7 09:06 raspiBackup.log
-rwxrwx--- 1 pi pi       852 Feb  7 09:06 raspiBackup.msg
-rwxrwx--- 1 pi pi 268435456 Feb  7 09:06 XPi4-32-backup.img
-rwxrwx--- 1 pi pi       512 Feb  7 09:06 XPi4-32-backup.mbr
-rwxrwx--- 1 pi pi       181 Feb  7 09:06 XPi4-32-backup.sfdisk
-rwxrwx--- 1 pi pi        45 Feb  7 09:06 XPi4-32-tgz-backup-20210207-090558.tgz

Logs
20210207-090558 DBG 1718:      --> initializeDefaultConfigVariables 
20210207-090558 DBG 1852:      <-- initializeDefaultConfigVariables 
20210207-090558 DBG 1857:      --> copyDefaultConfigVariables 
20210207-090558 DBG 1957:          --> substituteNumberArguments 
20210207-090558 DBG 1984:          <-- substituteNumberArguments 
20210207-090558 DBG 1928:      <-- copyDefaultConfigVariables 
20210207-090559 DBG 2699:      --> readConfigParameters 
20210207-090559 DBG 2755:      <-- readConfigParameters 
20210207-090559 DBG 1857:      --> copyDefaultConfigVariables 
20210207-090559 DBG 1957:          --> substituteNumberArguments 
20210207-090559 DBG 1984:          <-- substituteNumberArguments 
20210207-090559 DBG 1928:      <-- copyDefaultConfigVariables 
20210207-090559 DBG 1639:      --> logOptions Standard option files
20210207-090559 DBG 1641:      --- Options:  -a : -o : -m detailed
20210207-090559 DBG 1642:      --- APPEND_LOG=0
20210207-090559 DBG 1643:      --- APPEND_LOG_OPTION=-a
20210207-090559 DBG 1644:      --- BACKUPPATH=/backup
20210207-090559 DBG 1645:      --- BACKUPTYPE=tar
20210207-090559 DBG 1646:      --- AFTER_STARTSERVICES=
20210207-090559 DBG 1647:      --- BEFORE_STOPSERVICES=
20210207-090559 DBG 1648:      --- CHECK_FOR_BAD_BLOCKS=0
20210207-090559 DBG 1649:      --- COLORING=CM
20210207-090559 DBG 1650:      --- CONFIG_FILE=
20210207-090559 DBG 1651:      --- DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20210207-090559 DBG 1652:      --- DD_BLOCKSIZE=1M
20210207-090559 DBG 1653:      --- DD_PARMS=
20210207-090559 DBG 1654:      --- DD_WARNING=0
20210207-090559 DBG 1655:      --- DEPLOYMENT_HOSTS=
20210207-090559 DBG 1656:      --- DYNAMIC_MOUNT=
20210207-090559 DBG 1657:      --- EMAIL=x@@@@r(18)
20210207-090559 DBG 1658:      --- EMAIL_COLORING=SUBJECT
20210207-090559 DBG 1659:      --- EMAIL_PARMS=
20210207-090559 DBG 1660:      --- EXCLUDE_LIST=
20210207-090559 DBG 1661:      --- EXTENSIONS=
20210207-090559 DBG 1662:      --- FAKE=
20210207-090559 DBG 1663:      --- HANDLE_DEPRECATED=
20210207-090559 DBG 1664:      --- IGNORE_ADDITIONAL_PARTITIONS=0
20210207-090559 DBG 1665:      --- KEEPBACKUPS=3
20210207-090559 DBG 1666:      --- KEEPBACKUPS_DD=0
20210207-090559 DBG 1667:      --- KEEPBACKUPS_DDZ=0
20210207-090559 DBG 1668:      --- KEEPBACKUPS_TAR=0
20210207-090559 DBG 1669:      --- KEEPBACKUPS_TGZ=0
20210207-090559 DBG 1670:      --- KEEPBACKUPS_RSYNC=0
20210207-090559 DBG 1671:      --- LANGUAGE=EN
20210207-090559 DBG 1672:      --- LINK_BOOTPARTITIONFILES=0
[... 793 lines cut ...]
20210207-090648 DBG 3178:          --- subject: @HOSTNAME@: Backup finished successfully.
20210207-090648 DBG 3178:          --- coloring: 
20210207-090648 DBG 3019:              --> colorAnnotation 0
20210207-090648 DBG 3043:              <-- colorAnnotation 
20210207-090648 DBG 3201:          --- eMail: x@@@@r(18)
20210207-090648 DBG 3202:          --- eMail Program: mail
20210207-090648 DBG 3203:          --- Subject: @HOSTNAME@: Backup finished successfully.
20210207-090648 DBG 3203:          --- MIME-Version: 1.0
20210207-090648 DBG 3203:          --- Content-Type: text/html; charset=utf-8
20210207-090648 DBG 3204:          --- ColoringOption: 
20210207-090648 DBG 3205:          --- ContentType: 
20210207-090648 DBG 3205:          --- MIME-Version: 1.0
20210207-090648 DBG 3205:          --- Content-Type: text/html; charset=utf-8
20210207-090648 DBG 3206:          --- Parms: 
20210207-090648 DBG 3211:          --- mail  -s "@HOSTNAME@: Backup finished successfully.
20210207-090648 DBG 3211:          --- MIME-Version: 1.0
20210207-090648 DBG 3211:          --- Content-Type: text/html; charset=utf-8" x@@@@r(18)
20210207-090649 DBG 3214:          --- mail: RC: 0
20210207-090649 DBG 3257:          <-- sendEMail 
20210207-090649 DBG 3364:      --- Masquerading eMail
20210207-090650 DBG 3387:      --- Masquerading some mount options
20210207-090650 DBG 3405:      --- Masquerading home directory name
20210207-090650 DBG 3410:      --- Masquerading hostname
20210207-090650 DBG 3415:      --- Masquerading sensitive non local IPs
20210207-090651 DBG 3574:      <-- cleanup

xMadNet, 2021-02-07:
I've copied from other Pi
    /usr/local/bin/raspiBackup.sh
    /mnt/RasPi/System Volume Information/raspiBackup.sh
raspiBackup.sh V0.6.5.1 (c295a0e)
Back up WORKS !!!???

Logs
20210207-100453 DBG 1673:      --> initializeDefaultConfig 
20210207-100453 DBG 1805:      <-- initializeDefaultConfig 
20210207-100453 DBG 2548:      --> readConfigParameters 
20210207-100453 DBG 2604:      <-- readConfigParameters 
20210207-100453 DBG 7128:      --> checkOptionParameter -a :
20210207-100453 DBG 7149:      <-- checkOptionParameter :
20210207-100454 DBG 7128:      --> checkOptionParameter -o :
20210207-100454 DBG 7149:      <-- checkOptionParameter :
20210207-100454 DBG 7128:      --> checkOptionParameter -m detailed
20210207-100454 DBG 7149:      <-- checkOptionParameter detailed
20210207-100454 DBG 1810:      --> initializeConfig 
20210207-100454 DBG 1878:      <-- initializeConfig 
20210207-100454 DBG 1907:      --> substituteNumberArguments 
20210207-100454 DBG 1934:      <-- substituteNumberArguments 
20210207-100454 DBG 6968:      --> check4RequiredCommands 
20210207-100454 DBG 6989:      <-- check4RequiredCommands 
20210207-100454 DBG 7717:  --- RESTORE: 0 - fileParameter: 
20210207-100454 DBG 7736:  --- Enabling trap handler
20210207-100454 DBG 1321:  --- TRAP cleanup SIGINT SIGTERM EXIT
20210207-100454 DBG 2609:      --> setupEnvironment 
20210207-100454 DBG 2613:      --- Previous logfile: /home/@USER@/raspiBackup.log
20210207-100454 DBG 2718:      --- +================================================================================================================================================+
20210207-100454 DBG 2719:      --- | ===> A lot of sensitive information is masqueraded in this log file. Nevertheless please check the log carefully before you distribute it <=== |
20210207-100454 DBG 2720:      --- +================================================================================================================================================+
20210207-100454 DBG 2722:      --- LOG_OUTPUT: 2
20210207-100454 DBG 2723:      --- Using logfile /backup/@HOSTNAME@/@HOSTNAME@-tar-backup-20210207-100453/raspiBackup.log
20210207-100454 DBG 2724:      --- Using msgfile /backup/@HOSTNAME@/@HOSTNAME@-tar-backup-20210207-100453/raspiBackup.msg
20210207-100454 DBG 2732:      --- BACKUPTARGET_DIR: /backup/@HOSTNAME@/@HOSTNAME@-tar-backup-20210207-100453
20210207-100530 DBG 2733:      --- BACKUPTARGET_FILE: /backup/@HOSTNAME@/@HOSTNAME@-tar-backup-20210207-100453/@HOSTNAME@-tar-backup-20210207-100453.tar
20210207-100530 DBG 2735:      <-- setupEnvironment 
20210207-100530 DBG 2862:      --> colorAnnotation 1
20210207-100530 DBG 2886:      <-- colorAnnotation 
--- RBK0009I: @HOSTNAME@: raspiBackup.sh V0.6.5.1 (c295a0e) started at Sun 07 Feb 2021 10:05:30 AM CET.
20210207-100530 DBG 2862:      --> colorAnnotation 1
20210207-100530 DBG 2886:      <-- colorAnnotation 
--- RBK0128I: Using logfile /backup/@HOSTNAME@/@HOSTNAME@-tar-backup-20210207-100453/raspiBackup.log.
20210207-100530 DBG 2862:      --> colorAnnotation 1
20210207-100530 DBG 2886:      <-- colorAnnotation 
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
20210207-100531 DBG 7761:  --- Read config /usr/local/etc/raspiBackup.conf : 0.1.5
20210207-100531 DBG 7761:  --- VERSION_CONFIG="0.1.5"
20210207-100531 DBG 7761:  --- DEFAULT_BACKUPPATH="/backup"
20210207-100531 DBG 7761:  --- DEFAULT_BACKUPTYPE="tar"
20210207-100531 DBG 7761:  --- DEFAULT_ZIP_BACKUP="0"
20210207-100531 DBG 7761:  --- DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20210207-100531 DBG 7761:  --- DEFAULT_KEEPBACKUPS="3"
20210207-100531 DBG 7761:  --- DEFAULT_KEEPBACKUPS_DD=0
20210207-100531 DBG 7761:  --- DEFAULT_KEEPBACKUPS_DDZ=0
20210207-100531 DBG 7761:  --- DEFAULT_KEEPBACKUPS_TAR=0
20210207-100531 DBG 7761:  --- DEFAULT_KEEPBACKUPS_TGZ=0
20210207-100531 DBG 7761:  --- DEFAULT_KEEPBACKUPS_RSYNC=0
20210207-100531 DBG 7761:  --- DEFAULT_SMART_RECYCLE=0
20210207-100531 DBG 7761:  --- DEFAULT_SMART_RECYCLE_DRYRUN=1
20210207-100531 DBG 7761:  --- DEFAULT_SMART_RECYCLE_OPTIONS="7 4 12 1"
20210207-100531 DBG 7761:  --- DEFAULT_STOPSERVICES="systemctl stop plexmediaserver"
20210207-100531 DBG 7761:  --- DEFAULT_STARTSERVICES="systemctl start plexmediaserver"
20210207-100531 DBG 7761:  --- DEFAULT_BEFORE_STOPSERVICES=""
20210207-100531 DBG 7761:  --- DEFAULT_AFTER_STARTSERVICES=""
20210207-100531 DBG 7761:  --- DEFAULT_EMAIL="x@@@@r(18)"
20210207-100531 DBG 7761:  --- DEFAULT_SENDER_EMAIL=""
20210207-100531 DBG 7761:  --- DEFAULT_EMAIL_PARMS=""
20210207-100531 DBG 7761:  --- DEFAULT_MAIL_PROGRAM="mail"
20210207-100531 DBG 7761:  --- DEFAULT_TELEGRAM_TOKEN=""
[... 771 lines cut ...]
20210207-100900 DBG 2862:              --> colorAnnotation 1
20210207-100900 DBG 2886:              <-- colorAnnotation 
--- RBK0167I: Sending email.
20210207-100900 DBG 2862:              --> colorAnnotation 0
20210207-100900 DBG 2886:              <-- colorAnnotation 
20210207-100900 DBG 3044:          --- eMail: x@@@@r(18)
20210207-100900 DBG 3045:          --- eMail Program: mail
20210207-100900 DBG 3046:          --- Subject: @HOSTNAME@: Backup finished successfully.
20210207-100900 DBG 3046:          --- MIME-Version: 1.0
20210207-100900 DBG 3046:          --- Content-Type: text/html; charset=utf-8
20210207-100900 DBG 3047:          --- ColoringOption: 
20210207-100900 DBG 3048:          --- ContentType: 
20210207-100900 DBG 3048:          --- MIME-Version: 1.0
[... cut ...]
20210207-100900 DBG 3054:          --- Content-Type: text/html; charset=utf-8" x@@@@r(18)
20210207-100902 DBG 3057:          --- mail: RC: 0
20210207-100902 DBG 3100:          <-- sendEMail 
20210207-100902 DBG 3209:      --- Masquerading eMail
20210207-100902 DBG 3232:      --- Masquerading some mount options
20210207-100902 DBG 3250:      --- Masquerading home directory name
20210207-100902 DBG 3255:      --- Masquerading hostname
20210207-100902 DBG 3260:      --- Masquerading sensitive non local IPs

framps, 2021-02-07:
```
20210207-090646 DBG 1474: --> executeCmd Command: tar -cpi --one-file-system -z -f "/backup/@hostname@/@hostname@-tgz-backup-20210207-090558/@HOSTNAME@-tgz-backup-20210207-090558.tgz" --warning=no-xdev --numeric-owner --exclude="/" --exclude="/backup/@hostname@/@hostname@-tgz-backup-20210207-090558/raspiBackup.log" --exclude="/backup/@hostname@/@hostname@-tgz-backup-20210207-090558/raspiBackup.msg" --exclude='.gvfs' --exclude=/proc/ --exclude=/lost+found/* --exclude=/sys/* --exclude=/dev/* --exclude=/tmp/* --exclude=/boot/* --exclude=/run/* --exclude=/var/cache/* /
```
Great you attached the log. That way I was able to find out the issue is already fixed. You grabbed a buggy version of raspiBackup - unfortunately :disappointed: 

Please execute `sudo raspiBackup.sh -U -S` to grab the latest raspiBackup version :wink: 

Sorry for the inconvenience.

framps, 2021-02-07:
@xMadNet Given the severity of this issue .... Would you please confirm everything is OK with the latest version now?

xMadNet, 2021-02-08:
Indeed fixed with raspiBackup.sh V0.6.6 (6dc5832)

closed 2021-02-08 by xMadNet