framps/raspiBackup #313: !!! RBK0157W: No services to stop while configured with sudo raspiBackupInstallUI.sh

xMadNet, 2021-02-07:
My config
raspiBackup.sh V0.6.6 (a9c8b83)
Rpi4 4 GB : model name : ARMv7 Processor rev 3 (v7l)
Linux XPi4-32 5.10.10-v7l+ #1397 SMP Mon Jan 25 18:42:27 GMT 2021 armv7l GNU/Linux
NAME MAJ:MIN RM SIZE RO TYPE MOUNTPOINT
sda 8:0 0 1.8T 0 disk
└─sda1 8:1 0 1.8T 0 part /Nas
sdb 8:16 0 119.2G 0 disk
├─sdb1 8:17 0 256M 0 part /boot
└─sdb2 8:18 0 119G 0 part /

Commandes
sudo raspiBackup.sh -a : -o : -m detailed
Result
!!! RBK0157W: No services to stop while I have configured them with sudo raspiBackupInstallUI.sh
So get a lot of
tar: /var/lib/samba/private/msg.sock/19356: socket ignored
Is Backup valid ??????

In log (see below I can find)
20210207-213524 DBG 1697:      --- STARTSERVICES=systemctl start smbd && systemctl start plexmediaserver
20210207-213524 DBG 1698:      --- STOPSERVICES=systemctl stop plexmediaserver && systemctl stop smbd
**??Why Stop services is not excecuted ??**
Thanks

Log
20210207-213524 DBG 1721:      --> initializeDefaultConfigVariables 
20210207-213524 DBG 1855:      <-- initializeDefaultConfigVariables 
20210207-213524 DBG 1860:      --> copyDefaultConfigVariables 
20210207-213524 DBG 1960:          --> substituteNumberArguments 
20210207-213524 DBG 1987:          <-- substituteNumberArguments 
20210207-213524 DBG 1931:      <-- copyDefaultConfigVariables 
20210207-213524 DBG 2702:      --> readConfigParameters 
20210207-213524 DBG 2758:      <-- readConfigParameters 
20210207-213524 DBG 1860:      --> copyDefaultConfigVariables 
20210207-213524 DBG 1960:          --> substituteNumberArguments 
20210207-213524 DBG 1987:          <-- substituteNumberArguments 
20210207-213524 DBG 1931:      <-- copyDefaultConfigVariables 
20210207-213524 DBG 1642:      --> logOptions Standard option files
20210207-213524 DBG 1644:      --- Options:  -a : -o : -m detailed
20210207-213524 DBG 1645:      --- APPEND_LOG=0
20210207-213524 DBG 1646:      --- APPEND_LOG_OPTION=-a
20210207-213524 DBG 1647:      --- BACKUPPATH=/backup
20210207-213524 DBG 1648:      --- BACKUPTYPE=tar
20210207-213524 DBG 1649:      --- AFTER_STARTSERVICES=
20210207-213524 DBG 1650:      --- BEFORE_STOPSERVICES=
20210207-213524 DBG 1651:      --- CHECK_FOR_BAD_BLOCKS=0
20210207-213524 DBG 1652:      --- COLORING=CM
20210207-213524 DBG 1653:      --- CONFIG_FILE=
20210207-213524 DBG 1654:      --- DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20210207-213524 DBG 1655:      --- DD_BLOCKSIZE=1M
20210207-213524 DBG 1656:      --- DD_PARMS=
20210207-213524 DBG 1657:      --- DD_WARNING=0
20210207-213524 DBG 1658:      --- DEPLOYMENT_HOSTS=
20210207-213524 DBG 1659:      --- DYNAMIC_MOUNT=
20210207-213524 DBG 1660:      --- EMAIL=x@@@@r(18)
20210207-213524 DBG 1661:      --- EMAIL_COLORING=SUBJECT
20210207-213524 DBG 1662:      --- EMAIL_PARMS=
20210207-213524 DBG 1663:      --- EXCLUDE_LIST=
20210207-213524 DBG 1664:      --- EXTENSIONS=
20210207-213524 DBG 1665:      --- FAKE=
20210207-213524 DBG 1666:      --- HANDLE_DEPRECATED=
20210207-213524 DBG 1667:      --- IGNORE_ADDITIONAL_PARTITIONS=0
20210207-213524 DBG 1668:      --- KEEPBACKUPS=3
20210207-213524 DBG 1669:      --- KEEPBACKUPS_DD=0
20210207-213524 DBG 1670:      --- KEEPBACKUPS_DDZ=0
20210207-213524 DBG 1671:      --- KEEPBACKUPS_TAR=0
20210207-213524 DBG 1672:      --- KEEPBACKUPS_TGZ=0
20210207-213524 DBG 1673:      --- KEEPBACKUPS_RSYNC=0
20210207-213524 DBG 1674:      --- LANGUAGE=EN
[... 849 lines cut ...]
20210207-213929 DBG 3181:          --- subject: @HOSTNAME@: Backup finished successfully.
20210207-213929 DBG 3181:          --- coloring: 
20210207-213929 DBG 3022:              --> colorAnnotation 0
20210207-213929 DBG 3046:              <-- colorAnnotation 
20210207-213929 DBG 3204:          --- eMail: x@@@@r(18)
20210207-213929 DBG 3205:          --- eMail Program: mail
20210207-213929 DBG 3206:          --- Subject: @HOSTNAME@: Backup finished successfully.
20210207-213929 DBG 3206:          --- MIME-Version: 1.0
20210207-213929 DBG 3206:          --- Content-Type: text/html; charset=utf-8
20210207-213929 DBG 3207:          --- ColoringOption: 
20210207-213930 DBG 3208:          --- ContentType: 
20210207-213930 DBG 3208:          --- MIME-Version: 1.0
20210207-213930 DBG 3208:          --- Content-Type: text/html; charset=utf-8
20210207-213930 DBG 3209:          --- Parms: 
20210207-213930 DBG 3214:          --- mail  -s "@HOSTNAME@: Backup finished successfully.
20210207-213930 DBG 3214:          --- MIME-Version: 1.0
20210207-213930 DBG 3214:          --- Content-Type: text/html; charset=utf-8" x@@@@r(18)
20210207-213931 DBG 3217:          --- mail: RC: 0
20210207-213931 DBG 3260:          <-- sendEMail 
20210207-213931 DBG 3369:      --- Masquerading eMail
20210207-213931 DBG 3392:      --- Masquerading some mount options
20210207-213931 DBG 3410:      --- Masquerading home directory name
20210207-213931 DBG 3415:      --- Masquerading hostname
20210207-213931 DBG 3420:      --- Masquerading sensitive non local IPs
20210207-213932 DBG 3582:      <-- cleanup

framps, 2021-02-07:
> sudo raspiBackup.sh -a : -o : -m detailed

Please read https://www.linux-tips-and-tricks.de/de/raspibackup/#parameter. If you call raspiBackup with these options the configured stop/start options are overwritten and finally with `:` ignored :wink:

xMadNet, 2021-02-08:
Thanks. Don't remmenber why I'd set this paramenters.
Commande
sudo raspiBackup.sh -m detailed
Could you write on screen,msg a message : Stoppping services ...... Starting services
At least with -m detailed

xMadNet, 2021-02-08:
Sorry
Still have the "tar: /var/lib/samba/private/msg.sock/19356: socket ignored " warning
I suppose that services are not stopped
In log I can read : 20210208-081238 DBG 1646:      --- APPEND_LOG_OPTION=-a
Don't know if it is the same -a ?
It is not set in 
5 | /usr/local/etc/raspiBackup.conf
4 | ~/.raspiBackup.conf
3 | $(pwd)/.raspiBackup.conf
2 | -f <configFile>
1 | Invocation options

Log
20210208-081238 DBG 1721:      --> initializeDefaultConfigVariables 
20210208-081238 DBG 1855:      <-- initializeDefaultConfigVariables 
20210208-081238 DBG 1860:      --> copyDefaultConfigVariables 
20210208-081238 DBG 1960:          --> substituteNumberArguments 
20210208-081238 DBG 1987:          <-- substituteNumberArguments 
20210208-081238 DBG 1931:      <-- copyDefaultConfigVariables 
20210208-081238 DBG 2702:      --> readConfigParameters 
20210208-081238 DBG 2758:      <-- readConfigParameters 
20210208-081238 DBG 1860:      --> copyDefaultConfigVariables 
20210208-081238 DBG 1960:          --> substituteNumberArguments 
20210208-081238 DBG 1987:          <-- substituteNumberArguments 
20210208-081238 DBG 1931:      <-- copyDefaultConfigVariables 
20210208-081238 DBG 1642:      --> logOptions Standard option files
20210208-081238 DBG 1644:      --- Options:  -m detailed
20210208-081238 DBG 1645:      --- APPEND_LOG=0
20210208-081238 DBG 1646:      --- APPEND_LOG_OPTION=-a
20210208-081238 DBG 1647:      --- BACKUPPATH=/backup
20210208-081238 DBG 1648:      --- BACKUPTYPE=tar
20210208-081238 DBG 1649:      --- AFTER_STARTSERVICES=
20210208-081238 DBG 1650:      --- BEFORE_STOPSERVICES=
20210208-081238 DBG 1651:      --- CHECK_FOR_BAD_BLOCKS=0
20210208-081238 DBG 1652:      --- COLORING=CM
20210208-081238 DBG 1653:      --- CONFIG_FILE=
20210208-081238 DBG 1654:      --- DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20210208-081238 DBG 1655:      --- DD_BLOCKSIZE=1M
20210208-081238 DBG 1656:      --- DD_PARMS=
20210208-081238 DBG 1657:      --- DD_WARNING=0
20210208-081238 DBG 1658:      --- DEPLOYMENT_HOSTS=
20210208-081238 DBG 1659:      --- DYNAMIC_MOUNT=
20210208-081238 DBG 1660:      --- EMAIL=x@@@@r(18)
20210208-081238 DBG 1661:      --- EMAIL_COLORING=SUBJECT
20210208-081238 DBG 1662:      --- EMAIL_PARMS=
20210208-081238 DBG 1663:      --- EXCLUDE_LIST=
20210208-081238 DBG 1664:      --- EXTENSIONS=
20210208-081238 DBG 1665:      --- FAKE=
20210208-081238 DBG 1666:      --- HANDLE_DEPRECATED=
20210208-081238 DBG 1667:      --- IGNORE_ADDITIONAL_PARTITIONS=0
20210208-081238 DBG 1668:      --- KEEPBACKUPS=3
20210208-081238 DBG 1669:      --- KEEPBACKUPS_DD=0
20210208-081238 DBG 1670:      --- KEEPBACKUPS_DDZ=0
20210208-081238 DBG 1671:      --- KEEPBACKUPS_TAR=0
20210208-081238 DBG 1672:      --- KEEPBACKUPS_TGZ=0
20210208-081238 DBG 1673:      --- KEEPBACKUPS_RSYNC=0
20210208-081238 DBG 1674:      --- LANGUAGE=EN
20210208-081238 DBG 1675:      --- LINK_BOOTPARTITIONFILES=0
20210208-081238 DBG 1676:      --- LOG_LEVEL=1
20210208-081238 DBG 1677:      --- LOG_OUTPUT=2
20210208-081238 DBG 1678:      --- MAIL_ON_ERROR_ONLY=0
20210208-081238 DBG 1679:      --- MAIL_PROGRAM=mail
20210208-081238 DBG 1680:      --- MSG_LEVEL=0
20210208-081238 DBG 1681:      --- NOTIFY_START=0
20210208-081238 DBG 1682:      --- NOTIFY_UPDATE=1
20210208-081238 DBG 1683:      --- PARTITIONBASED_BACKUP=0
20210208-081238 DBG 1684:      --- PARTITIONS_TO_BACKUP=1 2
20210208-081238 DBG 1685:      --- RESIZE_ROOTFS=1
20210208-081238 DBG 1686:      --- RESTORE_DEVICE=
20210208-081238 DBG 1687:      --- ROOT_PARTITION=
[... 839 lines cut ...]
20210208-081711 DBG 3181:          --- subject: @HOSTNAME@: Backup finished successfully.
20210208-081711 DBG 3181:          --- coloring: 
20210208-081711 DBG 3022:              --> colorAnnotation 0
20210208-081711 DBG 3046:              <-- colorAnnotation 
20210208-081711 DBG 3204:          --- eMail: x@@@@r(18)
20210208-081711 DBG 3205:          --- eMail Program: mail
20210208-081711 DBG 3206:          --- Subject: @HOSTNAME@: Backup finished successfully.
20210208-081711 DBG 3206:          --- MIME-Version: 1.0
20210208-081711 DBG 3206:          --- Content-Type: text/html; charset=utf-8
20210208-081711 DBG 3207:          --- ColoringOption: 
20210208-081711 DBG 3208:          --- ContentType: 
20210208-081711 DBG 3208:          --- MIME-Version: 1.0
20210208-081711 DBG 3208:          --- Content-Type: text/html; charset=utf-8
20210208-081711 DBG 3209:          --- Parms: 
20210208-081711 DBG 3214:          --- mail  -s "@HOSTNAME@: Backup finished successfully.
20210208-081711 DBG 3214:          --- MIME-Version: 1.0
20210208-081711 DBG 3214:          --- Content-Type: text/html; charset=utf-8" x@@@@r(18)
20210208-081712 DBG 3217:          --- mail: RC: 0
20210208-081712 DBG 3260:          <-- sendEMail 
20210208-081712 DBG 3369:      --- Masquerading eMail
20210208-081712 DBG 3392:      --- Masquerading some mount options
20210208-081713 DBG 3410:      --- Masquerading home directory name
20210208-081713 DBG 3415:      --- Masquerading hostname
20210208-081713 DBG 3420:      --- Masquerading sensitive non local IPs
20210208-081714 DBG 3582:      <-- cleanup

framps, 2021-02-08:
> Could you write on screen,msg a message : Stoppping services ...... Starting services

They are written with option -m ist you don't use `-o : -a :`.

**Again** Please don't copy and paste the whole log into the github issue. Just attach the log file ! It's very annoying because I have to scroll down a lot of time just to get to the next reply :angry: 

> Still have the "tar: /var/lib/samba/private/msg.sock/19356: socket ignored " warning
> I suppose that services are not stopped

Looks like samba is not stopped.

xMadNet, 2021-02-08:
I confirm there is NO messages when stoping the service : Enhancement ?
But services are stopped 
Since I still have tar: /var/lib/samba/private/msg.sock/8784: socket ignored!!!

framps, 2021-02-08:
>  NO messages when stoping the service

I was able to reproduce this issue and fixed it right now. Just grab again the latest 0.6.6 version.

> Since I still have tar: /var/lib/samba/private/msg.sock/8784: socket ignored!!!

Do you know exclamation marks are impolite?

framps, 2021-02-09:
@xMadNet Anything else?

framps, 2021-02-11:
Closed because of inactivity

closed 2021-02-11 by framps