framps/raspiBackup #314: Monitor the raspi

mcmellow, 2021-02-13:
If you have a question about raspiBackup please execute all following actions first and provide all requested information:

* Read raspiBackup FAQ's which answers a lot of common questions [English](https://www.linux-tips-and-tricks.de/en/faq) - [German](https://www.linux-tips-and-tricks.de/de/faq)

* Provide the output of `sudo raspiBackup.sh --version` or `sudo raspiBackupInstallUI.sh --version`
* Provide the output of `cat /etc/os-release`

If any action was not executed or any requested information was not provided the question will be rejected.

mcmellow, 2021-02-13:
Hi,

Is there a way to monitor the raspiBackup process. I have monit up an running to monitor other processes. I hope somebody can give me some tips

framps, 2021-02-13:
> How can I monitor raspiBackup while running. I have monit up and running onmy RPi 3B+

I'm sorry but I don't get your question.

mcmellow, 2021-02-13:
I want to do a unattended backup ar 03:00 at night.  I want to monitor the backup to make sure all is OK. Take action if something goes wrong.  Monit is a linux program to monitor things.
I hope this explains my question a bit more.

framps, 2021-02-13:
There are various ways to check whether raspiBackup succeeded or failed.

1. Check log for success or failure message (RBK0005E and RBK0017I)
2. Check return code of raspiBackup, RC 0 -> OK, anything else -> failure
3. Analyze eMail subject if email is configured
4. Write a telegram bot which analyzed the message sent by raspiBackup if telegram is configured

mcmellow, 2021-02-15:
Thanks for your reply.  I set a pre- and after-  command on services stopped/started in the raspiBackup.conf  This works OK.

There is another point  enabling mail by entering my email address with raspiBackupInstallUI.sh does not change anything in the  raspiBackup.conf file.  No mail is send after a fake run.
Is there another way to setup mail with nullmailer?

framps, 2021-02-15:
> nullmailer

Nullmailer is officially not supported but in issue #241 a user successfully used the nullmailer :wink: 

> does not change anything in the raspiBackup.conf file.

The installer allows to change the eMail address and the eMail MTA. Both updates are done in the config file. There was an issue #241 when i added email coloring and the user used nullmailer successfully. According the logs he used the `mail` client which is the default MTA. 
if there is no eMail sent please check the debug log and search for `RBK0167I`. The following lines will log the exact command executed by raspiBackup to send the eMail. Please execute the command manually and check it works.

mcmellow, 2021-02-16:
Issue can be closed.  Sending mail with NullMailer works OK now

closed 2021-02-16 by mcmellow

framps, 2021-02-16:
Congratulations :smiley: 

> Sending mail with NullMailer works OK now

Would be great if you could summarize why it didn't work for you and how you fixed it :wink:

mcmellow, 2021-02-16:
Thanks to **git-olicat** (Oliver) He gave me this link.
https://itwelt.org/anleitungen-howto/linux-740817696/624-nullmailer-einrichten-mails-an-smtp-relay-weiterleiten 
I followed these steps, but discovered the nullmailer was already configured like this.

sudo raspiBackup.sh -F -e john@doe.com worked OK.  Then I discovered the lines were missing in the raspiBackup.conf
```
# email to send completion status
DEFAULT_EMAIL="john@doe.com"
``` 
After adding these lines to raspiBackup.conf it worked OK, also without the **-e** parameter