framps/raspiBackup #315: Rootpartitionserstellung endet fehlerhaft mit RC 1

Tobius69, 2021-02-20:
Hallo,

ich weiß, dass ein User das gleiche Problem in #267 hatte. Allerdings wollte er wohl das Backup auf eine NTFS Partition erstellen.

ich habe ein Backup (mittels rsync) auf mein gemountetes Synology-Laufwerk gemacht und will dieses nun zurückspielen.

Im Pi steckt ein Kartenleser mit der zu beschreibenden SD-Karte (hier ist noch ein altes Raspbian mit zwei Partitionen drauf, welches nun überschrieben werden soll).

 So sieht das ganze aus:
 sudo parted -l

Model: TS-RDF5 SD Transcend (scsi)
Disk /dev/sda: 32,0GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  273MB   268MB   primary  fat32        lba
 2      273MB   32,0GB  31,7GB  primary  ext4

Model: SD SD32G (sd/mmc)
Disk /dev/mmcblk0: 31,3GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  273MB   268MB   primary  fat32        lba
 2      273MB   31,3GB  31,0GB  primary  ext4

[image]

Tobius69, 2021-02-20:
OK - hat sich erledigt! Beim vierten Versuch hat es nun doch geklappt. Ich habe den Kartenleser im Verdacht. Hatte den nochmal rein und raus und dann gings.

framps, 2021-02-20:
Da waere ich nun nicht so weiteres drauf gekommen. Gut dass Du es noch ein paar mal probiert hast denn ich haette mir den Wolf gesucht :cry: 

Bitte close den Issue noch :wink:

closed 2021-02-20 by Tobius69