framps/raspiBackup #316: Fehler nach Update

Buzz2912, 2021-02-20:
Hi, 

ich habe heute ein älteres Image geflasht und auf einem neuen Pi eingesetzt. Dabei raspiBackup mit -U auf die neuste Version gebracht. Danach eine funktionierende .conf Datei eingespielt und das ganze mittels cifs Wrapper Skript ausgeführt.

Auf Pi Nr. 1 trat der Fehler einmal auf und nach nicht wieder.
Auf Pi Nr. 2 tritt der Fehler bei jedem Backup auf.

Gleiches Image, unterschiedliche Services.

--- /mnt/NAS already mounted
--- RBK0009I: lenipi: raspiBackup.sh V0.6.6 (c35ee98) Sa 20. Feb 21:00:29 CET 2021 gestartet.
/usr/local/bin/raspiBackup.sh: Zeile 6691: (???? - 2021) * 12 + (?? - 2) : Syntax Fehler: Operator erwartet. (Fehlerverursachendes Zeichen ist \"???? - 2021) * 12 + (?? - 2) \").
/usr/local/bin/raspiBackup.sh: Zeile 6721: ((: <= -6 : Syntax Fehler: Operator erwartet. (Fehlerverursachendes Zeichen ist \"<= -6 \").
--- RBK0141I: Nur der von den definierten Partitionen belegte Speicherplatz wird gesichert.
--- RBK0151I: Backuppfad /mnt/NAS/images mit Partitionstyp cifs wird benutzt.
--- RBK0003I: Backupgröße wird von 29.81 GiB auf 8.07 GiB reduziert.
--- RBK0085I: Backuperstellung vom Typ dd gestartet. Bitte Geduld.

Danke, Sebastian

framps, 2021-02-20:
raspiBackup hat einen Mechanismus der einen regelmaessig erinnert einmal einen Restoretest auszufuehren. Dazu wird die Datei /var/lib/restore.reminder gelesen. Der Fehlermeldung  nach exisitiert die Datei aber hat keinen Inhalt. Warum das so ist weiss ich nicht. Ich werde den Code in raspiBackup diesbezueglich robuster machen. Jedenfalls sollte es reichen diese Datei zu loeschen. Sie wird dann wieder neu mit korrektem Inhalt angelegt und die Fehlermeldung sollte nicht mehr auftreten.

Buzz2912, 2021-02-20:
Hat funktioniert. Danke

closed 2021-02-20 by Buzz2912