framps/raspiBackup #319: Allgemeine Verständnisfrage

neuerlinuxuser, 2021-02-22:
Hallo Framps,

habe dein Prgramm durch Zufall gefunden und bin von den Möglichkeiten begeistert. Leider muss ich gestehen, dass ich sowohl bei Linux wie auch beim Raspi ein absoluter Neuling bin, und bisher keine Antwort im Forum auf meine Frage gefunden habe. Ich möchte zunächst einmal wissen, ob folgender Plan ausführbar ist.
Möchte den RaspiBackup auf einen Debian Rechner installieren. Dann übers Netzwerk einen Raspi sichern. Die Sicherung sollte auf dem Debian Rechner gespeichert werden.
Ist dieser Plan möglich?
Würde mich über eine Antwort sehr freuen.
mit freundlichen Grüßen
P.

framps, 2021-02-22:
raspiBackup ist dazu gedacht dass sich eine laufende Raspberry selbst sichert. D.h. Du kannst raspiBackup leider nicht fuer Deinen Usecase benutzen.

Du kannst aber raspiBackup auf Deiner Raspberry laufen lassen und die Daten auf Dein Debian per nfs sichern. Allerdings muss dann das Debian immer verfuegbar sein. Alternativ besorgst Du Dir eine weitere Raspberry und benutzt die als Backup-NAS.

neuerlinuxuser, 2021-02-22:
Vielen Dank für die schnelle Antwort :+1:

framps, 2021-02-22:
Bitte den Issue closen wenn alles soweit geklärt ist :wink:

neuerlinuxuser, 2021-02-24:
Vielen Dank für die schnelle Antwort :

closed 2021-02-24 by neuerlinuxuser