framps/raspiBackup #320: tar: /var/lib/samba/private/msg.sock/3454: socket ignored

xMadNet, 2021-02-22:
Hi, 
My config :
logSystem
20210222-091540 DBG 1596:          *** uname -a
20210222-091541 DBG 1596:              Linux @HOSTNAME@ 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
20210222-091541 DBG 1597:          *** cat /etc/os-release
20210222-091541 DBG 1597:              PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
20210222-091541 DBG 1597:              NAME="Raspbian GNU/Linux"
20210222-091541 DBG 1597:              VERSION_ID="10"
20210222-091541 DBG 1597:              VERSION="10 (buster)"
20210222-091541 DBG 1597:              VERSION_CODENAME=buster
20210222-091541 DBG 1597:              ID=raspbian
20210222-091541 DBG 1597:              ID_LIKE=debian
20210222-091541 DBG 1597:              HOME_URL="http://www.raspbian.org/"
20210222-091541 DBG 1597:              SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
20210222-091541 DBG 1597:              BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
20210222-091541 DBG 1598:          *** cat /etc/debian_version
20210222-091541 DBG 1598:              10.8

 --> isNewVersionAvailable
20210222-091541 DBG 2312:              --- Versionsplit: 0.6.6 -
20210222-091541 DBG 2341:              --- Returning: 0.0 0.0 0.6.6
20210222-091541 DBG 2343:              <-- isNewVersionAvailable 99

Despite I stopped smbd service
20210222-091539 DBG 1697:      --- STARTSERVICES=systemctl start smbd && systemctl start plexmediaserver
20210222-091539 DBG 1698:      --- STOPSERVICES=systemctl stop plexmediaserver && systemctl stop smbd
20210222-091540 DBG 8001:  --- DEFAULT_STOPSERVICES="systemctl stop plexmediaserver && systemctl stop smbd"
20210222-091540 DBG 8001:  --- DEFAULT_STARTSERVICES="systemctl start smbd && systemctl start plexmediaserver"
20210222-091540 DBG 1697:      --- STARTSERVICES=systemctl start smbd && systemctl start plexmediaserver
20210222-091540 DBG 1698:      --- STOPSERVICES=systemctl stop plexmediaserver && systemctl stop smbd
20210222-091542 DBG 2358:                  --- systemctl stop plexmediaserver && systemctl stop smbd
20210222-091542 DBG 1502:                      --> executeShellCommand systemctl stop plexmediaserver && systemctl stop smbd
20210222-092031 DBG 2404:              --- systemctl start smbd && systemctl start plexmediaserver
20210222-092031 DBG 1502:                  --> executeShellCommand systemctl start smbd && systemctl start plexmediaserver

I still got a lot of those issues (about 40).
Is Backup Ok ?
Is it possible to avoid those Issuers, Warning ?
Thanks

framps, 2021-02-22:
Are you sure samba was stopped? I suggest to use option `--systemstatus` to check which services are up and which files are open. I don't use samba in my environment but as far as I understand these socket errors happen if samba is up and running. You defined samba should be stopped but if you use the invocation options `-a : -o :` the config options will be overwritten and samba will not be stopped. If you look into the debug file you can see which options are finally in place when raspiBackup starts.

> Is it possible to avoid those Issuers, Warning ?

If samba is down and you want to get rid of the messages just use option `-u` to exclude these socket files (See [here](https://www.linux-tips-and-tricks.de/en/backup/#parm_u) for details about option u)

framps, 2021-03-18:
Closing because of inactivity

closed 2021-03-18 by framps