framps/raspiBackup #321: Allow to ignore runtime errors from backuptools

framps, 2021-03-01:
As of now any RC other than 0 from the backup tools dd, tar and rsync abort raspiBackup. There are raspiBackup users who want not the backup process to abort for given RCs. Add an option for the backup tools to define RCs which should be accepted but write a warning message.

rsync RCs:
```
       0      Success
       1      Syntax or usage error
       2      Protocol incompatibility
       3      Errors selecting input/output files, dirs
       4      Requested  action  not supported: an attempt was made to manipu‐
              late 64-bit files on a platform that cannot support them; or  an
              option  was specified that is supported by the client and not by
              the server.
       5      Error starting client-server protocol
       6      Daemon unable to append to log-file
       10     Error in socket I/O
       11     Error in file I/O
       12     Error in rsync protocol data stream
       13     Errors with program diagnostics
       14     Error in IPC code
       20     Received SIGUSR1 or SIGINT
       21     Some error returned by waitpid()
       22     Error allocating core memory buffers
       23     Partial transfer due to error
       24     Partial transfer due to vanished source files
       25     The --max-delete limit stopped deletions
       30     Timeout in data send/receive
       35     Timeout waiting for daemon connection
```

tar RCs:
```
       0      Successful termination.
       1      Some files differ.  If tar was invoked with the --compare
              (--diff, -d) command line option, this means that some
              files in the archive differ from their disk counterparts.
              If tar was given one of the --create, --append or --update
              options, this exit code means that some files were changed
              while being archived and so the resulting archive does not
              contain the exact copy of the file set.
       2      Fatal error.  This means that some fatal, unrecoverable
              error occurred.
```

christianTF, 2021-03-04:
In my issue, I'll just track the result of yours ;-)

Let my think out "loud":
I saw you have around 30 exit codes (https://github.com/framps/raspiBackup/blob/94bd7dd23a161a0be34e01193fb2f80253c909f1/raspiBackup.sh#L285) in a range starting at exitcode 101.
Wouldn't it be handy to have a range of "warning exit codes", where one of them could be the "source file changed" warning.

This would make the behaviour downwards compatible:
If a calling scripts checks for exit code != 0, it still will recognize an error.
On the other hand, a calling script (like LoxBerry Backup) that would be aware of the "warning range", could execute different steps for error and warning exit codes. 

Such a warning situation should not delete the backup after execution. It is on the user or calling script to check the exit code or logfile. 

This would make it downwards compatible, the warning exit codes can be used in calling scripts, and no-one looses backup because of a warning.

lg, Christian

framps, 2021-03-04:
I like your proposal. Frankly I didn't spent much time on raspiBackups RC design because I don't knew anybody who actually cares about the RCs. But I see your point. If I understand you correctly you want to have a warning RC range (e.g. 50-99) for runtime issues which will not abort raspiBackup. This requires the severity of the issue be defined by me. An alternate much more flexible approach would be to have a new config (e.g. RC_IS_WARNING) which can be used to disable abort of raspiBackup for the defined RCs.

Example: 
Let's assume the new RC "source changed" is 132 and "sorce vanished" is 133 then you just have top add  `132 133` to RC_IS_WARNING which will cause raspiBackup not to abort and return 132 or 133 when done.

christianTF, 2021-03-04:
This would be a possible approach too. 
It requires different rc lists for different backup types.
It requires me and every user to Try&Error your return codes to differentiate warnings from errors for every backup type. 
The backup is still lost in case of a warning that the user couldn't know as it possibly appeared the first time. 
The backup will terminate with an error, so for a user it's not clear that this might only be a warning. 

As maintainer of the backup tool, you know the types of rc's that are warnings better than every user, therefore I would prefer a consolidated state instead require me to "prettify" error levels to warnings. I'm not the "backup guy" - mainly I've created a user interface.

christianTF, 2021-03-04:
BTW I need to mention, that you and me as developers follow completely different basic approaches. 
raspiBackup tries to make as much as possible customizable, whereas LoxBerry as core system expects Plugin developers to be as simple as possible and for easiness to use, with as little configuration as required.
This with all side effects, e. g. (as you already know) some LoxBerry users even can't interpret an error message that states in clear words, what the error is 😉

framps, 2021-03-09:
Last night I got an RC 23 with rsync and found following messages in the log:

```
rsync: readlink_stat("/home/@USER@/Pi-hole/CONTRIBUTING.md") failed: Structure needs cleaning (117)
rsync: readlink_stat("/home/@USER@/Pi-hole/advanced") failed: Structure needs cleaning (117)
rsync: readlink_stat("/home/@USER@/Pi-hole/LICENSE") failed: Structure needs cleaning (117)
IO error encountered -- skipping file deletion
rsync: readlink_stat("/usr/share/mc/bin") failed: Structure needs cleaning (117)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
```

It's caused by a dying SD card. I restored the last backup on another SD card and the system is up and running again. Took me approx 15 minutes.

But this means it's not possible to check for RC23 and assume it's because of a vanishing file. The log has to be scanned and all `rsync:` messages have to be inspected for error strings which signal a vanishing file or similar errors which should be ignored. I have no list of possible error messages for rsync and tar thus it's impossible for me to know which error messages may be written and to parse these message :cry:

framps, 2021-05-12:
Development note:  In order to guard against any blame because an incomplete backup fails to start the backup has to be marked as incomplete and a warning message has to be written during restore.

framps, 2021-06-14:
@christianTF I spent some time how this feature can be implemented. That's my proposal:

In raspiBackup config file it's possible to define a list of regex for every backup tool which ist used by raspiBackup to parse the backup tool output. Every regex has a corresponding RC defined. If the regex matches this RC is a possible RC returned and all matches are passed to an exit you have to implement which reads and interprets the RCs and returns an exit code which forces  raspiBackup to keep or to delete the backup.

closed 2021-06-23 by framps

christianTF, 2021-08-04:
Seems to a nice solution, as I also can roll out a new config file in case that unknown errors occur that need (or need not) to be handled.