framps/raspiBackup #323: Log permission denied errors at end of successfull job

NTMMFTS, 2021-03-18:
Hi, thanks for raspiBackup!  I'm having a logging permission issue...

At the very end of my daily backup job I'm getting the following five errors sent by email after the job success email is sent, and the job log is otherwise filled with normal logging for the job.

	/usr/local/bin/raspiBackup.sh: line 1536: /<job path>/raspiBackup.log: Permission denied
	/usr/local/bin/raspiBackup.sh: line 1536: /<job path>/raspiBackup.log: Permission denied
	/usr/local/bin/raspiBackup.sh: line 1536: /<job path>/raspiBackup.log: Permission denied
	/usr/local/bin/raspiBackup.sh: line 1536: /<job path>/raspiBackup.log: Permission denied
	/usr/local/bin/raspiBackup.sh: line 1536: /<job path>/raspiBackup.log: Permission denied

These are the last two lines of job log so I assumed the five logging attempts were afterwards:

	20210317-150211 DBG 3266:          <-- sendEMail 
	20210317-150211 DBG 3375:      --- Masquerading eMail

So I added this additional logging command to raspiBackup.sh after line 1536 that appends a separate log file with 777 permissions:

	echo "$USER $dte $m" >> /root/raspibackup_line_1536_error.log

And this is what was logged as the last 5 lines:
	
	20210317-150211 DBG 3398:      --- Masquerading some mount options
	20210317-150211 DBG 3416:      --- Masquerading home directory name
	20210317-150211 DBG 3421:      --- Masquerading hostname
	20210317-150211 DBG 3426:      --- Masquerading sensitive non local IPs
	20210317-150212 DBG 3588:      <-- cleanup 

The $USER var didn't show up even though the backup runs as root in cron.d.

After the job is done, the log file in the backup directory has 555 permissions, and the backup and the log are on a r/w cifs mount with no permission problems, as evidenced by the whole job otherwise completing successfully.

Any ideas why the job log is becoming unwritable for these last five log entries?

Thanks!

NTMMFTS, 2021-03-18:
Using V0.6.6 (83ecd03).

framps, 2021-03-18:
Would you please attach the debug log?

NTMMFTS, 2021-03-18:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/6164886/raspiBackup.log)

Thanks for taking a look.

framps, 2021-03-18:
Looks to me there is some kind of race condition which reveals a minor bug in raspiBackup.

Starting in line 3574 there is following code:
```
        if (( $FILEDESCRIPTORS_SWAPPED )); then
		exec >&3 2>&4 # free logfile
	fi

	if (( $LOG_LEVEL == $LOG_DEBUG )); then
		masqueradeSensitiveInfoInLog # and now masquerade sensitive details in log file
	fi

	if [[ -n "$DYNAMIC_MOUNT" ]] && (( $DYNAMIC_MOUNT_EXECUTED )); then
		logItem "Umount of $DYNAMIC_MOUNT scheduled"
		nohup umount $DYNAMIC_MOUNT &> /tmp/raspiBackup.nohup </dev/null &
	fi
```

Would you please change the code as following and start a backup again?

```
	if (( $LOG_LEVEL == $LOG_DEBUG )); then
		masqueradeSensitiveInfoInLog # and now masquerade sensitive details in log file
	fi

	if [[ -n "$DYNAMIC_MOUNT" ]] && (( $DYNAMIC_MOUNT_EXECUTED )); then
		logItem "Umount of $DYNAMIC_MOUNT scheduled"
		sleep 3; nohup umount $DYNAMIC_MOUNT &> /tmp/raspiBackup.nohup </dev/null &
	fi

       if (( $FILEDESCRIPTORS_SWAPPED )); then
		exec >&3 2>&4 # free logfile
	fi
```

I noticed you redacted following line:
```
20210317-150008 DBG 5045:              //<redacted> on /mnt/Backups type cifs (rw,relatime,vers=1.0,cache=strict,username=@@@@,domain=@@@@,uid=0,noforceuid,gid=0,noforcegid,addr=<redacted>,file_mode=0755,dir_mode=0755,soft,nounix,serverino,mapposix,rsize=61440,wsize=65536,bsize=1048576,echo_interval=60,actimeo=1)
```

which is fine. Could you give me a hint what's the format of the text? If possible I'd like to add code in raspiBackup which will masquerade what you redacted. Actually it looks to me it's an IP address and all external IP addresses are masqueraded by raspiBackup already. But because of the issue this didn't happen. Is my assumption correct?

NTMMFTS, 2021-03-18:
Thanks again, and I'll give that a try after work today and give you feedback.  The IP is actually local, so there's really no harm in sharing it here, and the RPi is on the same /24 subnet.

20210317-150008 DBG 5045:              //10.10.10.3/Jay/Raspberry Pi on /mnt/Backups type cifs (rw,relatime,vers=1.0,cache=strict,username=admin,domain=10.10.10.3,uid=0,noforceuid,gid=0,noforcegid,addr=10.10.10.3,file_mode=0755,dir_mode=0755,soft,nounix,serverino,mapposix,rsize=61440,wsize=65536,bsize=1048576,echo_interval=60,actimeo=1)
 20210317-150008 DBG 5045:              //10.10.10.3/Multimedia on /mnt/Multimedia type cifs (rw,relatime,vers=1.0,cache=strict,username=admin,domain=10.10.10.3,uid=0,noforceuid,gid=0,noforcegid,addr=10.10.10.3,file_mode=0755,dir_mode=0755,soft,nounix,serverino,mapposix,rsize=61440,wsize=65536,bsize=1048576,echo_interval=60,actimeo=1,x-systemd.automount)

NTMMFTS, 2021-03-19:
The updated code fixed the log file permission problem and the job still finished successfully, but it did log these sed errors at the end of the job.:

20210318-151031 DBG 3375:      --- Masquerading eMail
sed: cannot rename /mnt/Backups/pi.jaylewis.us/pi.jaylewis.us-tar-backup-20210318-150820/sedVsnboa: File exists
20210318-151032 DBG 3398:      --- Masquerading some mount options
sed: cannot rename /mnt/Backups/pi.jaylewis.us/pi.jaylewis.us-tar-backup-20210318-150820/sedgrQKLd: File exists
sed: cannot rename /mnt/Backups/pi.jaylewis.us/pi.jaylewis.us-tar-backup-20210318-150820/sedYhAIK9: File exists
20210318-151034 DBG 3416:      --- Masquerading home directory name
sed: cannot rename /mnt/Backups/pi.jaylewis.us/pi.jaylewis.us-tar-backup-20210318-150820/sedlQEjjd: File exists
20210318-151035 DBG 3421:      --- Masquerading hostname
sed: cannot rename /mnt/Backups/pi.jaylewis.us/pi.jaylewis.us-tar-backup-20210318-150820/sedwGxtLc: File exists
20210318-151036 DBG 3426:      --- Masquerading sensitive non local IPs
sed: cannot rename /mnt/Backups/pi.jaylewis.us/pi.jaylewis.us-tar-backup-20210318-150820/sedkZZIOR: File exists
sed: cannot rename /mnt/Backups/pi.jaylewis.us/pi.jaylewis.us-tar-backup-20210318-150820/sedk3vw0O: File exists
sed: cannot rename /mnt/Backups/pi.jaylewis.us/pi.jaylewis.us-tar-backup-20210318-150820/sedSGjcaM: File exists
20210318-151040 DBG 3601:      <-- cleanup 

The debug log is attached.
Thanks again :-)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/6168011/raspiBackup.log)

framps, 2021-03-19:
Now I remember why I added the exec before the masquerading stuff :smiley: .

Would you please use the following code now:
```
        if (( $FILEDESCRIPTORS_SWAPPED )); then
		exec >&3 2>&4 # free logfile
                sleep 5s      # give OS some time to close log files 
	fi

	if (( $LOG_LEVEL == $LOG_DEBUG )); then
		masqueradeSensitiveInfoInLog # and now masquerade sensitive details in log file
	fi

	if [[ -n "$DYNAMIC_MOUNT" ]] && (( $DYNAMIC_MOUNT_EXECUTED )); then
		logItem "Umount of $DYNAMIC_MOUNT scheduled"
		nohup umount $DYNAMIC_MOUNT &> /tmp/raspiBackup.nohup </dev/null &
	fi
```

NTMMFTS, 2021-03-19:
g'morning,

The backup still completes successfully and does not send email showing an error, but five of these errors are still generated on the console, and they are not in the log (attached):

/usr/local/bin/raspiBackup.sh: line 1536: /mnt/Backups/<hostname>/<hostname>-tar-backup-20210319-041819/raspiBackup.log: Permission denied

[raspiBackup.log](https://github.com/framps/raspiBackup/files/6173153/raspiBackup.log)

framps, 2021-03-19:
:cry: Ok. I will try to reproduce your issue on my site. Looks like there is some detailed debugging required. Hope I can get it done :crossed_fingers: 

I can see you use cifs for your backup partition. Frankly I'm a Linux guy and don't use cifs but nfs which is much faster. Would you please let me know which OS you use for your cifs partition and how is the partition fromatted? FAT32, EXT4, NTFS, ... ?

Anyhow the backup succeeds so there is no major issue. Just ignore the error messages. 

I will keep you posted :wink:

framps, 2021-03-19:
## This post is just to track my investigations on the issue. 

I mounted my Synology which exposes a cifs partition and get followin mount reply:
```
//192.168.0.9/Scratch on /backup type cifs (rw,relatime,vers=2.0,cache=strict,username=@@@@,uid=1000,forceuid,gid=0,noforcegid,addr=192.168.0.9,file_mode=0755,dir_mode=0755,soft,nounix,serverino,mapposix,rsize=65536,wsize=65536,bsize=1048576,echo_interval=60,actimeo=1)

``` 
The debug log shows
```
//10.10.10.3/Jay/Raspberry Pi on /mnt/Backups type cifs (rw,relatime,vers=1.0,cache=strict,username=@@@@,domain=@@@@,uid=0,noforceuid,gid=0,noforcegid,addr=10.10.10.3,file_mode=0755,dir_mode=0755,soft,nounix,serverino,mapposix,rsize=61440,wsize=65536,bsize=1048576,echo_interval=60,actimeo=1)
```

Major difference: I use v2.0 wheras user uses v1.0

framps, 2021-03-19:
@NTMMFTS Would you please also provide the output of
`df -T | grep /mnt/Backups`
?

NTMMFTS, 2021-03-19:
Hi,

My storage is a QNAP TS-469 Pro, and the partition is ext3.  Kind of old, but chugging along nicely.

Results from df -T | grep /mnt/Backups:
//10.10.10.3/Jay/Raspberry Pi cifs     5691351016 3191770140 2499580876  57% /mnt/Backups

I see your notes above, not sure why I specified cifs version 1.0, but I've been doing that for many years for mounts on other devices so I used the same syntax with the RPi without even questioning it.

This QNAP supports NFS, so let me know if you recommend trying it.

Thanks again...

framps, 2021-03-19:
Ok. I now know why fstype Pi was written in the message:
`//10.10.10.3/Jay/Raspberry\040Pi /mnt/Backups cifs vers=1.0,credentials=/root/.smbcredentials,x-systemd.automount 0 0`
\040 is a space and this causes the parser in raspiBackup to grap Pi instead of cifs :cry: 

But that's not the root cause for your issue - it's just a minor bug in raspiBackup. Frankly I just guess the issue is related to some cifs configuration on your side. If you switch to nfs you will get better performance and I'm confident the issue will be gone because the raspiBackup regression tests use nfs to save the backups.

There are other users of raspiBackup who use cifs and don't report any issues. That's why I think it's related to some cifs configuration on your side.

I suggest to switch to nfs. Make sure you use `no_root_squash`.

That's my line in my /etc/fstab:
`synology:/disks/silver   /backup  nfs     users,rw,sync,hard,intr,noauto,user     0       0`

That's my corresponding lin in /etc/exports:
`/disks/silver           *(rw,sync,no_subtree_check,no_root_squash)`

NTMMFTS, 2021-03-19:
Will definitely take your advice and test with NFS this weekend, and may also try cifs v2 or v3 if possible.  I'll give feedback here afterwards.

Thanks so much for your AMAZING support for this issue, really appreciate it.

framps, 2021-03-19:
It may be a combination of version and the other cifs configuration differences I found. 
It's up to you to test by trial and error the cifs stuff. I suggest to switch to nfs first. An then test in your spare time what's wrong with the cifs configuration :wink: 

> Thanks so much

You're welcome

NTMMFTS, 2021-03-20:
No errors at all with nfs which is awesome, but I did get some Interesting performance results.

With the cifs mount the last 7 backups took between 00:01:56 and 00:02:01.  With nfs with using the sync option, the backup took 00:11:44.  I changed it to async and the backup took 00:03:11.

This is /etc/fstab:

10.10.10.3:/Jay/Raspberry\040Pi /mnt/Backups nfs users,rw,hard,intr,auto,user 0 0

This is /proc/mounts:

10.10.10.3:/Jay/Raspberry\040Pi /mnt/Backups nfs4 rw,nosuid,nodev,noexec,relatime,vers=4.1,rsize=32768,wsize=32768,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=10.10.10.225,local_lock=none,addr=10.10.10.3 0 0

Any idea why so much slower with nfs than with cifs, especially when using async?  The RPi is a model 4B with 8GB of ram, assume it shouldn't be a horsepower issue with top looking really good:

top - 17:34:32 up 9 min,  2 users,  load average: 0.14, 0.34, 0.27
Tasks: 147 total,   2 running, 145 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.1 us,  0.7 sy,  0.0 ni, 99.2 id,  0.1 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   7812.4 total,   6231.7 free,    591.7 used,    988.9 buff/cache
MiB Swap:    100.0 total,    100.0 free,      0.0 used.   7104.6 avail Mem 

Thnx!

NTMMFTS, 2021-03-20:
Hold off on that...  there was another server doing heavy writes on the QNAP while I was testing. :-(

Can't test again until later, will let you know how it goes. :-)

framps, 2021-03-20:
No problem. nfs performs much better than cifs ... but you have to make sure the same environmental conditions exist :wink:

NTMMFTS, 2021-03-21:
Looking good, today's backup time was 00:02:22.  That's with the unmodified V0.6.6 (83ecd03) raspiBackup.sh and async nfs,  I may try sync again, but don't think it's critical unless you think otherwise.  And since you recommended nfs over cifs I've been changing the mounts on my other servers to nfs and they're all working well - including Veeam B&R 11 community edition. :-).  Thanks again!

framps, 2021-03-21:
I suggest to stick with async. Maybe [this article](https://www.admin-magazine.com/HPC/Articles/Useful-NFS-Options-for-Tuning-and-Management) about nfs performance tuning is worth reading :wink: 

Fankly I didn't fix your issue but given the fact it's some special environment issue on your side I think we can just close this issue. If that's OK with you please close the issue.

NTMMFTS, 2021-03-21:
Thanks for the article, and I'm good to go with this issue and I'm closing it.

I actually jumped the gun too soon with switching everything from cifs to nsf - I couldn't get permissions working for a plex docker container using nfs.  There's lots of ppl with the same issue when the container is start with a uid/gid of a user on the docker host, and most of the threads I found were ppl trying on a QNAP.  I think it's probably a poor nfs implementation on my 5+ year old QNAP TS-469 Pro, and there hasn't been a firmware up date in a long time.  I went back to a cifs mount for now until I figure out if I can fix it or not.

Thanks again for all your help, really appreciate it!

closed 2021-03-21 by NTMMFTS