framps/raspiBackup #325: raspiBackup.conf: parameter DEFAULT_DYNAMIC_MOUNT – no umount after backup

andyzle, 2021-03-22:
raspiBackup.conf: parameter DEFAULT_DYNAMIC_MOUNT – no umount after backup
Hello @framps,
after moving from my Raspberry Pi3 to Pi 4 I set up raspibackup from scratch and there in raspiBackup.conf I set the parameter DEFAULT_DYNAMIC_MOUNT for the first time.
According to the backup log the backup directory was correctly mounted but after successful backup it was not un-mounted, may you could check your script?

Here are the log details
According to log backup directory was mounted:
```
==> raspiBackup.log <==
20210321-140622 DBG 2069:          --> dynamic_mount /media/nas-backup
20210321-140622 DBG 2618:              --> isMounted /media/nas-backup
```
… and at the end of the successful backup process tried to umount:
```
==> raspiBackup.log <==
20210321-153053 DBG 3583:      --- Umount of /media/nas-backup scheduled
==> raspiBackup.msg <==
--- RBK0260I: Dynamischer umount von /media/nas-backup wird vorgenommen.
```
But when I check the mountpoint afterwards it was not unmounted:
```
pi@raspi4:/var/log $ mount | grep nfs
192.168.2.xxx:/volume1/raspi4backup on /media/nas-backup type nfs4 (rw,relatime,vers=4.0,rsize=131072,wsize=131072,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=192.168.2.yyy,local_lock=none,addr=192.168.2.xxx)
```

framps, 2021-03-22:
The command executed to umount the backup is

`nohup umount $DYNAMIC_MOUNT &> /tmp/raspiBackup.nohup </dev/null &`
which is executed asynchrounusly and no success/failure check can be done any more.

Would you check /tmp/raspiBackup.nohup for any messages?

andyzle, 2021-03-23:
/tmp/raspiBackup.nohup:
`umount.nfs4: /media/nas-backup: device is busy`
Maybe a file or directory via mountpoint was still open of the backup process, see e.g. [umount: device is busy. Why?](http://oletange.blogspot.com/2012/04/umount-device-is-busy-why.html)

framps, 2021-03-23:
Frankly that's what I expected: There is or are still some file(s) open. 

There may be some race condition and the log file used previously by raspiBackup was not freed for some timing reasons.

```
	if [[ -n "$DYNAMIC_MOUNT" ]] && (( $DYNAMIC_MOUNT_EXECUTED )); then
		logItem "Umount of $DYNAMIC_MOUNT scheduled"
		nohup umount $DYNAMIC_MOUNT &> /tmp/raspiBackup.nohup </dev/null &
	fi
```

Would you please update the code as follows:

```
	if [[ -n "$DYNAMIC_MOUNT" ]] && (( $DYNAMIC_MOUNT_EXECUTED )); then
		logItem "Umount of $DYNAMIC_MOUNT scheduled"
                sync
                sleep 1m          
                lsof +D $BACKUPTARGET_DIR
		nohup umount $DYNAMIC_MOUNT &> /tmp/raspiBackup.nohup </dev/null &
	fi
```

That way you will get a list of all files still open on the backup partition. Maybe there is no file open anymore due to the delay. Please provide the list of open files if you still get an umount failure.

andyzle, 2021-03-23:
New backup with changed code - /tmp/raspiBackup.nohup: same message as before (“device is busy”)
Last 3 lines of terminal output:
`--- RBK0167I: Eine eMail wird versendet.`
`lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs`
      `Output information may be incomplete.`
According to [this thread](https://unix.stackexchange.com/questions/171519/lsof-warning-cant-stat-fuse-gvfsd-fuse-file-system) I think lsof output is misleading in this case.
Maybe [that thread](https://askubuntu.com/questions/1324927/cannot-umount-tmp-both-fuser-and-lsof-show-nothing-locking-that-lv) might help.

framps, 2021-03-24:
Hm ... please post you /etc/fstab contents for /media/nas-backup. Looks like you use fuse to mount your backuppartition. 

I never used fuse for this. That's the fstab definition I use for my backuppartition:
```
asterix:/disks/silver   /backup  nfs     users,rw,sync,hard,intr,noauto,user     0       0
```

andyzle, 2021-03-25:
My fstab content for the mountpoint:
`# Entry for the NAS backup, mount with NFS version 4`
`192.168.2.xxx:/volume1/raspi4backup      /media/nas-backup       nfs     noauto,rw       0       0`
As you can see I do not register a fuse filesystem, and I have no idea why lsof shows a warning referring to this.
On my Raspberry 3 I use the raspiBackupWrapper script to mount and umount the mountpoint with the same configuration and it works perfectly.

framps, 2021-03-26:
> and I have no idea why lsof shows a warning referring to this.

Neither do I. I unfortunately cannot reproduce your issue on my system. Would you please add following line
```
 lsof / | awk 'NR==1 || $4~/[0-9][uw]/'
```

just before 
```
		logItem "Umount of $DYNAMIC_MOUNT scheduled"
```
?

That way we get a list of open files and - crossing fingers - an idea what's the root cause for the issue.

andyzle, 2021-03-27:
With the new lsof command that was the additional terminal output:
```
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
COMMAND     PID             USER   FD   TYPE DEVICE SIZE/OFF    NODE NAME
rsyslogd    358             root    7w   REG    8,2    50452  922254 /var/log/syslog
rsyslogd    358             root    8w   REG    8,2   181295  922587 /var/log/messages
rsyslogd    358             root    9w   REG    8,2   898214  915890 /var/log/daemon.log
rsyslogd    358             root   10w   REG    8,2    60870  922480 /var/log/auth.log
rsyslogd    358             root   11w   REG    8,2      518  922564 /var/log/user.log
rsyslogd    358             root   12w   REG    8,2   179741  922329 /var/log/kern.log
xrdp-sesm  5985             root    3w   REG    8,2    10898  922551 /var/log/xrdp-sesman.log
xrdp       5995             xrdp    3w   REG    8,2    59883  922553 /var/log/xrdp.log
applet.py  7161               pi    1w   REG    8,2    15512 1046550 /home/pi/.cache/lxsession/LXDE-pi/run.log
applet.py  7161               pi    2w   REG    8,2    15512 1046550 /home/pi/.cache/lxsession/LXDE-pi/run.log
applet.py  7480               pi    1w   REG    8,2    15512 1046550 /home/pi/.cache/lxsession/LXDE-pi/run.log
applet.py  7480               pi    2w   REG    8,2    15512 1046550 /home/pi/.cache/lxsession/LXDE-pi/run.log
applet.py 13442               pi    1w   REG    8,2    15512 1046550 /home/pi/.cache/lxsession/LXDE-pi/run.log
applet.py 13442               pi    2w   REG    8,2    15512 1046550 /home/pi/.cache/lxsession/LXDE-pi/run.log
applet.py 13684               pi    1w   REG    8,2    15512 1046550 /home/pi/.cache/lxsession/LXDE-pi/run.log
applet.py 13684               pi    2w   REG    8,2    15512 1046550 /home/pi/.cache/lxsession/LXDE-pi/run.log
applet.py 13922               pi    1w   REG    8,2    15512 1046550 /home/pi/.cache/lxsession/LXDE-pi/run.log
applet.py 13922               pi    2w   REG    8,2    15512 1046550 /home/pi/.cache/lxsession/LXDE-pi/run.log
applet.py 14166               pi    1w   REG    8,2    15512 1046550 /home/pi/.cache/lxsession/LXDE-pi/run.log
applet.py 14166               pi    2w   REG    8,2    15512 1046550 /home/pi/.cache/lxsession/LXDE-pi/run.log
cupsd     18103             root   11u   REG    8,2      224  922271 /var/log/cups/access_log
smbd      22878             root    2w   REG    8,2      504  915718 /var/log/samba/log.smbd
smbd      22878             root    5u   REG    8,2   430080  922496 /var/lib/samba/private/secrets.tdb
smbd      22878             root    9w   REG    8,2      504  915718 /var/log/samba/log.smbd
smbd      22878             root   22u   REG    8,2   421888  922500 /var/lib/samba/share_info.tdb
smbd      22878             root   23u   REG    8,2  1609728  922501 /var/cache/samba/gencache.tdb
smbd      22878             root   25u   REG    8,2      696  922502 /var/lib/samba/group_mapping.tdb
smbd      22878             root   26u   REG    8,2   421888  922503 /var/lib/samba/account_policy.tdb
smbd      22878             root   27u   REG    8,2   421888  922505 /var/lib/samba/private/passdb.tdb
smbd-noti 22880             root    2w   REG    8,2      504  915718 /var/log/samba/log.smbd
smbd-noti 22880             root    5u   REG    8,2   430080  922496 /var/lib/samba/private/secrets.tdb
smbd-noti 22880             root    9w   REG    8,2      504  915718 /var/log/samba/log.smbd
cleanupd  22881             root    2w   REG    8,2      504  915718 /var/log/samba/log.smbd
cleanupd  22881             root    5u   REG    8,2   430080  922496 /var/lib/samba/private/secrets.tdb
cleanupd  22881             root    9w   REG    8,2      504  915718 /var/log/samba/log.smbd
lpqd      22884             root    2w   REG    8,2      504  915718 /var/log/samba/log.smbd
lpqd      22884             root    5u   REG    8,2   430080  922496 /var/lib/samba/private/secrets.tdb
lpqd      22884             root   13w   REG    8,2      504  915718 /var/log/samba/log.smbd
lpqd      22884             root   22u   REG    8,2   421888  922500 /var/lib/samba/share_info.tdb
lpqd      22884             root   23u   REG    8,2  1609728  922501 /var/cache/samba/gencache.tdb
lpqd      22884             root   25u   REG    8,2      696  922502 /var/lib/samba/group_mapping.tdb
lpqd      22884             root   26u   REG    8,2   421888  922503 /var/lib/samba/account_policy.tdb
lpqd      22884             root   27u   REG    8,2   421888  922505 /var/lib/samba/private/passdb.tdb
nmbd      22909             root    2w   REG    8,2     3497  922515 /var/log/samba/log.nmbd
nmbd      22909             root    9w   REG    8,2     3497  922515 /var/log/samba/log.nmbd
```
In this output I do not see anything which might have an effect on the unmount command.

framps, 2021-03-27:
> In this output I do not see anything which might have an effect on the unmount command. 

Neither do I. The way the umount is scheduled is a little bit tricky - but didn't create any issues until now. I plan to rewrite the whole logging in next release because of other reasons and will not need this trick any more.  So I don't have any idea how to fix the issue.

I frankly have no clue where this fuse error comes from now. You wrote you use the wrapper script on another system. As a mitigation I suggest to use a wrapper script again, execute a mount just before raspiBackup is started and a umount at the end when it's finished. That's the way I did it on my systems before I added the dynamic mount feature. Crossing fingers you will not face the same issue when you use the wrapper script. Try the dynamic mount option again when next release is out.

Anyhow - thank you very much for your help to find the root cause for the issue even we were not able to find it.

andyzle, 2021-03-27:
Thank you for your support – I will use the wrapper script to mount/umount also for my new Raspberry Pi 4 and will check the dynamic mount feature after you released the new version.

framps, 2021-03-27:
Great. Please close this issue and let me know in this issue later on if you also have the same issue with the wrapper script.

DrDeath, 2021-11-09:
Hi, just found this old issue, but I have this issue also on V0.6.6.1 (6c10535)

The dynamic mount will not un-mount after backup.

framps, 2021-11-09:
Hm ... Please provide the debug log.

DrDeath, 2021-11-09:
`11-09-2021 14:38:45 --- RBK0009I: raspberrypi: raspiBackup.sh V0.6.6.1 (6c10535) Di 9. Nov 14:38:45 CET 2021 gestartet.
11-09-2021 14:38:46 --- RBK0259I: Dynamischer mount von /media/nas erfolgreich.
11-09-2021 14:38:46 --- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
11-09-2021 14:38:48 --- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
11-09-2021 14:38:50 --- RBK0151I: Backuppfad /media/nas/backups mit Partitionstyp nfs4 wird benutzt.
11-09-2021 14:38:51 --- RBK0008I: Services werden gestoppt: ' '.
11-09-2021 14:38:51 --- RBK0081I: Backup vom Typ rsync wird in /media/nas/backups/raspberrypi/raspberrypi-rsync-backup-20211109-143844 erstellt.
11-09-2021 14:38:51 --- RBK0036I: Partitionslayout wird gesichert.
11-09-2021 14:38:51 --- RBK0044I: Backup der Bootpartition wird in /media/nas/backups/raspberrypi/raspberrypi-rsync-backup-20211109-143844/raspberrypi-backup.img erstellt.
11-09-2021 14:40:10 --- RBK0044I: Backup des Partitionlayouts wird in /media/nas/backups/raspberrypi/raspberrypi-rsync-backup-20211109-143844/raspberrypi-backup.sfdisk erstellt.
11-09-2021 14:40:10 --- RBK0046I: Backup des Masterbootrecords wird in /media/nas/backups/raspberrypi/raspberrypi-rsync-backup-20211109-143844/raspberrypi-backup.mbr erstellt.
11-09-2021 14:40:10 --- RBK0133I: Verzeichnis /media/nas/backups/raspberrypi/raspberrypi-rsync-backup-last_buster wird für Hardlinks benutzt.
11-09-2021 14:40:11 --- RBK0158I: rsync Backup "/media/nas/backups/raspberrypi/raspberrypi-rsync-backup-20211109-143844" wird erstellt.
11-09-2021 14:40:11 --- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
11-09-2021 16:54:54 --- RBK0078I: Backupzeit: 02:16:03.
11-09-2021 16:54:55 --- RBK0033I: Bitte warten bis aufgeräumt wurde.
11-09-2021 16:54:55 --- RBK0007I: Services werden gestartet: ' '.
11-09-2021 16:54:55 --- RBK0218I: Wende smarte Backupstrategie an. Täglich:7 Wöchentlich:4 Monatlich:4 Jährlich:0
11-09-2021 16:54:58 --- RBK0217I: 1 Backups werden smart recycled. 14 Backups werden aufgehoben. Bitte Geduld.
11-09-2021 16:54:58 --- RBK0220I: Smart Backup Strategie löscht Backup /media/nas/backups/raspberrypi/raspberrypi-rsync-backup-last_buster.
11-09-2021 17:21:14 --- RBK0260I: Dynamischer umount von /media/nas wird vorgenommen.
11-09-2021 17:21:14 --- RBK0017I: Backup erfolgreich beendet.
11-09-2021 17:21:14 --- RBK0010I: raspberrypi: raspiBackup.sh V0.6.6.1 (6c10535) Di 9. Nov 17:21:14 CET 2021 beendet mit Returncode 0.
11-09-2021 17:21:14 --- RBK0167I: Eine eMail wird versendet.`

`pi@raspberrypi:~ $ cat /tmp/raspiBackup.nohup
umount.nfs4: /media/nas: device is busy`

DrDeath, 2021-11-09:
/etc/fstab

`192.168.2.10:/volume1/raspberry  /media/nas/     nfs     defaults,user,rsize=32768,wsize=32768,tcp,rw,nfsvers=4 0  0`

framps, 2021-11-09:
That's not the debug log. It's just the messages :wink: . But I don't need the debug log any more. Looks like you have some files open on /media/nas and therefore umount fails. 

Please check with `lsof /media/nas` which files are still used.

DrDeath, 2021-11-10:
just checked with : `sudo lsof /media/nas`   ... empty output.
mountpoint `/media/nasl` still mounted.

I'll insert `lsof /media/nas > /tmp/lsof_raspiBackup.txt`  before `logItem "Umount of $DYNAMIC_MOUNT scheduled"` and wait for next backup tomorrow.

framps, 2021-11-10:
Great idea :+1:  Another root cause may be there is still something temporarily used and it takes some time until the file is no longer used. Maybe the force option `-f` may be useful or the umount should be delayed for some seconds. Let's wait for tomorrow and check the output of lsof.

DrDeath, 2021-11-11:
Hm, no output in `/tmp/lsof_raspiBackup.txt `  ( file was **not** created and /media/nas still mounted )

The Codeblock looks like this:

```
        if [[ -n "$DYNAMIC_MOUNT" ]] && (( $DYNAMIC_MOUNT_EXECUTED )); then
                lsof /media/nas > /tmp/lsof_raspiBackup.txt
                logItem "Umount of $DYNAMIC_MOUNT scheduled"
                nohup umount $DYNAMIC_MOUNT &> /tmp/raspiBackup.nohup </dev/null &
        fi
```

I'll umount /media/nas manually and wait for next run...

framps, 2021-11-11:
I just tested the dynamic umount in my environment and it works perfect. My /etc/fstab entry:

```
192.168.0.12:/disks/silver /backup nfs rw,noauto,acl 0 0
```

So I'm unable to reproduce your issue and we have to debug on your system - unfortunately :cry: 

Line 3765 is only cleanup for initial startup issues. For your issue it's the code on 3881 which kicks of the umount at the end of the backup run and fails.

DrDeath, 2021-11-12:
next run: same behavior.
`/tmp/lsof_raspiBackup.txt` was written, but empty
`raspiBackup.nohup` contains `umount.nfs4: /media/nas: device is busy`
Maybe the path pointer stays in /media/nas/ and prevent the umount.
I'll add a `cd /home` before....

framps, 2021-11-12:
Maybe it's somehow related to path /media/nas. That's the place used to mount any connected external devices. I use a dedicated backup path /backup all the time. Maybe it's also related to the fact you don't use `noauto`. Just guessing ...

DrDeath, 2021-11-12:
nice hint.
Reconfigured `raspiBackup.sh` / `raspiBackup.conf` and `/etc/fstab` to use the new mount point `/diskstation`
waiting for next run....

DrDeath, 2021-11-13:
Ok, looks like this was root cause.

The `/media/xxx` seems reacting slower for umount commands.
Since changing the mount point to `/diskstation`, the error went away. 

Thanks for pointing me in the right direction.

framps, 2021-11-13:
Great it works now for you. Your issue was very helpful because I'm sure there will be future raspiBackup users who also use `/media` as the backup path and will face hte same issue.

DrDeath, 2021-11-18:
Ok, additional hint:
After testing a few days, it happens again two times... umount was not done (even on `/diskstation`).

My personal workaround is to add the parameter `-l`  ( `umount --lazy` ) to the umount command in line 3811:
`nohup umount -l $DYNAMIC_MOUNT &> /tmp/raspiBackup.nohup </dev/null &`

framps, 2021-11-18:
Good catch :+1: I didn't know about this umount feature.

But then as far as I can see the code can be changed to

`umount -l $DYNAMIC_MOUNT` &>>$LOG_FILE and moved after `logFinsh()` call

which removed the ugly nohup call I frankly don't like.

framps, 2021-11-19:
I modified and tested the modified code https://github.com/framps/raspiBackup/blob/bf695b8862646c49210a698c8a2e56a3ab429093/raspiBackup.sh#L4177 and it works like a charm. Thank you very much your pointed me to umount option `-l`

The fix is not published right now but will be published latest with next release.

closed 2021-11-19 by framps