framps/raspiBackup #327: some errors when displaying help with --help

hugalafutro, 2021-04-03:
It's not an error in operation and as such there are no logs. I've recently been tinkering with raspiBackup config on one of my pi's and while running `sudo raspiBackup.sh --help` I noticed the help output also prints out some errors (I added arrows as either I'm blind or code authoring service like github doesn't support highlighting):
```
raspiBackup.sh 0.6.6, 2020-12-16/09:58:36 - 7989828
Usage: raspiBackup.sh [option]* {backupDirectory}

-General options-
-b {dd block size} (default: )
-D "{additional dd parameters}" (default: no)
-e {email address} (default: no)
-E "{additional email call parameters}" (default: no)
-f {config filename}
-g Display progress bar
-G {message language} (EN or DE) (default: )
-h display this help text
/usr/local/bin/raspiBackup.sh: line 7214: LOG_LEVELs: bad array subscript     <----------------------------------------------
-l {log level} ( Debug | Off) (default: )
/usr/local/bin/raspiBackup.sh: line 7215: MSG_LEVELs: bad array subscript    <----------------------------------------------
-m {message level} ( Verbose | Detailed | Minimal) (default: )
-M {backup description}
-n notification if there is a newer scriptversion available for download (default: no)
-s {email program to use} (mail,ssmtp,msmtp,sendEmail,mailext) (default: )
--timestamps Prefix messages with timestamps (default: no)
-u "{excludeList}" List of directories to exclude from tar and rsync backup
-U current script version will be replaced by the most recent version. Current version will be saved and can be restored with parameter -V
-v verbose output of backup tools (default: no)
-V restore a previous version
-z compress backup file with gzip (default: no)

-Backup options-
-a "{commands to execute after Backup}" (default: )
-B Save bootpartition in tar file (Default: )
-F Backup is simulated
-k {backupsToKeep} (default: )
-o "{commands to execute before Backup}" (default: no)
-t {backupType} (dd|rsync|tar) (default: )
-T "{List of partitions to save}" (Partition numbers, e.g. "1 2 3"). Only valid with parameter -P (default: )

-Restore options-
-0 SD card will not be formatted
-1 Formatting errors on SD card will be ignored
-C Formating of the restorepartitions will check for badblocks (Standard: )
-d {restoreDevice} (default: no) (Example: /dev/sda)
-R {rootPartition} (default: restoreDevice) (Example: /dev/sdb1)
--resizeRootFS (Default: no)
```

framps, 2021-04-03:
That's the source code which creates the error messages:

```
	echo "-l {log level} ($POSSIBLE_LOG_LEVELs) (default: ${LOG_LEVELs[$DEFAULT_LOG_LEVEL]})"
	echo "-m {message level} ($POSSIBLE_MSG_LEVELs) (default: ${MSG_LEVELs[$DEFAULT_MSG_LEVEL]})"
```

Looks like `DEFAULT_LOG_LEVEL` and `DEFAULT_MSG_LEVEL` are incorrect in the config file. The first one has to be either 0 or 1 or Off or Debug and the second one has to be either 0 or 1 or Minimal or Detailed.

Please check /usr/local/etc/raspiBackup.conf.

hugalafutro, 2021-04-03:
I've got a Zero, Zero-W, 3x Pi3 running raspbian buster and a rpi4 running ubuntu. I've ran `sudo cat /usr/local/etc/raspiBackup.conf | grep _LEVEL` on every one of them which returns only `DEFAULT_MSG_LEVEL="0"` indicating `DEFAULT_LOG_LEVEL` is not present at all.

framps, 2021-04-03:
I found the root cause of the issue: It's already fixed in latest 0.6.6 version :wink: .

Please update your local version of 0.6.6 with `sudo raspiBackup.sh -U -S`. You then should have 
`raspiBackup         : Version: 0.6.6      Date: 2021-02-26 16:16:38  Sha: 83ecd03`

Please verify the fix and close the issue.

hugalafutro, 2021-04-03:
Fixed, thanks!

log output:
```
pi@pi-hole:~ $ sudo raspiBackup.sh -U -S
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
--- RBK0195I: Update raspiBackup.sh to version 0.6.6.
--- RBK0038I: Are you sure? y/N
y
--- RBK0057I: Downloading file raspiBackup.sh from https://www.linux-tips-and-tricks.de.
--- RBK0072I: /usr/local/bin/raspiBackup.sh updated from version 0.6.6 to version 0.6.6. Previous version saved as /usr/local/bin/raspiBackup.0.6.6.sh. Don't forget to test backup and restore with the new version now.
pi@pi-hole:~ $ sudo raspiBackup.sh --help
raspiBackup.sh 0.6.6, 2021-02-26/16:16:38 - 83ecd03
Usage: raspiBackup.sh [option]* {backupDirectory}

-General options-
-b {dd block size} (default: 1MB)
-D "{additional dd parameters}" (default: no)
-e {email address} (default: no)
-E "{additional email call parameters}" (default: no)
-f {config filename}
-g Display progress bar
-G {message language} (EN or DE) (default: )
-h display this help text
-l {log level} ( Debug | Off) (default: Debug)
-m {message level} ( Verbose | Detailed | Minimal) (default: Minimal)
-M {backup description}
-n notification if there is a newer scriptversion available for download (default: yes)
-s {email program to use} (mail,ssmtp,msmtp,sendEmail,mailext) (default: mail)
--timestamps Prefix messages with timestamps (default: no)
-u "{excludeList}" List of directories to exclude from tar and rsync backup
-U current script version will be replaced by the most recent version. Current version will be saved and can be restored with parameter -V
-v verbose output of backup tools (default: no)
-V restore a previous version
-z compress backup file with gzip (default: no)

-Backup options-
-a "{commands to execute after Backup}" (default: )
-B Save bootpartition in tar file (Default: 0)
-F Backup is simulated
-k {backupsToKeep} (default: 3)
-o "{commands to execute before Backup}" (default: no)
-t {backupType} (dd|rsync|tar) (default: dd)
-T "{List of partitions to save}" (Partition numbers, e.g. "1 2 3"). Only valid with parameter -P (default: 1 2)

-Restore options-
-0 SD card will not be formatted
-1 Formatting errors on SD card will be ignored
-C Formating of the restorepartitions will check for badblocks (Standard: 0)
-d {restoreDevice} (default: no) (Example: /dev/sda)
-R {rootPartition} (default: restoreDevice) (Example: /dev/sdb1)
--resizeRootFS (Default: yes)
```

closed 2021-04-03 by hugalafutro