framps/raspiBackup #328: Synology NAS Question Regarding raspiBackup Files

SJMarty, 2021-04-07:
I have five Raspberry Pis and use raspiBackup on all of them.  I back them up to an NFS share on NAS 1 (Synology DS-1517+).  I use HyperBackup (native to Synology) to backup all of the shares on NAS 1 to NAS 2 (a second DS-1517+) every night.  The issue I'm seeing in that in some of the raspiBackups from some of the Pis, this shows up in the Synology HyperBackup logs...

```
Information | 4/6/2021 5:26 | SYSTEM | [Network][NFS Share] Backup task finished successfully. [2238939 files scanned] [223509 new files] [2015430 files unchanged]
-- | -- | -- | --
Warning | 4/6/2021 5:17 | SYSTEM | [Network][NFS Share] Incomplete file backup. (The file is a special file.)   [/volume1/NFS/[HOST-NAME]/[HOST-NAME]-rsync-backup-20210406-011241/root/.gnupg/S.gpg-agent.extra]
Warning | 4/6/2021 5:17 | SYSTEM | [Network][NFS Share] Incomplete file backup. (The file is a special file.)   [/volume1/NFS/[HOST-NAME]/[HOST-NAME]-rsync-backup-20210406-011241/root/.gnupg/S.gpg-agent.browser]
Warning | 4/6/2021 5:17 | SYSTEM | [Network][NFS Share] Incomplete file backup. (The file is a special file.)   [/volume1/NFS/[HOST-NAME]/[HOST-NAME]-rsync-backup-20210406-011241/root/.gnupg/S.gpg-agent]
Warning | 4/6/2021 5:17 | SYSTEM | [Network][NFS Share] Incomplete file backup. (The file is a special file.)   [/volume1/NFS/[HOST-NAME]/[HOST-NAME]-rsync-backup-20210406-011241/root/.gnupg/S.gpg-agent.ssh]
Information | 4/6/2021 5:00 | SYSTEM | [Network][NFS Share] Backup task started.
```

As I mentioned, when these warnings are logged, it's not every backup nor every Pi.  In all cases, however, it always references files in the /root/.gnupg/ folder.  I've done my NAS backups this way for over four years and have never seen these warnings before.  raspiBackup itself logs no errors during the backup and reports the backup completed successfully.

At this point, I'm not clear if the issue is with my raspiBackup configuration or with my Synology NAS backups.  Are there additional command line options or something else I should be doing differently with raspiBackup's configuration?  I also don't understand why the NAS backup process complains about these files on some raspiBackups but not others/all of them.

Looking forward to suggestions/guidance.

Thanks.

framps, 2021-04-07:
> At this point, I'm not clear if the issue is with my raspiBackup configuration or with my Synology NAS backups.

raspiBackup uses rsync to create the backups. If there are no errors reported from rsync the backup should be OK. I think it's some issue with hyperbackup.

> It's not every backup nor every Pi

That's interesting. `nor every pi`  might be because you don't have theses files on every pi?
`not every backup` Do you sometimes get the error messages for the same pi and sometimes not?

What I would do is 
1) Check whether the files which hyperbackup complaints about are identical on pi and the rsync backup. 

2) Next I would check whether they are identical with the files included in hyperbackup. According the message it sounds to me they will not be identical. 

If the files in (1) are identical and (2) are different this proves it's an issue with hyperbackup. There are some hits for this message in the net but I didn't find any good explanation for the message. Maybe you should open a ticket in the Synology Support Center.

SJMarty, 2021-04-11:
Correct on the files...they are not on every Pi.  However, the files that HyperBackup reports warnings on, it does so for every backup HyperBackup performs of those files.  In other words, once the problematic files are present, HyperBackup has trouble with them every time it tries to back them up.

I compared the source files on the Pis with the files placed on the NFS share on the NAS via raspiBackup and they appear to match.  I checked the HyperBackup destination and the files that it warns about are not being backed up.

Here is another file HyperBackup had trouble with...

```
Warning | 4/11/2021 5:06 | SYSTEM | [Network][NFS Share] Incomplete file backup. (The file is a special file.) [/volume1/NFS/[HOST-NAME]/[HOST-NAME]-rsync-backup-20210407-171043/root/.cache/pcmanfm-socket--1]
```

Here is a directory listing for that file on the Pi...

`srwxr-xr-x  1 root root    0 Mar 31 16:43 pcmanfm-socket--1`

The 's' file type indicates socket.  Would have probably guessed that from the name.  :)

I went ahead and opened a case with Synology.  They came back with a suggestion that the file may be locked by the Pi and to disconnect the Pi from the share and rerun the backup.  I didn't expect that to be the answer and think they might just be guessing based on the file name but I tried it anyway and the same warnings logged.  Waiting for their next suggestion.

SJMarty, 2021-04-12:
I'm not getting very far with Synology support.  They want remote access to my NAS devices to "run some tests" (their words), manually run some backups, and who knows what else.  They haven't actually asked me for any information on the files that are producing the warning messages.  I don't see any evidence that this issue is specific to my NAS so I pushed back on their remote access requests so we'll see where it goes.

I also posted in their forums but no responses to my thread.

framps, 2021-04-12:
Thank you very much for the update. 

I think you should continue on the Synology path ... but maybe we should find some mitigation path in parallel :wink: 

Is there any pattern you can identify from the files which cannot be saved by Hyperbackup? You mentioned `/root/.gnupg/S.gpg-agent.*` which sounds so be some special files because of security/privacy. If we can identify the files which cause the issue and identify a pattern we can decide whether they are important for the Hyperbackup or just ignore the files in raspiBackup and create some exclude rules in raspiBackup.

SJMarty, 2021-04-26:
The initial problem files were on several Pis and located at `/root/.gnupg/S.gpg-agent.*` but after several backups, those files aged off.  The files are no longer on the Pis and no longer on the NFS share that Hyper Backup backs up.  So at this point, all I have left that is giving Hyper Backup issues is a single file on one Pi `/root/.cache/pcmanfm-socket--1` and as I mentioned before, this is a socket file.  After two weeks of back and forth with Synology support (and multiple requests to have remote access to my NAS), here is what they came back with...

> Hello,
> 
> The file is unable to be backed up as it is a limitation in backup software regarding socket files. The socket file itself isn't technically a viewable/modifiable file. It is used to communicate between different applications. The task seems to finish successfully but just skips the files as it cannot back them up. As far as I'm able to determine the backup is working properly.
> 
> Let me know if you have any questions.
> 
> Regards,
> Matt

I don't recall if the `/root/.gnupg/S.gpg-agent.*` files were socket files but since they are gone now, even if they weren't, I can only assume they were transient.  As far as the actual socket file `/root/.cache/pcmanfm-socket--1`, it has persisted so based on the location (a .cache directory), I deleted the file with no apparent or immediate ill effects.  The file was originally created on 03/31/21 and has not been recreated since it was deleted.  I ran a manual backup and without that file present, Hyper Backup was able to back up the resulting raspiBackup with no problems.  This particular Pi backs up every Monday morning so it backed up this morning.  I retain three backups so I expect the last one to age off in another week.

Hopefully, this will help someone else at some point.

I'll continue checking the Hyper Backup logs to see if anything else shows up in there indicating a special file that can't be backed up.  Is there anything else you'd like me to ask Synology before they close my ticket?

framps, 2021-04-27:
> Hyper Backup issues is a single file on one Pi `/root/.cache/pcmanfm-socket--1` and as I mentioned before, this is a socket file.

As far as I found out in the net this is a socket created by PCMan Filemanager which should be deleted when PCMan is closed. Maybe PCMan crashed and the file was not deleted.

>  The file was originally created on 03/31/21 and has not been recreated since it was deleted.

Did you start PCMan as root again when you deleted the file? 

> Hopefully, this will help someone else at some point.

I appreciate you share your experiences you make with socket files and Synology HyperBackup. 

> The file is unable to be backed up as it is a limitation in backup software regarding socket files.

That's an important statement from Synology and explains the issue. 

As far as I can see socket files are created by applications (e.g. gpg and PCMan) which are deleted when the application is stopped. Because you create a backup of a running system these sockets exist and either you ignore the error because a new socket will be created when the backup is started, you stop the applications before you create the backup or you add the socket files in the exclude list of raspiBackup.

>  Is there anything else you'd like me to ask Synology before they close my ticket?

We now know HyperBackup has an issue with socket files and that's the root cause for your issue. I don't think they will fix this issue and there is no further question from my side.

> I'll continue checking the Hyper Backup logs to see if anything else shows up in there indicating a special file that can't be backed up.

Any open socket on the raspberry will create the issue again. As long as you don't use gpg an PCMan as root you shouldn't get the issue again.

SJMarty, 2021-05-01:
I did ask one additional question of Synology.  I asked "Are there any other file types that Hyper Backup is unable to back up?" and this is the response I received...

> Hello,
> 
> As far as I'm aware those are the only files. This is the first time I've experienced this error myself, and I talked to people that know more about Hyper backup. Regular files, folders, and symlinks are able to be backed up by Hyper Backup. Socket or "special" files are not. Those are the only files I am aware of that are unable to be backed up.
> 
> Regards,
> Matt

SJMarty, 2021-05-01:
Well, I just checked my Hyper Backup logs and the files from /root/.gnupg showed up again on two of my Pis.  I checked and all four of the files are socket files.  I also verified that they disappear after a reboot of the Pis.

I have no idea why these socket files showed up.  The Pis back up once a week and the previous two backups did not have the files present.

What are your thoughts on just excluding these files from the backups?  I'm not confident this is the best route because I don't know why the files are being created but at this point, I'm not sure how else to handle them.

These are the four files (same files on both Pis)...

```
/root/.gnupg/S.gpg-agent.ssh
/root/.gnupg/S.gpg-agent.extra
/root/.gnupg/S.gpg-agent.browser
/root/.gnupg/S.gpg-agent
```

Thanks.

framps, 2021-05-04:
Looks like somebody used [GPG](https://en.wikipedia.org/wiki/GNU_Privacy_Guard) as root on your system. As far as I understand this is used sometimes when you update SW on your system as root. 

> What are your thoughts on just excluding these files from the backups?

I don't know whether it creates any other issues if these files are missing in the backup. But as far as I understand you didn't get any issues when you deleted these files. Given this I think it's safe to exclude the files from the backup and eliminate the warning message.

SJMarty, 2021-05-09:
Can you clarify the syntax I should use to add exclusions to the raspiBackup.conf file?  From reading the help pages, I gather I should be using the DEFAULT_EXCLUDE_LIST parameter.  I'm a bit confused on what I should be placing in the quotes.

I want to exclude these four files...
```
/root/.gnupg/S.gpg-agent.ssh
/root/.gnupg/S.gpg-agent.extra
/root/.gnupg/S.gpg-agent.browser
/root/.gnupg/S.gpg-agent
```
Is one of these is correct...?

`DEFAULT_EXCLUDE_LIST="--exclude=/root/.gnupg/S.gpg-agent*"`

`DEFAULT_EXCLUDE_LIST="/root/.gnupg/S.gpg-agent*"`

framps, 2021-05-11:
You have to add the exclude options in the syntax rsync requires and thus also [include `--exclude=`](https://www.linux-tips-and-tricks.de/en/backup/#parm_u). In your case 

`DEFAULT_EXCLUDE_LIST="--exclude=/root/.gnupg/"`

will exclude the whole gnupg subdirectory. 

If you use

`DEFAULT_EXCLUDE_LIST="--exclude=/root/.gnupg/*"`

the directory itself will be kept. 

`DEFAULT_EXCLUDE_LIST="--exclude=/root/.gnupg/S.gpg-agent*"`

will also work as far as I can see. But given the fact the rsync exclude semantic is a bit tricky I'm not 100% sure. Please test whether the statement you use works as expected.

In order to speed up the test process use option `-v` and

`DEFAULT_RSYNC_BACKUP_ADDITIONAL_OPTIONS="-n"`

to enable the dry run of rsync and check in the log whether the files you want to exclude are indeed excluded :wink:

SJMarty, 2021-05-13:
This worked perfectly...

`DEFAULT_EXCLUDE_LIST="--exclude=/root/.gnupg/S.gpg-agent*"`

Thanks for all of your help!

closed 2021-05-13 by SJMarty