framps/raspiBackup #329: RaspiBackup cannot repartition to slightly smaller partition on rsync restore?!

RURon, 2021-04-11:
(I could not find information where a restore log is created (the one in / did not contain this program run and neither the one in the backup-dir)

Today I had the situation where the rsync backup of the root partition (sdcard) was a little larger than the target (usb-ssd). So raspiBackup tried to shrink the target root. But this failed:

`pi@raspberrypi4:~ $ sudo raspiBackup.sh -d /dev/sdb /mnt/ssd/backups/raspberrypi4-Extra_Backup/raspberrypi4-rsync-backup-20210411-125936/
--- RBK0009I: raspberrypi4: raspiBackup.sh V0.6.6 (928df55) So 11. Apr 14:58:51 CEST 2021 gestartet.
!!! RBK0006W: Ziel /dev/sdb mit 29.81 GiB ist kleiner als die Backupquelle mit 29.82 GiB. Die root Partition wird entsprechend verkleinert. HINWEIS: Der Restore kann fehlschlagen wenn sie zu klein wird.
!!! RBK0065W: Gerät /dev/sdb wird repartitioniert und die gesamten Daten werden gelöscht.
--- RBK0067I: Momentane Partitionen auf /dev/sdb:
Number  Start   End      Size     Type     File system  Flags
 1      4,19MB  273MB    268MB    primary  fat32        lba
 2      273MB   32017MB  31744MB  primary  ext4
--- RBK0066I: Gerät /dev/sdb wird überschrieben mit der gesicherten Boot- und Rootpartition.
--- RBK0069I: Bootpartition /dev/sdb1 wird formatiert und erhält die zurückgespielte Bootpartition.
--- RBK0070I: Rootpartition /dev/sdb2 wird formatiert und erhält die zurückgespielte Rootpartition.
--- RBK0038I: Bist Du sicher? j/N
j
--- RBK0050I: Backup wird von /mnt/ssd/backups/raspberrypi4-Extra_Backup/raspberrypi4-rsync-backup-20210411-125936 zurückgespielt.
--- RBK0004I: Zweite Partition wird von 29.57 GiB auf 29.56 GiB angepasst.
--- RBK0053I: Erste Partition (Bootpartition) wird auf /dev/sdb1 zurückgespielt.
??? RBK0185E: Rootpartitionserstellung endet fehlerhaft mit RC 1.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0010I: raspberrypi4: raspiBackup.sh V0.6.6 (928df55) So 11. Apr 14:59:32 CEST 2021 beendet mit Returncode 117.
??? RBK0077E: Restore wurde fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.`

Unfortunately the error code is not yet explained/documented to I dont really know what to do. What I now did try is to specify -0 option (currently restoring...)

Other than that, I greatly thank you for this piece of software! Really saved my ass a couple of times already!

RURon, 2021-04-11:
(hm, linefeeds seem to be my problem right now)

RURon, 2021-04-11:
`pi@raspberrypi4:~ $ sudo raspiBackup.sh -d /dev/sdb /mnt/ssd/backups/raspberrypi4-Extra_Backup/raspberrypi4-rsync-backup-20210411-125936/
--- RBK0009I: raspberrypi4: raspiBackup.sh V0.6.6 (928df55) So 11. Apr 14:58:51 CEST 2021 gestartet.
!!! RBK0006W: Ziel /dev/sdb mit 29.81 GiB ist kleiner als die Backupquelle mit 29.82 GiB. Die root Partition wird entsprechend verkleinert. HINWEIS: Der Restore kann fehlschlagen wenn sie zu klein wird.
!!! RBK0065W: Gerät /dev/sdb wird repartitioniert und die gesamten Daten werden gelöscht.
--- RBK0067I: Momentane Partitionen auf /dev/sdb:
Number  Start   End      Size     Type     File system  Flags
 1      4,19MB  273MB    268MB    primary  fat32        lba
 2      273MB   32017MB  31744MB  primary  ext4
--- RBK0066I: Gerät /dev/sdb wird überschrieben mit der gesicherten Boot- und Rootpartition.
--- RBK0069I: Bootpartition /dev/sdb1 wird formatiert und erhält die zurückgespielte Bootpartition.
--- RBK0070I: Rootpartition /dev/sdb2 wird formatiert und erhält die zurückgespielte Rootpartition.
--- RBK0038I: Bist Du sicher? j/N
j
--- RBK0050I: Backup wird von /mnt/ssd/backups/raspberrypi4-Extra_Backup/raspberrypi4-rsync-backup-20210411-125936 zurückgespielt.
--- RBK0004I: Zweite Partition wird von 29.57 GiB auf 29.56 GiB angepasst.
--- RBK0053I: Erste Partition (Bootpartition) wird auf /dev/sdb1 zurückgespielt.
??? RBK0185E: Rootpartitionserstellung endet fehlerhaft mit RC 1.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0010I: raspberrypi4: raspiBackup.sh V0.6.6 (928df55) So 11. Apr 14:59:32 CEST 2021 beendet mit Returncode 117.
??? RBK0077E: Restore wurde fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
`

framps, 2021-04-11:
> I dont really know what to do

Please check the debug log for additional error messages. For restore the debug log is created in the directory you are in when you kick off the restore.

Whether option `-0` helps depends on the issue you face. RC1 

Please attach the debug log for further analysis.

RURon, 2021-04-11:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/6292365/raspiBackup.log)

The error appears to be a mount problem - though when looking at lsblk:

`pi@raspberrypi4:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 223,6G  0 disk
└─sda1        8:1    0 223,6G  0 part /mnt/ssd
sdb           8:16   0  29,8G  0 disk
├─sdb1        8:17   0   256M  0 part
└─sdb2        8:18   0  29,6G  0 part
sdc           8:32   1   120M  0 disk
└─sdc1        8:33   1 119,3M  0 part /media/usb3
mmcblk0     179:0    0  29,8G  0 disk
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0  29,6G  0 part /
zram0       254:0    0   750M  0 disk [SWAP]
zram1       254:1    0   150M  0 disk /opt/zram/zram1`

There is no mountpoint. I also tried sudo "umount -f /dev/sdb2" - hmmm.
(btw, my attempt to use -0 option ended with "RBK0183E: Rootpartitionscheck endet fehlerhaft mit RC 8." - will now check if it still boots successfully)

RURon, 2021-04-11:
(it's booting just fine *yay!*)

framps, 2021-04-11:
> it's booting just fine _yay!_)

Congrats :smile: . Would you please let me know what you did in order to fix the issue? Just in case somebody else using raspiBackup will face the same issue. Actually it's strange /dev/sdb2 is already mounted.

RURon, 2021-04-11:
sure, I only ignored the error RBK0183E - I dont know what is checked at that step and what exactly failed, but system looks good as far as I can tell. If you think it could help, I could attach that log as well (which is now on the sd card from which I started the restore (and is now ejected from my headless Pi), but I could get the log tomorrow)

framps, 2021-04-12:
Ok. I see. That's interesting. There is some information in the debug log

```
20210411-192545 DBG 6428:              *** blkid
20210411-192547 DBG 6428:                  /dev/mmcblk0p1: LABEL_FATBOOT="M2.Boot" LABEL="M2.Boot" UUID="4BBD-D3E7" TYPE="vfat" PARTUUID="6f22a926-01"
20210411-192547 DBG 6428:                  /dev/mmcblk0p2: UUID="4736d476-478a-433d-9408-aa8c1a251ae6" TYPE="ext4" PARTUUID="6f22a926-02"
20210411-192547 DBG 6428:                  /dev/sda1: LABEL="Crucial240GB" UUID="d26d8ee8-5b92-4a81-91ae-ddc4874dcf3d" TYPE="ext4" PARTUUID="41931ec9-01"
20210411-192547 DBG 6428:                  /dev/zram0: LABEL="zram-config0" UUID="0890c810-a04e-4093-8c62-3e22e6c40fda" TYPE="swap"
20210411-192547 DBG 6428:                  /dev/zram1: UUID="96082339-f44b-448c-989b-b9200320a989" TYPE="ext4"
20210411-192547 DBG 6428:                  /dev/sdb1: LABEL_FATBOOT="M2.Boot" LABEL="M2.Boot" UUID="4BBD-D3E7" TYPE="vfat" PARTUUID="6f22a926-01"
20210411-192547 DBG 6428:                  /dev/sdb2: UUID="38b8373a-ca5d-4139-a59b-e5719cc868c7" TYPE="ext4" PARTUUID="6f22a926-02"
20210411-192547 DBG 6428:                  /dev/mmcblk0: PTUUID="6f22a926" PTTYPE="dos"
20210411-192547 DBG 6428:                  /dev/sdc1: SEC_TYPE="msdos" UUID="A87B-A154" TYPE="vfat" PARTUUID="9d11ef09-01"
```

which may point to the root cause of the issue: 
You have /dev/mmcblk0 and /dev/sdb with the same UUID and PARTUUID. Looks like you already did some restore on /dev/sdb before and the UUIDs and PARTUUIDs were cloned. I already had an issue when somebody also had the same UUID and PARTUUID on two devices and as far as I remember there was also a mount/umount issue. I tried to find the issue in this repo to refresh my memory but failed :cry: 

Maybe you want to try again but clear the partition table on /dev/sdb first with
`dd if=/dev/zero of=/dev/sdb bs=512 count=1  `?

WARNING: Make sure you clear the correct disk!

RURon, 2021-04-12:
Ah, interesting, indeed :-) As the system is already running again (with some changes already, so that I would need to recreate those), I would skip this test but next time I need a restore on these disks, I will hopefully remember the bookmark I will now create ;-) Thanks for looking at the log!

RURon, 2021-04-12:
(maybe you could let RaspiBackup state a warning if it finds same UUIDs on source and destination?)

closed 2021-04-12 by RURon

framps, 2021-04-12:
> state a warning if it finds same UUIDs

I added some code to check for this but removed it because it created irritations for raspiBackup users because of false positives. As a leftover there is an option `--updateUUIDs` which makes sure a new UUID and PARTUUID is generated during restore and thus you don't get this ugly issue.