framps/raspiBackup #330: Restore Problem mit raspiBackup

gerd-1, 2021-04-13:
Hallo Framp´s
ich habe ein raspiBackup auf einem 32 GB USB Stick und möchte das Backup testweise auf eine 32GB SD-Karte zurücksichern.
Dabei endet das restore mit folgenden Meldungen, die mir aber leider nciht weiterhelfen.

` $ sudo raspiBackup.sh -d /dev/sda /media/pi/RaspiBackup01/raspberrypi/raspberrypi-rsync-backup-20210413-171547/
--- RBK0009I: raspberrypi: raspiBackup.sh V0.6.6 (83ecd03) Di Apr 13 21:36:42 CEST 2021 gestartet.
!!! RBK0006W: Ziel /dev/sda mit 29.12 GiB ist kleiner als die Backupquelle mit 29.72 GiB. Die root Partition wird entsprechend verkleinert. HINWEIS: Der Restore kann fehlschlagen wenn sie zu klein wird.
!!! RBK0065W: Gerät /dev/sda wird repartitioniert und die gesamten Daten werden gelöscht.
Warning: Unable to open /dev/sda read-write (Das Dateisystem ist nur lesbar).  /dev/sda has been opened read-only.
Warning: Unable to open /dev/sda read-write (Das Dateisystem ist nur lesbar).  /dev/sda has been opened read-only.
Warning: Unable to open /dev/sda read-write (Das Dateisystem ist nur lesbar).  /dev/sda has been opened read-only.
--- RBK0067I: Momentane Partitionen auf /dev/sda:
Model: Generic USB SD Reader (scsi)                                       
Number  Start   End      Size     Type     File system  Flags
 1      4,19MB  31267MB  31263MB  primary  fat32        lba
--- RBK0066I: Gerät /dev/sda wird überschrieben mit der gesicherten Boot- und Rootpartition.
--- RBK0069I: Bootpartition /dev/sda1 wird formatiert und erhält die zurückgespielte Bootpartition.
--- RBK0070I: Rootpartition /dev/sda2 wird formatiert und erhält die zurückgespielte Rootpartition.
--- RBK0038I: Bist Du sicher? j/N 
j
--- RBK0050I: Backup wird von /media/pi/RaspiBackup01/raspberrypi/raspberrypi-rsync-backup-20210413-171547 zurückgespielt.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0010I: raspberrypi: raspiBackup.sh V0.6.6 (83ecd03) Di Apr 13 21:37:18 CEST 2021 beendet mit Returncode 112.
??? RBK0077E: Restore wurde fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
`
Bitte um deine Mithilfe.
Danke im Voraus!

framps, 2021-04-13:
`Warning: Unable to open /dev/sda read-write (Das Dateisystem ist nur lesbar).  /dev/sda has been opened read-only.`

Die Meldung sagt dass Du Deine SD Karte - ich nehme an die liegt auf /dev/sda - nur lesbar gemounted hast. Dann funktioniert natuerlich kein Restore :smiley: 

Hats Du vielleicht den Schreibschutz den SD Karte eingeschaltet?

gerd-1, 2021-04-13:
Der Schreibschutz ist nicht eingeschaltet.
Ich weiß aber leider nicht wie ich die Karte sonst beschreibbar machen kann.
Mir ist auch nicht klar warum sie schreibgeschüzt ist.
Es handelt sich dabei um eine ganz neue Karte die ich mittels Kartenleser an den Raspi gehängt und nach der autom. Rückfrage mit meinem Passwort gemountet habe.

framps, 2021-04-13:
> Ich weiß aber leider nicht wie ich die Karte sonst beschreibbar machen kann.
> Mir ist auch nicht klar warum sie schreibgeschüzt ist.

Ich sehe jetzt mehrere Moeglichkeiten fuer Dich Dein Problem zu loesen:

1) Du suchst im Netz wie man eine SD Karte unter Linux R/W mounted. 
2) Du fragst in einen Linux- oder Raspberryforum nach was Die Ursache sein kann. 

In [FAQ47](https://www.linux-tips-and-tricks.de/de/faq/#a47) findest Du dazu weitere Informationen :wink: 

Ich hoffe Du verstehst dass ich hier keine Hilfe zu Linuxfragen sondern nur zu raspiBackup gebe.

closed 2021-04-14 by gerd-1