framps/raspiBackup #331: WD My Cloud EX2 Ultra als NAS benutzen

Lutz1967, 2021-04-24:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Version: 0.6.6 CommitSHA: 83ecd03 CommitDate: 2021-02-26 CommitTime: 16:16:38

Hi,

ich habe eine WD My Cloud als backup - system für die Rechner in meinem Heimnetzwerk laufen und würde auch gern meinen raspi darauf sichern. Ich habe mich soweit durchgehangelt. Leider bekomme ich bei dem empfohlenen rsync Format den Fehler: 

RBK0266E: Es fehlt die Berechtigung um Linux Dateiattribute auf /nfs/Backup zu erstellen (Dateisystem: nfs).

DD funktioniert. Tar werde ich noch testen, gehe aber davon aus, dass es funktionieren wird.

Nachdem was ich gelesen habe, ist das Filesystem der WD My Cloud EX2 Ultra ext4, es sollte also funktionieren. Bei der NFS Freigabe gibt es nicht so viele Möglichkeiten, siehe auch hier: 

https://getmob.de/2021/04/16/mount-wd-my-cloud-ex2-nas-in-raspberry/  

Ich habe schon eine weile rumgesucht, kann aber zu diesem NAS nichts finden und auch eine Fehlerbeschreibung habe ich nicht gefunden. Aus den Release notes habe ich gefolgert, dass das evtl. so gewollt ist?

Danke!

framps, 2021-04-25:
Es gibt zwei Dinge die erfuellt sein muessen damit ein rsync Backup auf nfs funktioniert: 

1) ext3/4 Filesystem (hast Du wie ich denke)
2) Das Filesystem muss no_root_squash exportiert werden. Das wird vermutlich die Ursache sein. Ich koennte mir denken das die `oeffentlich` Einstellung bei Deiner NAS das bewirkt. Da musst Du ein wenig probieren. Genau kannst Du das pruefen wenn Du Dir die Datei /etc/exports ansiehst. Weiss nicht ob Du Systemzugriff auf die NAS hast.Bei mir sieht eine Freigabezeilefuer `/disks/silver` wie folgt aus:
```
/disks/silver           *(rw,sync,no_subtree_check,no_root_squash)
```

Lutz1967, 2021-04-25:
Hallo,

Erstmal vielen Dank für die schnelle Antwort. Bei mir sieht die /etc/exports so aus:

"/nfs/raspi" RASPI.IP(rw,all_squash,sync,no_wdelay,insecure_locks,insecure,no_subtree_check,anonuid=501,anongid=1000)

Also all_squash und nicht no_root_squash . Damit ist es klar, dass das nicht funktioniert. 
Da ich alles Andere als ein Linux - Experte ist werde ich da nicht dran rumpfuschen. Ich habe mal ein bisschen gegoogled und da scheinen sich Andere auch die Zähne dran auszubeissen. Also wenn es keine "normale" Konfiguration auf der My Cloud gibt, die das behebt, werde ich mal schön tar backups benutzen. Sicher ist sicher. 

Vielen Dank für den tollen Job mit dem script.

framps, 2021-04-25:
> Da ich alles Andere als ein Linux - Experte ist werde ich da nicht dran rumpfuschen. Ich habe mal ein bisschen gegoogled und da scheinen sich Andere auch die Zähne dran auszubeissen.

Eigentlich ist das recht einfach und mit no_root_squash funktioniert es zuverlaessig. Deshalb verstehe ich nicht ganz wo Du gefunden hast dass sich andere die Zaehne daran ausgebissen haben. Aber natuerlich kannst Du auch tar nutzen. Dann erstellst Du immer einen Vollbackup der natuerlich auch laenger dauert. Auch ist das partielle Rauspicken von Teilbackups nicht ganz so komfortabel. Aber wichtig ist dass Du ueberhaupt ein Backup hast :+1: 

Du kennst ja den Spruch: Kein Backup - kein Mitleid :wink: 

Bitte den Issue schliessen wenn Du keine weitere Frage mehr hast.

closed 2021-04-25 by Lutz1967