framps/raspiBackup #332: RBK0208W Es konnte keine UUID in /boot/cmdline.txt für root erneuert werden

GregorSchirmer, 2021-04-27:
Es wurde ein komplett neues Raspberry OS aufgespielt (2021-03-04-raspios-buster-armhf-lite). raspiBackup wurde nicht für Backups eingerichtet, sondern nach der Installation direkt ein Restore auf der gleichen Raspberry OS Version versucht.
Es wurde ein tgz mit dem Pi auf eine baugleiche microSD Karte ohne externer Partition restored.
Der Restore wurde bei allen Versuchen mit dem Fehler RBK0208W "erfolgreich" abgeschlossen.
Auch mit Option --updateUUIDs, -0 und -1 ist der Fehler "RBK0208W Es konnte keine UUID in /boot/cmdline.txt für root erneuert werden" aufgetreten. Wie im Fehler, beschrieben lässt sich die restorte SD-Karte im Pi nicht booten.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/6387046/raspiBackup.log)

framps, 2021-04-27:
Es zeigt sich leider dass ein paar wichtige Informationen nicht im Debuglog stehen :cry: 

Aber was ich jedenfalls schon mal gesehen habe ist dass /boot/cmdline.txt nicht updated wurde mit der neuen PARTUUID.

in /etc/fstab des restorten Images steht
```
20210427-183256 DBG 7084:              --- Upd /tmp/root/etc/fstab
20210427-183256 DBG 7085:                  *** cat /tmp/root/etc/fstab
20210427-183256 DBG 7085:                      proc                                         /proc           proc    defaults          0       0
20210427-183256 DBG 7085:                      PARTUUID=b36b97b9-01                         /boot           vfat    defaults          0       2
20210427-183256 DBG 7085:                      PARTUUID=b36b97b9-02                         /               ext4    defaults,noatime  0       1
20210427-183256 DBG 7085:                      UUID=7992dc3d-f399-4cc1-9d27-dd502530c45f    /media/backup   ext4    defaults          0       0
20210427-183256 DBG 7085:                      # a swapfile is not a swap partition, no line here
20210427-183256 DBG 7085:                      #   use  dphys-swapfile swap[on|off]  for that
```

d.h. die UUID der Rootpartition der SD Karte ist `PARTUUID=b36b97b9-02`. Du musst in /boot/cmdline.txt (deren Inhalt ich leider nicht im Debuglog habe) so updaten dass dort die PARTUUID wie sie in der /etc/fstab definiert ist steht. Dann sollte das restorte System booten. 

Um rauszufinden warum die /boot/cmdline.txt nicht gefunden wird muss ich einen Patch fuer Dich bauen und mehr debug Statements einfuegen.

G-Kraxen, 2021-04-28:
Die UUIDs in /boot/cmdline.txt und /etc/fstab stimmen mit den, per sudo blkid ausgelesenen Werten überein. Der Fehler kommt vermutlich aus einem anderen Eck.

framps, 2021-04-28:
Hm ... kann es sein dass in der fstab noch was steht was gemounted werden soll aber nicht abgeschlossen ist? Dann hängt es auch beim Booten

G-Kraxen, 2021-04-28:
Daran habe ich auch gedacht aber leider nein.
Der Pi bootet gar nicht.
Selbst die rote LED leuchtet nicht einmal auf.

G-Kraxen, 2021-04-28:
Ich habe den Pi heute mit Raspberry OS (2021-03-04-raspios-buster-armhf-lite) komplett neu aufgesetzt.
Ein Backup gezogen und wieder eingespielt. Leider mit dem selben Resultat -> RBK0208W Es konnte keine UUID in /boot/cmdline.txt für root erneuert werden.

Von gesicherten System:
[image]
Von gesicherten System:
[image]
Hier noch bkid vom gesicherten System:
[image]

Vom zurückgespielten System:
[image]
Vom zurückgespielten System:
[image]

Hier noch der log dazu:
[raspberrypi.log](https://github.com/framps/raspiBackup/files/6393977/raspberrypi.log)

framps, 2021-04-29:
[Im Raspberry Forum](https://forum-raspberrypi.de/forum/thread/52554-raspibackup-restore-probleme-mittlerweile-der-2-raspi-der-beim-restore-nicht-boo/?postID=490187#post490187) hat jemand dasselbe Problem beschrieben und eine Loesung gefunden.

Er nutzt wie Du `TAR_BOOT_PARTITION_ENABLED=1`. Nachdem er das auf `TAR_BOOT_PARTITION_ENABLED=0` gesetzt hat funktionierte alles.

Leidr habe ich momentan keinen Zugriff auf eine Raspberry um das Problem nachzustellen. Einen Issue #334 habe ich aber schon mal dafuer erstellt. Kannst Du auch mal den tar bootpartition ausschalten und testen ob es dann funktioniert?

Raemsna, 2021-04-30:
Hallo nun auch hier. 

Noch ein Hinweis: 
Wenn ich die restorete SD Karte in einem anderen System mounte ergeben sich folgende Unterschiede (mount der boot Partition): 

TAR_BOOT_PARTITION_ENABLED=1
==> cmdline.txt liegt unter ==> /mnt/SDCard/**boot**/cmdline.txt
==> ich denke die cmdline.txt wird deshalb nicht gefunden, die UUID kann nicht geändert werden und das ganze System bootet nicht

TAR_BOOT_PARTITION_ENABLED=0
==> cmdline.txt liegt unter ==> /mnt/SDCard/cmdline.txt
==> UUID kann geändert werden. System bootet

ich bin froh, dass endlich gefunden zu haben, ich beschäftige mich schon seit gefühlt Wochen damit und habe schon an meinem Geisteszustand gezweifelt :D

Raemsna, 2021-04-30:
> Es wurde ein komplett neues Raspberry OS aufgespielt (2021-03-04-raspios-buster-armhf-lite). raspiBackup wurde nicht für Backups eingerichtet, sondern nach der Installation direkt ein Restore auf [...]

Und ich hab schon gedacht, ich bin der einzige mit dem Problem :) Lustig, dass das jetzt so synchron auftaucht

framps, 2021-04-30:
> Lustig, dass das jetzt so synchron auftaucht
Ich finde es eher nicht lustig ... Bin aber froh dass schon jemand anderes die Ursache gefunden hat. Ich wäre nicht so schnell darauf gekommen dass dort die Ursache liegt.

> ich beschäftige mich schon seit gefühlt Wochen damit und habe schon an meinem Geisteszustand gezweifelt :D

Sorry for the inconvenience.

Ich mache diesen Issue jetzt als Duplicate von #334 zu.

closed 2021-04-30 by framps

G-Kraxen, 2021-04-30:
Ich kann bestätigen, dass das Sichern und Rückspielen ohne TAR_BOOT_PARTITION_ENABLED=1 problemlos funktioniert.
@Raemsna konnte dein Pi normal booten nachdem du den Pfad zum cmdline.txt händisch korrigiert hast?

Raemsna, 2021-04-30:
> 
> 
> Ich kann bestätigen, dass das Sichern und Rückspielen ohne TAR_BOOT_PARTITION_ENABLED=1 problemlos funktioniert.
> @Raemsna konnte dein Pi normal booten nachdem du den Pfad zum cmdline.txt händisch korrigiert hast?

das habe ich ehrlich gesagt nicht versucht.

framps, 2021-05-03:
Ich habe einen Hotfix gebaut. Er findet sich [hier](https://github.com/framps/raspiBackup/blob/issue334_tar_boot/raspiBackup.sh). Waere gut wenn er noch von Euch verifiziert werden wuerde :wink:

framps, 2021-05-04:
Issue fixed. Please update your current version with sudo raspiBackup.sh -U -S.