framps/raspiBackup #333: Dropbox/Rclone mount path not accepted issue

PotemkinMcNasty, 2021-04-27:
I have mounted a Dropbox drive on my Raspberry Pi OS using rclone mount. I can browse the drive and list contents, but the raspibackup configuration tool reports "Backup path /home/pi/dropbox folder/ does not exist", so I cannot complete the initial configuration.

Contents of debug log below:
20210427-223053 DBG 1721:      --> initializeDefaultConfigVariables 
20210427-223053 DBG 1855:      <-- initializeDefaultConfigVariables 
20210427-223053 DBG 1860:      --> copyDefaultConfigVariables 
20210427-223053 DBG 1960:          --> substituteNumberArguments 
20210427-223053 DBG 1987:          <-- substituteNumberArguments 
20210427-223053 DBG 1931:      <-- copyDefaultConfigVariables 
20210427-223053 DBG 2702:      --> readConfigParameters 
20210427-223053 DBG 2758:      <-- readConfigParameters 
20210427-223053 DBG 1860:      --> copyDefaultConfigVariables 
20210427-223053 DBG 1960:          --> substituteNumberArguments 
20210427-223053 DBG 1987:          <-- substituteNumberArguments 
20210427-223053 DBG 1931:      <-- copyDefaultConfigVariables 
20210427-223053 DBG 1642:      --> logOptions Standard option files
20210427-223053 DBG 1644:      --- Options:  -l Debug
20210427-223053 DBG 1645:      --- APPEND_LOG=0
20210427-223053 DBG 1646:      --- APPEND_LOG_OPTION=-a
20210427-223053 DBG 1647:      --- BACKUPPATH=/backup
20210427-223053 DBG 1648:      --- BACKUPTYPE=rsync
20210427-223053 DBG 1649:      --- AFTER_STARTSERVICES=
20210427-223053 DBG 1650:      --- BEFORE_STOPSERVICES=
20210427-223053 DBG 1651:      --- CHECK_FOR_BAD_BLOCKS=0
20210427-223053 DBG 1652:      --- COLORING=CM
20210427-223053 DBG 1653:      --- CONFIG_FILE=
20210427-223053 DBG 1654:      --- DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20210427-223053 DBG 1655:      --- DD_BLOCKSIZE=1M
20210427-223053 DBG 1656:      --- DD_PARMS=
20210427-223053 DBG 1657:      --- DD_WARNING=0
20210427-223053 DBG 1658:      --- DEPLOYMENT_HOSTS=
20210427-223053 DBG 1659:      --- DYNAMIC_MOUNT=
20210427-223053 DBG 1660:      --- EMAIL=
20210427-223053 DBG 1661:      --- EMAIL_COLORING=SUBJECT
20210427-223053 DBG 1662:      --- EMAIL_PARMS=
20210427-223053 DBG 1663:      --- EXCLUDE_LIST=
20210427-223053 DBG 1664:      --- EXTENSIONS=
20210427-223053 DBG 1665:      --- FAKE=
20210427-223053 DBG 1666:      --- HANDLE_DEPRECATED=
20210427-223053 DBG 1667:      --- IGNORE_ADDITIONAL_PARTITIONS=0
20210427-223053 DBG 1668:      --- KEEPBACKUPS=3
20210427-223053 DBG 1669:      --- KEEPBACKUPS_DD=0
20210427-223053 DBG 1670:      --- KEEPBACKUPS_DDZ=0
20210427-223053 DBG 1671:      --- KEEPBACKUPS_TAR=0
20210427-223053 DBG 1672:      --- KEEPBACKUPS_TGZ=0
20210427-223053 DBG 1673:      --- KEEPBACKUPS_RSYNC=0
20210427-223053 DBG 1674:      --- LANGUAGE=EN
20210427-223053 DBG 1675:      --- LINK_BOOTPARTITIONFILES=0
20210427-223054 DBG 1676:      --- LOG_LEVEL=1
20210427-223054 DBG 1677:      --- LOG_OUTPUT=2
20210427-223054 DBG 1678:      --- MAIL_ON_ERROR_ONLY=0
20210427-223054 DBG 1679:      --- MAIL_PROGRAM=mail
20210427-223054 DBG 1680:      --- MSG_LEVEL=0
20210427-223054 DBG 1681:      --- NOTIFY_START=0
20210427-223054 DBG 1682:      --- NOTIFY_UPDATE=1
20210427-223054 DBG 1683:      --- PARTITIONBASED_BACKUP=0
20210427-223054 DBG 1684:      --- PARTITIONS_TO_BACKUP=1 2
20210427-223054 DBG 1685:      --- RESIZE_ROOTFS=1
20210427-223054 DBG 1686:      --- RESTORE_DEVICE=
20210427-223054 DBG 1687:      --- ROOT_PARTITION=
20210427-223054 DBG 1688:      --- RSYNC_BACKUP_ADDITIONAL_OPTIONS=
20210427-223054 DBG 1689:      --- RSYNC_BACKUP_OPTIONS=-aHAx --delete
20210427-223054 DBG 1690:      --- RSYNC_IGNORE_ERRORS=
20210427-223054 DBG 1691:      --- SENDER_EMAIL=
20210427-223054 DBG 1692:      --- SKIP_DEPRECATED=
20210427-223054 DBG 1693:      --- SKIPLOCALCHECK=0
20210427-223054 DBG 1694:      --- SMART_RECYCLE=0
20210427-223054 DBG 1695:      --- SMART_RECYCLE_DRYRUN=1
20210427-223054 DBG 1696:      --- SMART_RECYCLE_OPTIONS=7 4 12 1
20210427-223054 DBG 1697:      --- STARTSERVICES=:
[... 505 lines cut ...]
20210427-223056 DBG 3310:                                  -rw-r--r-- 1 root root   252 Apr 27 22:30 raspiBackup.msg
20210427-223056 DBG 3027:                              --> colorAnnotation 1
20210427-223056 DBG 3051:                              <-- colorAnnotation 
--- RBK0049I: Messages saved in /home/@USER@/raspiBackup.msg.
20210427-223056 DBG 3027:                              --> colorAnnotation 1
20210427-223056 DBG 3051:                              <-- colorAnnotation 
--- RBK0026I: Debug logfile saved in /home/@USER@/raspiBackup.log.
20210427-223056 DBG 3329:                          <-- cleanupBackupDirectory 
20210427-223056 DBG 3762:                      <-- cleanupBackup 
20210427-223056 DBG 3768:                      --> cleanupTempFiles 
20210427-223056 DBG 3775:                      <-- cleanupTempFiles 
20210427-223056 DBG 3509:                  --- Terminate now with rc 102
20210427-223056 DBG 3027:                      --> colorAnnotation 1
20210427-223056 DBG 3051:                      <-- colorAnnotation 
--- RBK0010I: @HOSTNAME@: raspiBackup.sh V0.6.6 (83ecd03) stopped at Tue 27 Apr 2021 10:30:56 PM BST with rc 102.
20210427-223056 DBG 3027:                      --> colorAnnotation 1
20210427-223056 DBG 3051:                      <-- colorAnnotation 
??? RBK0005E: Backup failed. Check previous error messages for details.
20210427-223056 DBG 3143:                      --> sendEMail 
20210427-223056 DBG 3265:                      <-- sendEMail 
20210427-223057 DBG 3397:                  --- Masquerading some mount options
20210427-223057 DBG 3415:                  --- Masquerading home directory name
20210427-223057 DBG 3420:                  --- Masquerading hostname
20210427-223057 DBG 3425:                  --- Masquerading sensitive non local IPs
20210427-223057 DBG 3587:                  <-- cleanup

framps, 2021-04-28:
According the log you get
`??? RBK0027E: No external device mounted on /backup. root partition would be used for the backup.`

which maps to 

```
20210427-223056 DBG 4424:                      *** mount
20210427-223056 DBG 4424:                          /dev/mmcblk0p2 on / type ext4 (rw,noatime)
20210427-223056 DBG 4424:                          devtmpfs on /dev type devtmpfs (rw,relatime,size=3867520k,nr_inodes=966880,mode=755)
20210427-223056 DBG 4424:                          sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
20210427-223056 DBG 4424:                          proc on /proc type proc (rw,relatime)
20210427-223056 DBG 4424:                          securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
20210427-223056 DBG 4424:                          tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
20210427-223056 DBG 4424:                          devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
20210427-223056 DBG 4424:                          tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
20210427-223056 DBG 4424:                          tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
20210427-223056 DBG 4424:                          tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
20210427-223056 DBG 4424:                          cgroup2 on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
20210427-223056 DBG 4424:                          cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,name=systemd)
20210427-223056 DBG 4424:                          none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
20210427-223056 DBG 4424:                          cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
20210427-223056 DBG 4424:                          cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
20210427-223056 DBG 4424:                          cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)
20210427-223056 DBG 4424:                          cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
20210427-223056 DBG 4424:                          cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
20210427-223056 DBG 4424:                          cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
20210427-223056 DBG 4424:                          cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
20210427-223056 DBG 4424:                          cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
20210427-223056 DBG 4424:                          systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=34,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
20210427-223056 DBG 4424:                          mqueue on /dev/mqueue type mqueue (rw,relatime)
20210427-223056 DBG 4424:                          debugfs on /sys/kernel/debug type debugfs (rw,relatime)
20210427-223056 DBG 4424:                          sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
20210427-223056 DBG 4424:                          configfs on /sys/kernel/config type configfs (rw,relatime)
20210427-223056 DBG 4424:                          /dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
20210427-223056 DBG 4424:                          tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=800080k,mode=700,uid=1000,gid=1000)
20210427-223056 DBG 4424:                          gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
20210427-223056 DBG 4424:                          fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
20210427-223056 DBG 4424:                          dropbox: on /home/@USER@/dropbox type fuse.rclone (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
```

Please update `BACKUPPATH=/backup` in config file or use installer to set it accordingly.

PotemkinMcNasty, 2021-04-28:
Hi Framp
Thanks by the way for an awesome tool, and your support.
I think this is because the tool cannot be configured for the correct path (/home/pi/dropbox) as the config tool doesn’t accept that as a valid path.
Thanks
Richard

framps, 2021-04-28:
I see. Please try `/home/@USER@/dropbox` or `dropbox:` (Substitute @USER with your real user)

PotemkinMcNasty, 2021-04-28:
Thank you - unfortunately I am getting the same result

framps, 2021-04-29:
Please provide the debug log when you use `/home/@USER@/dropbox` and `dropbox:`.

I unfortunately don't use dropbox and cannot recreate your scenario :disappointed:

PotemkinMcNasty, 2021-04-30:
Will do, but won’t they be the same i.e. the config tool never updates the config file as it cannot “find” the path?
Thanks
Richard

framps, 2021-05-02:
You are right. You have to update the `BACKUPPATH=`  line in the config file `/usr/local/etc/raspiBackup.conf` as follows:

Try
`BACKUPPATH="/home/@USER@/dropbox"`
and
`BACKUPPATH="dropbox:"`

PotemkinMcNasty, 2021-05-03:
Ok that worked, kind of.  I now get an error that the file system doesn’t support Sym links.  Seems Dropbox stopped supporting those.

framps, 2021-05-03:
> Try
> `BACKUPPATH="/home/@USER@/dropbox"`
> and
> `BACKUPPATH="dropbox:"`

Which config setting do you use? I want to fix the parser in the installer :wink: 

> file system doesn’t support Sym links

tar should work.

framps, 2021-06-09:
Closing because of inactivity

closed 2021-06-09 by framps