framps/raspiBackup #334: DEFAULT_TAR_BOOT_PARTITION_ENABLED=1 may be buggy

framps, 2021-04-29:
For details see here in the [German Raspberry Forum](https://forum-raspberrypi.de/forum/thread/52554-raspibackup-restore-probleme-mittlerweile-der-2-raspi-der-beim-restore-nicht-boo/)

Raemsna, 2021-05-04:
Backup mit der Hotfix Version lief ohne Fehler durch
`
20210504-183602 DBG 1721:      --> initializeDefaultConfigVariables 
20210504-183602 DBG 1855:      <-- initializeDefaultConfigVariables 
20210504-183602 DBG 1860:      --> copyDefaultConfigVariables 
20210504-183602 DBG 1960:          --> substituteNumberArguments 
20210504-183602 DBG 1987:          <-- substituteNumberArguments 
20210504-183602 DBG 1931:      <-- copyDefaultConfigVariables 
20210504-183602 DBG 2702:      --> readConfigParameters 
20210504-183603 DBG 2758:      <-- readConfigParameters 
20210504-183603 DBG 1860:      --> copyDefaultConfigVariables 
20210504-183603 DBG 1960:          --> substituteNumberArguments 
20210504-183603 DBG 1987:          <-- substituteNumberArguments 
20210504-183603 DBG 1931:      <-- copyDefaultConfigVariables 
20210504-183603 DBG 1642:      --> logOptions Standard option files
20210504-183603 DBG 1644:      --- Options: 
20210504-183603 DBG 1645:      --- APPEND_LOG=0
20210504-183603 DBG 1646:      --- APPEND_LOG_OPTION=-a
20210504-183603 DBG 1647:      --- BACKUPPATH=/mnt/Backup/RaspWohnzimmer
20210504-183603 DBG 1648:      --- BACKUPTYPE=tar
20210504-183603 DBG 1649:      --- AFTER_STARTSERVICES=
20210504-183603 DBG 1650:      --- BEFORE_STOPSERVICES=
20210504-183603 DBG 1651:      --- CHECK_FOR_BAD_BLOCKS=0
20210504-183603 DBG 1652:      --- COLORING=CM
20210504-183603 DBG 1653:      --- CONFIG_FILE=
20210504-183603 DBG 1654:      --- DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20210504-183603 DBG 1655:      --- DD_BLOCKSIZE=1M
20210504-183603 DBG 1656:      --- DD_PARMS=
20210504-183603 DBG 1657:      --- DD_WARNING=0
20210504-183603 DBG 1658:      --- DEPLOYMENT_HOSTS=
20210504-183603 DBG 1659:      --- DYNAMIC_MOUNT=
20210504-183603 DBG 1660:      --- EMAIL=r@@@@e(23)
20210504-183603 DBG 1661:      --- EMAIL_COLORING=SUBJECT
20210504-183603 DBG 1662:      --- EMAIL_PARMS=
20210504-183603 DBG 1663:      --- EXCLUDE_LIST=--exclude=/home/@USER@/Downloads/* --exclude=/mnt/shared/* --exclude=/mnt/Backup/* --exclude=/mnt/Backupserver/* --exclude=/mnt/Mediaserver/* --exclude=/mnt/SDCard1/* --exclude=/var/log/syslog --exclude=/var/log/daemon.log --exclude=/var/lib/samba/private/msg.sock/*
20210504-183603 DBG 1664:      --- EXTENSIONS=
20210504-183603 DBG 1665:      --- FAKE=
20210504-183603 DBG 1666:      --- HANDLE_DEPRECATED=
20210504-183603 DBG 1667:      --- IGNORE_ADDITIONAL_PARTITIONS=0
20210504-183603 DBG 1668:      --- KEEPBACKUPS=5
20210504-183603 DBG 1669:      --- KEEPBACKUPS_DD=0
20210504-183603 DBG 1670:      --- KEEPBACKUPS_DDZ=0
20210504-183603 DBG 1671:      --- KEEPBACKUPS_TAR=0
20210504-183603 DBG 1672:      --- KEEPBACKUPS_TGZ=0
20210504-183603 DBG 1673:      --- KEEPBACKUPS_RSYNC=0
20210504-183603 DBG 1674:      --- LANGUAGE=EN
20210504-183603 DBG 1675:      --- LINK_BOOTPARTITIONFILES=0
20210504-183603 DBG 1676:      --- LOG_LEVEL=1
20210504-183603 DBG 1677:      --- LOG_OUTPUT=2
20210504-183603 DBG 1678:      --- MAIL_ON_ERROR_ONLY=0
20210504-183603 DBG 1679:      --- MAIL_PROGRAM=ssmtp
20210504-183603 DBG 1680:      --- MSG_LEVEL=0
20210504-183603 DBG 1681:      --- NOTIFY_START=0
20210504-183603 DBG 1682:      --- NOTIFY_UPDATE=1
20210504-183603 DBG 1683:      --- PARTITIONBASED_BACKUP=0
20210504-183603 DBG 1684:      --- PARTITIONS_TO_BACKUP=*
20210504-183603 DBG 1685:      --- RESIZE_ROOTFS=1
20210504-183603 DBG 1686:      --- RESTORE_DEVICE=
20210504-183603 DBG 1687:      --- ROOT_PARTITION=
20210504-183603 DBG 1688:      --- RSYNC_BACKUP_ADDITIONAL_OPTIONS=
20210504-183603 DBG 1689:      --- RSYNC_BACKUP_OPTIONS=-aHAx --delete
20210504-183603 DBG 1690:      --- RSYNC_IGNORE_ERRORS=
20210504-183603 DBG 1691:      --- SENDER_EMAIL=r@@@@e(17)
20210504-183603 DBG 1692:      --- SKIP_DEPRECATED=
20210504-183603 DBG 1693:      --- SKIPLOCALCHECK=0
20210504-183603 DBG 1694:      --- SMART_RECYCLE=0
20210504-183603 DBG 1695:      --- SMART_RECYCLE_DRYRUN=1
20210504-183603 DBG 1696:      --- SMART_RECYCLE_OPTIONS=7 4 12 1
20210504-183603 DBG 1697:      --- STARTSERVICES=sudo service fhem start && sudo service watchdog start && sudo service apache2 start
20210504-183603 DBG 1698:      --- STOPSERVICES=sudo service watchdog stop && sudo service fhem stop && sudo service apache2 stop
[... 771 lines cut ...]
20210504-184403 DBG 3255:          --- --- RBK0151I: Backuppfad /mnt/Backup/RaspWohnzimmer mit Partitionstyp cifs wird benutzt.<br/>
20210504-184403 DBG 3255:          --- --- RBK0036I: Partitionslayout wird gesichert.<br/>
20210504-184403 DBG 3255:          --- --- RBK0158I: tar Backup "/mnt/Backup/RaspWohnzimmer/@HOSTNAME@/@HOSTNAME@-tar-backup-20210504-183602/@HOSTNAME@-tar-backup-20210504-183602.tar" wird erstellt.<br/>
20210504-184403 DBG 3255:          --- --- RBK0085I: Backuperstellung vom Typ tar gestartet. Bitte Geduld.<br/>
20210504-184403 DBG 3255:          --- --- RBK0078I: Backupzeit: 00:07:10.<br/>
20210504-184403 DBG 3255:          --- --- RBK0159I: 5 Backups werden für den Backuptyp tar aufbewahrt. Bitte Geduld.<br/>
20210504-184403 DBG 3255:          --- --- RBK0033I: Bitte warten bis aufgeräumt wurde.<br/>
20210504-184403 DBG 3255:          --- --- RBK0049I: Meldungen wurden in /mnt/Backup/RaspWohnzimmer/@HOSTNAME@/@HOSTNAME@-tar-backup-20210504-183602/raspiBackup.msg gesichert.<br/>
20210504-184403 DBG 3255:          --- --- RBK0026I: Debug Logdatei wurde in /mnt/Backup/RaspWohnzimmer/@HOSTNAME@/@HOSTNAME@-tar-backup-20210504-183602/raspiBackup.log gesichert.<br/>
[... cut ...]
20210504-184406 DBG 3394:      --- Masquerading sender eMail
20210504-184406 DBG 3409:      --- Masquerading some mount options
20210504-184407 DBG 3427:      --- Masquerading home directory name
20210504-184407 DBG 3432:      --- Masquerading hostname
20210504-184407 DBG 3437:      --- Masquerading sensitive non local IPs
20210504-184409 DBG 3599:      <-- cleanup 
`

Restore läuft gerade noch (aber mit der stable Version auf einem anderen Raspi - hoffe beim Restore ist die hotfix Version nicht anders als die stable?)

framps, 2021-05-04:
Vielen Dank fuer den Test. Ist immer besser wenn noch ein Betroffener testet ob der Fix OK ist.

> hoffe beim Restore ist die hotfix Version nicht anders als die stable

Nope. Ich musste sowohl am Backup Codepath als auch Restore CodePath aendern.

> ??? RBK0183E: Rootpartitionscheck endet fehlerhaft mit RC 4.

Das ist ein anderer Fehler als bisher. Nimm mal auch zum Restore auch die neue Version.

Raemsna, 2021-05-04:
ok neuer Restore mit neuer Version läuft
Log von dem fehlgeschlagenen Restore: 
[raspiBackup.log](https://github.com/framps/raspiBackup/files/6422920/raspiBackup.log)

Raemsna, 2021-05-04:
Update: Restore lief durch, Gerät mit restoreter SD bootet. Vielen Dank

`sudo raspiBackup.sh -d /dev/sda /mnt/Backup/RaspWohnzimmer/RaspiWohnzimmer/RaspiWohnzimmer-tar-backup-20210504-183602/
--- RBK0009I: RaspiBuero: raspiBackup.sh V0.6.6-hotfix (f3de486) Di 4. Mai 19:25:41 CEST 2021 gestartet.
!!! RBK0173W: =========> HINWEIS <=========
!!! RBK0173W: Dieses ist ein temporärer Hotfix der auf die nächste Version upgraded werden muss sobald eine verfügbar ist.
!!! RBK0173W: =========> HINWEIS <=========
!!! RBK0006W: Ziel /dev/sda mit 29.75 GiB ist kleiner als die Backupquelle mit 59.47 GiB. Die root Partition wird entsprechend verkleinert. HINWEIS: Der Restore kann fehlschlagen wenn sie zu klein wird.
!!! RBK0065W: Gerät /dev/sda wird repartitioniert und die gesamten Daten werden gelöscht.
--- RBK0067I: Momentane Partitionen auf /dev/sda:
Number  Start   End      Size     Type     File system  Flags
 1      4,19MB  31954MB  31950MB  primary  fat32        lba
--- RBK0066I: Gerät /dev/sda wird überschrieben mit der gesicherten Boot- und Rootpartition.
--- RBK0069I: Bootpartition /dev/sda1 wird formatiert und erhält die zurückgespielte Bootpartition.
--- RBK0070I: Rootpartition /dev/sda2 wird formatiert und erhält die zurückgespielte Rootpartition.
--- RBK0038I: Bist Du sicher? j/N
j
--- RBK0050I: Backup wird von /mnt/Backup/RaspWohnzimmer/RaspiWohnzimmer/RaspiWohnzimmer-tar-backup-20210504-183602 zurückgespielt.
--- RBK0004I: Zweite Partition wird von 59.22 GiB auf 29.50 GiB angepasst.
--- RBK0053I: Erste Partition (Bootpartition) wird auf /dev/sda1 zurückgespielt.
--- RBK0055I: Zweite Partition (Rootpartition) wird auf /dev/sda2 zurückgespielt.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0010I: RaspiWohnzimmer: raspiBackup.sh V0.6.6-hotfix (f3de486) Di 4. Mai 20:09:43 CEST 2021 beendet mit Returncode 0.
--- RBK0076I: Restore erfolgreich beendet.
`

framps, 2021-05-04:
Vielen Dank fuer den Test :+1: 

Ich merge den Fix gleich und publishe ihn.

framps, 2021-05-04:
Mit `sudo raspiBackup.sh -U -S` wird der Fix eingespielt.

framps, 2021-05-04:
> Update: Restore lief durch, Gerät mit restoreter SD bootet. Vielen Dank

Dann bitte den issue closen :wink: 

Noch mal vielen Dank fuer die Hilfe.

framps, 2021-05-04:
Sehe gerade das ich den Issue erstellt habe und closen muss ...

closed 2021-05-04 by framps