framps/raspiBackup #338: Restart stopped services before to perform intelligent backup strategy

framps, 2021-05-17:
In order to keep timeout of running services short the services should be started immediately when the backup finished. Cleanup of existing backups/backup strategy processing may take some additional time - in particular for rsync backups

teemue, 2021-05-23:
Thanks for opening an issue on my behalf :-)

framps, 2021-06-09:
@teemue I fixed the issue right now. Would be great if you verify the fix. Should I provide the fix for you for testing?

teemue, 2021-06-10:
Thanks, I would appreciate it and test right away!

framps, 2021-06-10:
Great. Thank you very much. 

Please grab [this file](https://github.com/framps/raspiBackup/blob/development/raspiBackup.sh) and copy it into your home directory. Then call `sudo ./raspiBackup.sh` (not the leading period !) .

teemue, 2021-06-10:
/home/pi or /home/ ? I tried but running the script ran previous version 0.6.6, which is installed

framps, 2021-06-10:
I updated the previous description. Note of the leading period in the command :wink:

teemue, 2021-06-11:
Worked perfectly. Thank you!

framps, 2021-06-11:
Great. Thank you for your verfication :+1:

closed 2021-06-11 by framps