framps/raspiBackup #340: Error while trying to backup with script

Vannixxo, 2021-06-22:
Hello I always used the script by terminal and successfully restored my rpi two times!

Now I restored the image created with raspiBackup to an external USB hdd and not the SD anymore.
The systems boots and it's fine, but when I try to create a new backup I get this error:

sudo bash /home/pi/RaspiBackup/RaspiBackup.sh start -c /media/pi/Time\ Capsule/RaspiBackup/RaspiBackup$(date +%d-%m-%Y).img
blockdev: cannot open /dev/mmcblk0: No such file or directory
blockdev: cannot open /dev/mmcblk0: No such file or directory
 Starting SD Image backup process
 Creating sparse /media/pi/Time
dd: invalid number: ‘’
 /media/pi/Time Capsule/RaspiBackup/RaspiBackup22-06-2021.img has not been created or has zero size

framps, 2021-06-22:
Please provide the debug log.

closed 2021-06-22 by Vannixxo

framps, 2021-06-22:
@Vannixxo You just closed the issue. Would be great if you let me know what's the reason. Did you find the root cause for the issue?

Vannixxo, 2021-06-23:
> @Vannixxo You just closed the issue. Would be great if you let me know what's the reason. Did you find the root cause for the issue?
You're right sorry, the reason is that I noticed I was using a really outdated script!