framps/raspiBackup #341: Not working - command not found

91fabbai91, 2021-06-25:
Hey guys,

I have installed raspiBackup on my system with Ubuntu 20.04 64bit, but the backup creation is not working. Find attached the logs. There are several messages that some functions of bash scripts are not found. So I cannot detect the free space on the samba share and the device itself. 
The mounting on the Raspberry Pi works and I can see the directories on the HDD.

Do you have any idea why this doesn't work.

Thanks in advance and best regards,
Fabian

My setup:
Raspberry Pi4 8GB
Ubuntu 20.04 64Bit
1TB SanDisk SSD via USB3

Fritzbox 7490
Fritz.OS 7.27
1TB HDD via USB3

[raspiBackup.log](https://github.com/framps/raspiBackup/files/6714532/raspiBackup.log)

framps, 2021-06-25:
```
??? RBK0039E: Mail Program mail ist nicht installiert um eMail zu senden.
```
Please either install an eMail client or disable eMail notification :wink: 

```
/usr/local/bin/raspiBackup_mem_post.sh: line 34: getMemoryFree: command not found
```

That's a follow on error caused by RBK0039E and will disappear if the eMail issue is fixed.

91fabbai91, 2021-06-25:
So thanks I wasn't aware that the issue with the missing mail client is the reason that it's not working.
I installed ssmtp via apt and it fixed it. 

Thanks you for your help

closed 2021-06-25 by 91fabbai91