framps/raspiBackup #342: Multicore Processing while do (compressed) tar backup

91fabbai91, 2021-06-25:
Hi, 

I have a Raspberry Pi4 with 1TB SSD attached and have a 1TB SMB Share, where I can store the backup.
So I can't use rsync because of missing hardlinks on that device. So I tried it with the use of tar, but it is really time consuming. As I looked into htop I saw that just one of four cores is used. Is there a possibility to accelerate it by using multicores?

On my laptop I usually use [pigz](https://www.zlib.net/pigz/) for parallel compressing, but I am not sure if it also accelerates the tar-balling itself.

Best regards and thanks in advance

framps, 2021-06-25:
There exists a tar option `--use-compress-program=<compressionprogram> . 

You can use `DEFAULT_TAR_BACKUP_ADDITIONAL_OPTIONS` to add this option with pigz to the tar call. I never tested this. Just give it a try and keep me posted :wink:

framps, 2021-07-10:
Any update on this issue from you side?

framps, 2021-07-15:
Closing because of inactivity

closed 2021-07-15 by framps