framps/raspiBackup #343: raspiBackup beendet mit Returncode 42

H-a-m-i, 2021-07-07:
Hallo.

Ich habe gestern raspiBackup auf meinem Raspberry installiert und heut bei einem manuellen Backup dann den Fehlercode "RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen." Vorhergehende Meldung ist: "RBK0010I: raspberrypi: raspiBackup.sh V0.6.6 (c8d3058) Mi 7. Jul 12:49:41 CEST 2021 beendet mit Returncode 42." In den FAQ habe ich gelesen, dass eine Problembeschreibung hier in github präferiert wird.

Sorry, ich hab eigentlich eher weniger Ahnung von Linux etc. und hab die Installation von raspiBackup nur mit einer Schritt-für-Schritt-Anleitung geschafft xD Ich weiß nicht mal, wie ich die Debug Logdatei hier anhänge, soll ich die komplett hier reinkopieren oder wie mach ich das am besten?

Danke im voraus

framps, 2021-07-07:
Den RC 42 sollte raspiBackup eigentlich nicht melden. Der kommt nur wenn irgendeine falsche Eingabe nicht richtig abgefangen wurde oder ein unerwarteter Fehler auftritt.

Um die Ursache zu finden brauche ich das Debuglog. Bitte nicht den ganzen Text hier per copy & paste reinschreiben sondern die Logdatei per drag'n'drop anhaengen.

H-a-m-i, 2021-07-07:
Hmm ok. Habe gestern schon mal ein manuelles Backup gemacht, da kam kein Fehler (außer dass keine Mail gesendet werden konnte, was ich mittlerweile behoben habe). Seit dem hatte ich PiVPN bzw. wireguard installiert, wobei ich auch diverse Probleme hatte und es deshalb mehrfach deinstalliert und wieder installiert habe. Nun läuft das aber und ich wollte eben ein manuelles backup ausführen.

> sondern die Logdatei per drag'n'drop anhaengen.

Ahh so geht das =) Alles klar

[raspiBackup.log](https://github.com/framps/raspiBackup/files/6778279/raspiBackup.log)

framps, 2021-07-07:
Der rsync Backup startet - wird aber aus nicht ersichtlichen Gruenden bei 83% abgebrochen. 

```
  1,124,543,137  83%  129.44kB/s    0:29:14  
  1,124,805,281  83%  158.00kB/s    0:23:55  
  1,125,067,425  83%  203.50kB/s    0:18:33  
  1,125,329,569  83%  170.10kB/s    0:22:10  
  1,125,591,713  83%  179.30kB/s    0:21:00  20210707-124807 DBG 3494:                          --> cleanup 
```

Nimm doch bitte noch die Option `-v` mit auf und liefere dann das debuglog. Ich hoffe dann wird rsync gespraechiger und sagt warum der rsync abgebrochen wird.

H-a-m-i, 2021-07-07:
Also ich hatte es jetzt nochmal durchlaufen lassen mit -v also mit ` sudo raspiBackup.sh -m -v detailed` (ich hoffe das war richtig so?!) Allerdings ist die Lodgatei nun kleiner. Zudem hatte ich `DEFAULT_RSYNC_BACKUP_ADDITIONAL_OPTIONS="--info=NAME0"` vorhin beim ausführen nicht gesehen. Mache es also nun nochmal. 
Einfach nur `DEFAULT_RSYNC_BACKUP_ADDITIONAL_OPTIONS="--info=NAME0"` im Terminal eingeben?
Und dann eben ` sudo raspiBackup.sh -m -v detailed` ?

Trotzdem hier mal die logdatei von vorhin:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/6780213/raspiBackup.log)

H-a-m-i, 2021-07-08:
So hier dann die aktuelle logdatei:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/6780781/raspiBackup.log)

Diesmal hat es schon nach paar Minuten abgebrochen

H-a-m-i, 2021-07-08:
Guten Morgen,

hier noch mal eine logdatei von einem Backup das nach 1 1/2 Stunden abgebrochen hat:

[raspiBackup.log](https://github.com/framps/raspiBackup/files/6782776/raspiBackup.log)

framps, 2021-07-08:
Im Log findest sich wieder keine Indikation warum rsync einfach so anhaelt :cry: Ich habe dieses Verhalten von rsync in den bald 10 Jahren die raspiBackup existiert noch nie gesehen. Solange rsync einfach ohne Fehlermeldung stehenbleibt kann raspiBackup natuerlich auch kein Backup erfolgreich erstellen.

Ich habe mal zusammengeschrieben was mir so auffaellt. Vielleicht klingelt es dann bei Dir und Du hast eine Idee wo die Ursache zu suchen ist.

1. Ich habe [diesen Link](https://serverfault.com/questions/640973/rsync-termiates-after-a-while-unexpectedly) gefunden. Kann es sein dass Du zu wenig Memory hast? In der Antwort wir geraten waehrend des Backups mit rsync mal den Speicher zu monitoren.

2. Auch ist mir

```
20210708-031255 DBG 5317:                  *** cat /proc/cmdline
20210708-031255 DBG 5317:                      coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=d9b3f436-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
```
aufgefallen. Du benutzt zwar ein normales Buster aber hast ziemlich viel zusaetzlich in der cmdline reingeschrieben. U.a. auch Memory Settings. Weisst Du was die Settings alle bewirken? Ich nicht.

3. 
> habe gestern schon mal ein manuelles Backup gemacht, da kam kein Fehler

Heisst das dass Du doch mal ein Backup mit raspiBackup erstellen konntest und nach weiteren Aenderungen am System bricht raspiBackup ab? Dann liegt die Ursache irgendwo in den Aenderungen.

4. 1 Stunde um die doch relativ wenigen Daten zu kopieren finde ich sehr lange. Kann es sein dass Dein Backupmedium buggy ist? Suche mal in /var/log/syslog nach Fehlermeldungen. Oder probiere mal ob ein anderes Backupmedium dasselbe Problem hat.

H-a-m-i, 2021-07-08:
> Im Log findest sich wieder keine Indikation warum rsync einfach so anhaelt 😢 Ich habe dieses Verhalten von rsync in den bald 10 Jahren die raspiBackup existiert noch nie gesehen.

Ist ja echt merkwürdig. Und dann passiert das gerade bei mir, wo ich doch eigentlich keine Ahnung davon habe und alles nur nach Schritt-für-Schritt-Anleitungen machen kann 🙈😄

> 1. Ich habe [diesen Link](https://serverfault.com/questions/640973/rsync-termiates-after-a-while-unexpectedly) gefunden. Kann es sein dass Du zu wenig Memory hast? In der Antwort wir geraten waehrend des Backups mit rsync mal den Speicher zu monitoren.

Hmm zu wenig Memory? Meinst Arbeitsspeicher? Hab den Pi 3 B+ der hat glaub 1 GB.
Oder meinst die Speicherkarte? Das kann fast nicht sein, ich hab ne 32 GB Karte drin und bis vor paar Tage hatte ich ausschließlich pihole laufen, für was anderes hatte ich den pi letztes jahr nicht gekauft. Und die Tage kam dann eben noch raspiBackup, unattended-upgrades und wireguard dazu. So voll kann die Karte also nicht sein. Wie gesagt, habe ich echt überhaupt keine Ahnung von der Materie, muss morgen erstmal den link von dir genau angucken und googlen wie ich das mit dem monitoren mache. 

> Du benutzt zwar ein normales Buster aber hast ziemlich viel zusaetzlich in der cmdline reingeschrieben. U.a. auch Memory Settings. Weisst Du was die Settings alle bewirken? Ich nicht.

Auch davon habe ich echt kein plan, habe selbst nie etwas in diese cmdline reingeschrieben. 🙈 Habe immer nur nach Anleitungen letztes Jahr pihole und dann die Tage den Rest installiert. Wenn nicht mal du weißt was das bewirkt, dann hab ich ja echt die a-karte. 

> Heisst das dass Du doch mal ein Backup mit raspiBackup erstellen konntest und nach weiteren Aenderungen am System bricht raspiBackup ab? Dann liegt die Ursache irgendwo in den Aenderungen.

Jepp nachdem ich, glaub am Dienstag, raspiBackup nach deiner Anleitung installiert und danach manuell ausgeführt hatte, lief es ohne Fehler (bis auf das mit der Mail) durch. Danach kam dann wireguard und unattended-upgrades dazu und seitdem kommt der Fehler. Hatte aber eben auch echt probleme mit PiVPN und wireguard und es glaub insgesamt drei mal deinstalliert und neu installiert. Vielleicht hab ich da irgendwas versaut was raspiBackup nicht mag. 

> 1 Stunde um die doch relativ wenigen Daten zu kopieren finde ich sehr lange. Kann es sein dass Dein Backupmedium buggy ist? Suche mal in /var/log/syslog nach Fehlermeldungen. Oder probiere mal ob ein anderes Backupmedium dasselbe Problem hat.

Ich muss zugeben, dass mein Stick der da dran hängt schon ein ziemlich alter Schinken ist. Werde es nachher oder morgen früh nochmal mit einem neueren versuchen und dann berichten. Auch die anderen Sachen werd ich morgen mal recherchieren und dann gucken ob ich mit Tante Google was rausfinden kann. Mal sehen ob ich in der syslog was finde (wenn ich es dann überhaupt als Fehler erkenne 😄)

H-a-m-i, 2021-07-09:
So, nachdem ich heute Nacht nen neuen Stick eingerichtet und ein manuelles Backup durchgeführt habe, hat dies ohne Fehler funktioniert. Dauerte aber auch 50 Minuten. Das automatische Backup, das dann um 4 Uhr durchgeführt wurde, war auch nach knapp 20 Minuten erfolgreich. Das Ganze schien also wirklich nur an dem alten Stick zu liegen. Hätte ich das mal lieber sofort versucht, bevor ich dich hier beanspruche. Tut mir leid, hab eben echt kein Plan davon 🙈

Allerdings kam gleich nach der "Backup erfolgreich Mail" diese Nachricht:

[image]

framps, 2021-07-09:
> Das Ganze schien also wirklich nur an dem alten Stick zu liegen. Hätte ich das mal lieber sofort versucht, bevor ich dich hier beanspruche. Tut mir leid, hab eben echt kein Plan davon see_no_evil

Kein Problem. Wie gesagt bekomme ich seit knapp 10 Jahren immer wieder Fehlerberichte aber dass rsync ohne Kommentar einfach so endet wenn das Backupmedium alt bzw irgendwie nicht OK ist ist mir auch noch nicht untergekommen. Hauptsache es funktioniert jetzt :wink: 

Zu der Fehlermeldung: 
Bist Du sicher dass Du cups stoppen willst? Druckt bei Dir jemand in der Nacht um 3:00 Uhr? Ich wuerde das rausnehmen. Oder Du aenderst die folgende Zeile

```
STOPSERVICES=systemctl stop cron && systemctl stop pihole-FTL && systemctl stop cups && systemctl stop unattended-upgrades && systemctl stop xrdp && systemctl stop lighttpd
```
in
```
STOPSERVICES=systemctl stop cron && systemctl stop pihole-FTL && systemctl stop cups &>/dev/null && systemctl stop unattended-upgrades && systemctl stop xrdp && systemctl stop lighttpd
```
Bitte dann den Issue closen.

H-a-m-i, 2021-07-09:
> Hauptsache es funktioniert jetzt 😉

Ja zum Glück 🙂 

> Bist Du sicher dass Du cups stoppen willst?

Ja da sieht man mal wie wenig Ahnung ich hab, die ganzen Dienste die gestoppt werden sollen hab ich nämlich auch blind aus irgend nem Forum übernommen 😬 keine Ahnung dass das ein Druckdienst ist 🙈 der braucht natürlich nicht gestoppt werden.

Ich danke dir wirklich vielmals für deine Zeit und deine Hilfe!

closed 2021-07-09 by H-a-m-i