framps/raspiBackup #344: RaspiBackup / Sicherung von Linux Mint 20.2 EFI-Installation

Mitch187, 2021-07-11:
Hallo,

erstmal möchte ich mich für Raspibackup bedanken. Ich nutze es seit mehreren Jahren auf meinen Raspberry PIs. Jetzt möchte ich eine Raspberry PI durch neue Hardwareund ein neues Betriebssystem ersetzen. Ich nutze Rsync mittels eingehangenem NFS.

Scheinbar hat Rasbibackup ein Problem mit EFI-Partitionen,was aber auf einem UEFI-Board notwendig ist:
**??? RBK0203E: Boot device kann nicht erkannt werden. Bitte das Problem mit einem Debuglog welches mit Option '-l debug' erstellt wird berichten.**

Ich habe mal einen Ausschnitt des Debug.log angehangen. Ich kann mir nicht vorstellen, dass es am Btrfs Dateisystem von sda2 liegt.

Hier der mount-point der EFI Partition:
_/dev/sda1 on /boot/efi type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)_

Es wäre schon, wenn man mit Raspibackup auch EFI Partionen sichern könnte. Ich habe versucht, dass Problem noch zu umgehen, indem ich auf  partitionsorintiertes Backup umgestellt habe, um nur sda2 zu sichern:

Optionen fuer partitionsorientierten Modus
_DEFAULT_PARTITIONBASED_BACKUP="1"
DEFAULT_PARTITIONS_TO_BACKUP="2"_

Aber leider ignoriert er dann nicht sda1 und bringt weiterhin den Fehler. So hätte ich als kleines Workarround nur sda2 sichern können, um anschließend den EFI anders zu sichern.

Gruss,
Mitch aka Michael

[Fehlermeldung Raspibackup.txt](https://github.com/framps/raspiBackup/files/6796661/Fehlermeldung.Raspibackup.txt)

framps, 2021-07-11:
Ich koennte jetzt antworten: Nicht unterstuetztes OS :smile: 

Aber sofern Du bereit bist zu helfen und ich kein Environment aufsetzen muss um Dein Problem bei mir lokal zu reporduzieren will ich gerne mal reinsehen. Sofern ein Fix einfach moeglich ist baue ich Dir den auch :wink: 

Die Meldung ist etwas ungenau: Sie kommt auch wenn das Root Device nicht erkannt wird und das ist bei Dir der Fall.

Es scheint bei Dir `/dev/sda2[/@]` zu sein. Kannst Du mal die Ausgabe von `mount` von Deinem System zeigen?

Mitch187, 2021-07-11:
Hallo,
das ist total super, dass Du versuchst mir zu helfen. Ich weiss schon, warum ich damals dich per Paypal untersützt habe, würde aber nochmal für den extra Service eine Kleinigkeit nachwerfen. Ich habe selten so ein geniales OpenSource Porjekt gesehen, was auch wirklich immer zuverlässig funktioniert hat. Ich habe ungefähr 200 Sicherungen durchgeführt und ca. 5 Resto
[raspiBackup.log](https://github.com/framps/raspiBackup/files/6797082/raspiBackup.log)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/6797083/raspiBackup.log)
[raspibackup log und msg.tar.gz](https://github.com/framps/raspiBackup/files/6797086/raspibackup.log.und.msg.tar.gz)
[raspibackup log und msg.tar.gz](https://github.com/framps/raspiBackup/files/6797087/raspibackup.log.und.msg.tar.gz)
[log-und-msg.zip](https://github.com/framps/raspiBackup/files/6797090/log-und-msg.zip)
res. Ohne Probleme!

 /dev/sda2 ist tatsächlich das Btrfs Filesystem. Dies habe bewusst gewählt, weil ich mittels Timeshift kurze und schnelle Snapshots anlegen wollte, um noch flexibeler zu sein. Aber zwigend darauf angewiesen bin ich nicht, mir wäre tatsächlich raspibackup wichtiger, als zuverläössige Backup und Restore-Lösung. Also falls es daran liegen sollte, werde ich Btrfs raus.

[mount.txt](https://github.com/framps/raspiBackup/files/6797078/mount.txt)
[source.txt](https://github.com/framps/raspiBackup/files/6797079/source.txt)
[raspibackup-debug.txt](https://github.com/framps/raspiBackup/files/6797080/raspibackup-debug.txt)

framps, 2021-07-11:
Sofern jetzt btrfs nicht noch irgendwie in die Suppe spuckt sollte es jetzt wohl funktionieren :wink: Den Fix findest Du [hier](https://raw.githubusercontent.com/framps/raspiBackup/issue344/raspiBackup.sh). 

Wenn der Backup jetzt durchlaufen sollte bitte auch den Restore testen. raspiBackup kann bereits btrfs restoren denn Code ist dafuer bereits vorhanden. Ob der Restore mit btrfs aber noch funktioniert weiss ich nicht denn btrfs ist nicht in der raspiBackup Regressionsuite drin.

Mitch187, 2021-07-11:
Der Fehler bleibt aktuell der selbe. Siehe Anhang.
[log und msg.zip](https://github.com/framps/raspiBackup/files/6797452/log.und.msg.zip)

framps, 2021-07-11:
> Soll ich es lieber vorher nochmal mit etx4 versuchen?

Up to you. 

Es gab einen klitzekleinen Fehler bei meinem Fix den ich eben korrigiert habe. Ich schlage vor dass Du es jetzt noch mal mit der geaenderten Version probierst :wink:

Mitch187, 2021-07-11:
Hi,

ich bin zwischenzeitlich auf ext4 gewechselt. Aber ich möchte es anschliessend nochmal mit Btrfs probieren.

Also die Sichberung war erfolgreich, aber der Restore auf einem USB-Stick nicht (gesichert wurde von einer SSD):
--- RBK0009I: mm-retro: raspiBackup.sh V0.6.6-hotfix (8953b57) So 11. Jul 23:37:47 CEST 2021 gestartet.
!!! RBK0173W: =========> HINWEIS <========= 
!!! RBK0173W: Dieses ist ein temporärer Hotfix der auf die nächste Version upgraded werden muss sobald eine verfügbar ist. 
!!! RBK0173W: =========> HINWEIS <=========
!!! RBK0018W: Ziel /dev/sdb mit 14.72 GiB ist größer als die Backupquelle mit 9.50 KiB. Die root Partition wird entsprechend vergrößert um den ganzen Platz zu benutzen.
!!! RBK0065W: Gerät /dev/sdb wird repartitioniert und die gesamten Daten werden gelöscht.
--- RBK0146I: Keine Partitionstabelle auf /dev/sdb gefunden.
--- RBK0066I: Gerät /dev/sdb wird überschrieben mit der gesicherten Boot- und Rootpartition.
--- RBK0069I: Bootpartition /dev/sdb1 wird formatiert und erhält die zurückgespielte Bootpartition.
--- RBK0070I: Rootpartition /dev/sdb2 wird formatiert und erhält die zurückgespielte Rootpartition.
--- RBK0038I: Bist Du sicher? j/N 
j
--- RBK0050I: Backup wird von /backup/mm-retro/mm-retro-rsync-backup-20210711-225143 zurückgespielt.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0010I: mm-retro: raspiBackup.sh V0.6.6-hotfix (8953b57) So 11. Jul 23:37:51 CEST 2021 beendet mit Returncode 112.
??? RBK0077E: Restore wurde fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.

log:

20210711-234150 DBG 4493:                  --- Creating mountpoint /tmp/raspiBackup
20210711-234150 DBG 4496:                  --- Umounting boot partition /dev/sdb1
umount: /dev/sdb1: not mounted.
20210711-234150 DBG 4503:                  --- Umounting root partition /dev/sdb2
umount: /dev/sdb2: not mounted.
Checking that no-one is using this disk right now ... OK

Disk /dev/sdb: 14,74 GiB, 15812526080 bytes, 30883840 sectors
Disk model: USB Flash Drive
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Last LBA specified by script is out of range.
Last LBA specified by script is out of range.
Failed to apply script headers, disk label not created.
Leaving.

20210711-234150 DBG 1452:                      --> exitError 112
20210711-234150 DBG 1459:                      <-- exitError 112
20210711-234150 DBG 3494:                      --> cleanup
20210711-234150 DBG 3027:                          --> colorAnnotation 1
20210711-234150 DBG 3051:                          <-- colorAnnotation
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
20210711-234150 DBG 3606:                          --> cleanupRestore
20210711-234150 DBG 3610:                          --- Got trap EXIT

framps, 2021-07-13:
Vermutlich wegen `Last LBA specified by script is out of range.` kann keine Partitionierung vorgenommen werden. Kannst Du mal das gesamte Debuglog zeigen?

framps, 2021-07-26:
> Der Fehler bleibt aktuell der selbe

Willst Du dass ich an dem Issue weiter arbeite? Ansonsten bitte closen :wink:

Mitch187, 2021-08-02:
Vielen Dank, ich habe dann doch eine andere Lösung gefunden.

closed 2021-08-02 by Mitch187

framps, 2021-08-02:
> ich habe dann doch eine andere Lösung gefunden.

Perfekt :+1: . Aber moechtest Du nicht noch in 2-3 Saetzen kurz Deine andere Loesung beschreiben? Dann kann eventuell jemand anderes aus der Community von Deiner Loesung profitieren :wink: