framps/raspiBackup #345: Installationsscript wird abgebrochen

miracuruzrh, 2021-07-15:
Guten Tag

Als erstes vielen Dank für ein so umfangreiches und hilfreiches Programm.
Nun möchte ich gerne eine Fehlermeldung bei der Installation melden.

Das Installationsscript tätige ich als normaler User (simon), welche sudo Rechte hat.
`curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/installation/install.sh | sudo bash`
Ich erhalte einen blauen Bildschirm mit der Info, dass die Internetverbindung geprüft wird und komme bis zum Schirm, welcher die Navigation erklärt. Beim Bestägigen von "OK" mit der Enter Taste bricht das script ab und sehe dann noch Teile des blauen Bildschirms mit den letzen Informationen:
`<[4]  + 2061 done                   curl -s  | 
────       2062 suspended (tty input)  sudo bash────`

Ein Aufruf mit logfile mittels:
`curl -sL https://raspibackup.linux-tips-and-tricks.de/install | sudo bash -s -- -d`
liefert nicht viele Hinweise:
` ⚡ root@DeskPi  ~  cat raspiBackupInstallUI.log 
20210715-021959: DBG: 3615 -  raspiBackupInstallUI.sh 0.4.3.5, 2021-06-27/21:13:36 - ee5d6a1
20210715-021959: DBG: 3616 -  whiptail version: whiptail (newt): 0.52.20
20210715-021959: DBG: 0684 -  Checking for /usr/local/bin
20210715-021959: DBG: 0684 -  Checking for /usr/local/etc
20210715-021959: DBG: 0684 -  Checking for /etc/cron.d
20210715-021959: DBG: 3419 -  -> uiInstall 
20210715-021959: DBG: 0799 -     -> isInternetAvailable 
20210715-021959: DBG: 0804 -     <- isInternetAvailable 0
20210715-021959: DBG: 3350 -     -> downloadPropertiesFile_do 
20210715-022000: DBG: 3381 -        -> shouldRenewDownloadPropertiesFile 
20210715-022000: DBG: 3395 -           20210715 : 20210715
20210715-022000: DBG: 3398 -           Skip download
20210715-022000: DBG: 3404 -        <- shouldRenewDownloadPropertiesFile 1
20210715-022000: DBG: 3355 -     <- downloadPropertiesFile_do 
20210715-022000: DBG: 3317 -     -> parsePropertiesFile 
20210715-022000: DBG: 3325 -        Script properties: v: 0.6.6 i:  b: 0.6.6.1
20210715-022000: DBG: 3333 -        Installer properties: v: 0.4.3.5 i:  b: 
20210715-022000: DBG: 3335 -     <- parsePropertiesFile 
20210715-022000: DBG: 3340 -     -> parseCurrentVersions 
20210715-022000: DBG: 3343 -        Current script version: 
20210715-022000: DBG: 3344 -        Current installer version: 
20210715-022000: DBG: 3345 -     <- parseCurrentVersions `

ich kann danach manuell weiterarbeiten mittels
`sudo -i su` und das angelegte schript :"raspiBackupInstallUI.sh" ausführbar machen, mittels:
`chmod +x ./raspiBackupInstallUI.sh` und danach weiter arbeiten.

Denke aber, irgendwo müsste ein Fehler sein.

Das Programm läuft auf einem Raspberry4 mit 8GB Ram und alle Pakete sind aktuell.
Es läuft ausserdem noch ein pi-hole und ein nginx-server.
Ansonsten ist die Installation unverändert.

miracuruzrh, 2021-07-15:
Nun hatte ich doch erfolg. Beim ersten mal hatte ich wohl einen Schreibfehler gemacht.
Ich musste "bash" durch "zsh" ersetzen. Somit hat es funktioniert.
Also mit diesem Befehl:
`curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/installation/install.sh | sudo zsh`
Vielleicht könnte dies noch in die Installationsanleitung aufgenommen werden?

Beste Grüsse Simon

framps, 2021-07-15:
Vielen Dank dass Du das Problem berichtest. Welches OS nutzt Du?

> Allerdings verwende ich zsh als default shell mit der oh-my-zsh Erweiterung.
> Könnte dies evtl. der Grund dafür sein? Wenn ja, wüsste ich grad nicht, wie ich das script anders aufrufen müsste.

Das ist ziemlich sicher der Grund. Du wuerdest mir helfen wenn Du folgende Befehle auf Deinem System eingeben wuerdest. Dabei wird nur der Installer gestartet - aber er bricht ja bei bash gleich wieder ab. Somit wird also nichts auf Deinem System geaendert. Aber ich bekommen detailiertere Info was die Ursache sein kann :wink: 

```
script 
sudo bash -x raspiBackupInstallUI.sh
exit
```
Danach den Inhalt der Datei `typescript` im aktuellen Verzeichnis hier hochladen.

miracuruzrh, 2021-07-15:
Vielen Dank für die Rückmeldung

> Vielen Dank dass Du das Problem berichtest. Welches OS nutzt Du?

Ich nehme an, dass das OS vom Raspberry selbst gemeint ist.
Installiert ist "Raspberry Pi OS Lite (32-bit)" (ehemals Raspian) Installationsdatum ist ca. 13.07.21. Die reine Shell Version.

Hier einige Angaben zum OS.
```
 simon@DeskPi  ~  sudo su
 ⚡ root@DeskPi  /home/simon  cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
 ⚡ root@DeskPi  /home/simon  uname -a
Linux DeskPi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
 ⚡ root@DeskPi  /home/simon  cat /proc/version
Linux version 5.10.17-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1421 SMP Thu May 27 14:00:13 BST 2021
 ⚡ root@DeskPi  /home/simon  vcgencmd version
May 27 2021 14:02:15 
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start)
 ⚡ root@DeskPi  /home/simon  
```

Allerdings musste ich das Vorgehen etwas abändern, da ich die Kommandos auf diese Weise nicht verwenden konnte.
Der Grund ist, dass mit dem eigentlichen Installationsscript die raspiBackupInstallUI.sh im /root Verzeichnis abgelegt wird, auch wenn ich den Aufruf von `curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/installation/install.sh | sudo bash` von meinem User "simon" in dessen home Verzeichnis /home/simon tätige.
Mit dem user "simon" habe ich ja nicht ins /root Verzeichnis wechseln und von dort aus das Kommando starten (genau wegen dem ".").

Also habe ich die Datei raspiBackupInstallUI.sh nach /home/simon kopiert und mit chown den Besitzer zu simon gewechselt und dann die oben vorgeschlagenen Kommandos ausgeführt.

Ich hoffe, dass alles richtig geklappt hat, da ich mich mit script / typescript leider nicht auskenne kann ich es leider nicht kontrollieren. Daher lade ich mal 2 Dateien hoch. Die besagte "typescript" und was nach dem Exit Befehl noch alles auf der shell erschienen ist als Logfile.

Ich bin noch "github neuling" und daher nicht sicher, welches Verzeichnis gemeint ist.
Daher habe ich die beiden Dateien einfach mal an diesen Post angehängt. Hoffe, dass das so korrekt ist.

Falls ich noch etwas weiter helfen kann, bitte lass es mich einfach wissen.

Beste Grüsse
Simon
[RaspiBackup Log und typescript.zip](https://github.com/framps/raspiBackup/files/6826603/RaspiBackup.Log.und.typescript.zip)

framps, 2021-07-16:
Vielen Dank fuer die Dateien. Leider sehe ich dort nicht was ich suche :cry: Ich habe aber eine modifizierte Version des Installers erstellt. Du findest ihn [hier](https://github.com/framps/raspiBackup/blob/issue345/installation/raspiBackupInstallUI.sh).  Als raw anzeigen und dann downloaden.

Bitte kopiere sie in Dein /home/simon Verzeichnis. Danach bitte folgende Befehle eingeben und die jeweilige Ausgabe hier zeigen. (Achte auf den fuehrenden Punkt da sonst die installierte Version aufgerufen wird)

0) `cd /home/simon`
1) `./raspiBackupInstallUI.sh`
2) `bash -x ./raspiBackupInstallUI.sh`
3) `zsh -x ./raspiBackupInstallUI.sh`

> Allerdings musste ich das Vorgehen etwas abändern, da ich die Kommandos auf diese Weise nicht verwenden konnte.

Stimmt. Mein Fehler :disappointed: Aber Du hast alles richtig gemacht :+1: 

> Ich bin noch "github neuling" und daher nicht sicher, welches Verzeichnis gemeint ist.
> Daher habe ich die beiden Dateien einfach mal an diesen Post angehängt. Hoffe, dass das so korrekt ist.

Alles richtig gemacht :wink: 

> Falls ich noch etwas weiter helfen kann, bitte lass es mich einfach wissen.

Vielen Dank fuer Deine Unterstuetzung.

miracuruzrh, 2021-07-16:
Besten Dank für die modifizierte Version.

Ich bin mir nicht 100% sicher, ob die Ausgabe wirklich mit der ersten Zeile beginnt.
Im Terminal (Terminal von mac OS) habe ich zwar die Buffergrösse erhöht, sehe aber keinen Unterschied.
Daher habe ich mal probiert die Ausgabe mit dem Anhang "... > log.txt" in eine Datei zu schreiben was aber leider nicht funktionierte. Oder einen zusätzlichen Eintrag im Script am Anfang um ein logfile zu erstellen hat leider auch nicht funktioniert. Hoffe aber, dass die Ausgabe komplett ist.

Kann aber sagen, dass alle 3 Varianten funktioniert haben :-).
Die Ausgaben von Punkt 2 und 3 habe ich im Anhang eingefügt.
[ScriptLog.zip](https://github.com/framps/raspiBackup/files/6832508/ScriptLog.zip)

framps, 2021-07-16:
> Ich bin mir nicht 100% sicher, ob die Ausgabe wirklich mit der ersten Zeile beginnt.

Leider nein :cry: . Es fehlt das fuer mich Wichtigste was am Anfang steht. Um die Ausgabe in eine Datei zu redirecten musst Du immer `&> log.txt` anhaengen. Kurze Erklaerung: Dadurch wird die Standardausgabe und die Standarerrorausgabe zusammen redirected. Di Standarderrorausgabe enthaelt die Debugausgabe die durch `-x` eingeschaltet wird :wink:

miracuruzrh, 2021-07-16:
Danke vielmals für die Erklärung.

> Heisst das das der Installer nun in allen drei Varianten auch erfolgreich durchlaeuft?
Ich muss leider nochmals korrigieren. Da ich nochmals getestet habe.
`sudo zsh -x ./raspiBackupInstallUI.sh` bringt nun Fehler. Evtl. hatte dies zuvor funktioniert, oder ich hatte mich irgendwie vertan.

Etwas komisch ist, dass Punkt 1 ja auch mit zsh aufgerufen wird und punkt 3 auch, nur mit dem Unterschied dass mit dem -x das logging aktiviert wird.

Um ganz sicher zu sein, dass die zsh die aktive shell ist, hier noch die Ausgabe einiger Überprüfungen:
`
 simon@DeskPi  ~  ps -p $$
  PID TTY          TIME CMD
10848 pts/0    00:00:01 zsh
 simon@DeskPi  ~  echo "$SHELL"
/usr/bin/zsh
 simon@DeskPi  ~  echo $0
-zsh
 simon@DeskPi  ~  readlink /proc/$$/exe
/usr/bin/zsh
 simon@DeskPi  ~  grep "^$USER" /etc/passwd
simon:x:1001:1001:Simon Boonstra,,,:/home/simon:/usr/bin/zsh
`

Hier noch eine Auflistung:
1. sudo ./raspiBackupInstallUI.sh -> Läuft durch -> Okay
2. sudo bash -x ./raspiBackupInstallUI.sh -> Läuft durch -> Okay
2a. sudo bash -x ./raspiBackupInstallUI.sh &> log.txt -> nicht 100% sicher. logfile wird generiert, bash hängt aber zum Schluss und bringe diese nicht mehr mit ctrl+c zum Abbruch. Aber da im Logfile "exit 0" steht, scheint dies i.O. zu sein. Ohne dem Anhang .. &> log.txt funktioniert es ja.
3. sudo zsh -x ./raspiBackupInstallUI.sh -> Fehler -> Siehe Logfile
[Logfiles.zip](https://github.com/framps/raspiBackup/files/6832906/Logfiles.zip)

Ich möchte aber  nicht unnötig Arbeit generieren. Die Installation ist für mich ja kein Problem.
Wenn es aber hilft, das Programm weiter zu Entwickeln bin ich natürlich sehr gerne dabei.

framps, 2021-07-16:
Vielen Dank. Jetzt kommen wir der Sache schon naeher :wink: 

Ich habe zwei Sachen geaendert:
1) `#!/usr/bin/env bash` wurde `#!/bin/bash`
2) `$BASH_VERSION` wird geprueft ob es gesetzt ist

Damit kann der Installer nur noch mit bash aufgerufen werden. zsh erzeugt eine Fehlermeldung dass bash fehlt (Siehe logfile)

Da 1) und 2) jetzt OK ist und 3) wie erwartet eine Fehlermeldung erzeugt sieht es fuer mich jetzt eigentlich ganz gut aus. Warum 2a) Probleme macht weiss ich nicht - ist aber auch nicht der Normalfall.

D.h. wenn ich diese kleine aenderung noch in die aktuelle 0.6.6.1 Beta reinbringe kann der Installer auch mit zsh aufgerufen werden :+1: 

> Ich möchte aber nicht unnötig Arbeit generieren.

Ich auch nicht denn eigentlich ist zsh nicht unterstuetzt. Aber ich versuche immer raspiBackup noch ein wenig zu verbessern und moegliche Fehler entweder durch Code abzufangen und Meldungen zu schreiben oder zu fixen. Eine eigene Umgebung mit zsh aufzusetzen waere fuer mich unnoetige Arbeit. Aber solange sich jemand wie Du findet ein paar Tests auf seinem System durchzufuehren um der Ursache auf den Grund zu gehen ist das OK und keine unnoetige Arbeit.

Ob jetzt `curl -sL https://raspibackup.linux-tips-and-tricks.de/install | sudo bash` funktioniert ist noch nicht ganz klar. Ich koennte fuer Dich eine geaenderte `install` erstellen so dass Du das bei Dir auch noch einmal final testen kannst. Ist das OK fuer Dich?

miracuruzrh, 2021-07-16:
Okay alles klar soweit.

Gerne teste ich eine geänderte `install`. Bitte einfach sagen, wenn es soweit ist, dann kann ich das mal probieren.
Ja denke auch, dass das zuviel des guten ist, eine eigene Umgebung für zsh aufzusetzen. Der nächste hat dann wieder eine andere shell usw. :-p.

framps, 2021-07-16:
> Aktuell bricht es noch ab

Auch in `install ` muss ich noch den shebang aendern :wink: . Dann sollte es eigentlich funktionieren.

framps, 2021-07-16:
Kannst  Du jetzt `curl -sL https://raspibackup.linux-tips-and-tricks.de/install345 | sudo bash` erfolgreich aufrufen?

miracuruzrh, 2021-07-16:
Sehr merkwürdig.
Mit dem Aufruf von:
`curl -sL https://raspibackup.linux-tips-and-tricks.de/install345 | sudo bash`
bricht das Script ab. Einen screenshot, wie das aussieht habe ich angehängt.

Wahrscheinlich ist der Befehl so nicht ganz korrekt. Ich probiere halt oft mit "trial & error".
Ob das so funktionieren kann, weiss ich daher nicht. Lustigerweise funktioniert so aber die Navigation. Das logxy.txt wird allerdings nicht befüllt.
`curl -sL https://raspibackup.linux-tips-and-tricks.de/install345 | sudo bash | &> logxy.txt`
[image]

framps, 2021-07-17:
> Lustigerweise funktioniert so aber die Navigation.

D.h. eigentlich funktioniert es soweit nur gibt es diese merkwuerdige Anzeige mit dem suspended tty output.

Probier doch mal `curl -sL https://raspibackup.linux-tips-and-tricks.de/install345 | sudo bash &>log.txt`. Dann sollte was in dem Log drinstehen

miracuruzrh, 2021-07-17:
Entschuldige die späte Antwort

Soweit so gut. Das Log wurde erstellt. Ob es hilft, wäre die andere Frage :-).
[log.txt](https://github.com/framps/raspiBackup/files/6835256/log.txt)

framps, 2021-07-17:
> Entschuldige die späte Antwort

Kein Problem. Wir haben hier ja kein Sev1 Problem :laughing: 

> Ob es hilft, wäre die andere Frage :-).

Hast Recht. Es fehlt ein -x :cry: Also `curl -sL https://raspibackup.linux-tips-and-tricks.de/install345 | sudo bash -x &>log.txt`

miracuruzrh, 2021-07-17:
Nun hat es funktioniert. Sorry habe ich auch gerade nicht gesehen.
Hier die Ausgabe von curl `-sL https://raspibackup.linux-tips-and-tricks.de/install345 | sudo bash -x`  im Anhang.
Aber wie gesagt, wenn ich mehr Arbeit generiere, als Dir geholfen ist, bitte einfach sagen. Für mich aber kein Problem, wenn es etwas nützt. Für mich ist das 'hobby' etc. :-)

[log.txt](https://github.com/framps/raspiBackup/files/6835272/log.txt)

framps, 2021-07-17:
Das ist fuer mich auch Hobby :wink: . Aber vielleicht sollten wir aber mal rekapitulieren. Bitte korrigiere mich sofern etwas falsch ist:

1) Der Aufruf mit zsh mit `curl -sL https://raspibackup.linux-tips-and-tricks.de/install345 | sudo bash` funktioniert funktional - also die Menus kommen und man kann raspiBackup installieren
2) Initial ist das Menu etwas korrupt. Im weiteren Verlauf ist aber alles wieder lesbar.

Sofern das alles stimmt denke ich lassen wir es so. Es ist nicht 100% OK - aber ausreichend. Wenn ich bei mir das nachvollziehe in einem zsh Window laeuft alles ohne Probleme und ohne korruptes Menu. Bei Dir scheint noch irgendwas Spezielles zu sein was den Fehler verursacht. Ich glaube es macht keinen Sinn da noch weiter Aufwand reinzustecken.

miracuruzrh, 2021-07-17:
Das ist fast korrekt.

1. `curl -sL https://raspibackup.linux-tips-and-tricks.de/install345 | sudo bash` generiert ein 'korruptes' Menu mit suspended jobs beim exit. Es wird dann aber die Datei /root/raspiBackupInstallUI345.sh generiert. Dann weiter mit folgenden Schritten:
2. `sudo -i su` (root werden und ins home Verzeichnis von root wechseln)
3. `chmod +x ./raspiBackupInstallUI345.sh`
4. `./raspiBackupInstallUI345.sh` -> Funktionsfäiges Menu und raspiBackup lässt sich installieren.

framps, 2021-07-19:
>     1. generiert ein 'korruptes' Menu mit suspended jobs beim exit.

D.h. man kann nicht normal installieren mit dem kleinen Menufehler? Und der exit ist forced vom Script und nicht ein exit vom Benutzer? Dann habe ich das voellig falsch verstanden. 

Wenn ich den curl in der zsh unter Benutzer pi in seinem Homeverzeichnis /home/pi ausfuehre wird ein `raspiBackupInstallUI345.sh` in /home/pi erstellt. Bist Du sicher dass bei Dir es in /root erstellt wird?

framps, 2021-07-26:
In der aktuellen Beta 0.6.6.1 habe ich jetzt sichergestellt dass alles Shell Script nur per bash aufgerufen werden koennen.

Ich denke damit kann der Issue von Dir geclosed weren :wink:

framps, 2021-08-10:
Closed because of inactivity

closed 2021-08-10 by framps

miracuruzrh, 2021-09-01:
Entschuldigung, dass ich mich nicht mehr gemeldet hatte.
Da waren noch Ferien und weitere Verpflichtungen dazwischen.
Denke auch, dass das geschlossen bleiben kann. Is tja jetzt ein Beta fix vorhanden.
Vielen Dank

framps, 2021-09-01:
> Entschuldigung, dass ich mich nicht mehr gemeldet hatte.

Kein Problem. Ich schliesse Issues immer nach einer gewissen Zeit. Man kann einen Issue ja auch wieder oeffnen :wink: