framps/raspiBackup #347: Executing raspiBackupRestore2Image.sh on different machine

vguna, 2021-07-31:
```
# raspiBackup.sh --version
Version: 0.6.6 CommitSHA: 9df3c1e CommitDate: 2021-07-21 CommitTime: 19:47:04
```
```
raspiBackupInstallUI.sh 0.4.3.5, 2021-07-21/19:47:04 - 9df3c1e
```
```
# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

Hey,
I'm wondering whether `raspiBackupRestore2Image.sh` can also be execute without any sideeffects on a different machine which has the backups locally stored.

I tried to restore such an created image via Etcher, but it didn't boot. I also skipped the pishrink part - to isolate the issue but still no luck.

My setup is, that I run raspiBackup on my Pi and let it backup my SSD (no sdcard) with the rsync option to a mounted NFS share from my NAS. For this the Pi wakes the NAS, performs the backup and the NAS goes to sleep after some minutes of inactivity.

To speedup the restore process later on, I would also like to create a DD image from the rsync backup to have it easier to write the image back to the SSD once an issue occurs from my Windows machine - without having to setup a RaspianOS or similar first. To not waste resources to create the DD image every time, the creation is postponed to a different point in time, when the NAS is maybe used for other things anyway. So it checks whether an DD image exists for the latest backup, if not, it will create one on-demand using `raspiBackupRestore2Image.sh`.

As mentioned, this does not seem to work for me - since the written back image is not booting from my SSD. Now I'm wondering if there's anything special, that would not allow me to create the DD image from the NAS instead of the Pi - where the actual system resides on. I also made a "normal" install of raspiBackup on the NAS with the needed configs via `raspiBackupInstallUI.sh` to match the one from the Pi (the backup dir is of course different).

I checked the output of the image creation, but I couldn't spot any error besides the `Re-reading the partition table failed.: Invalid argument` message - not sure if that is an issue.

```
root@nas:/usr/local/bin# raspiBackupRestore2Image.sh /store1/backup/Openhab/openHABianPi/openHABianPi-rsync-backup-20210726-200018
raspiBackupRestore2Image.sh v0.1.6, 2021-05-20/21:07:10 - 70652dd
===> Backup source disk size: 114471 (MiB)
0+0 records in
0+0 records out
0 bytes copied, 0.000563674 s, 0.0 kB/s
Checking that no-one is using this disk right now ... OK

Disk /dev/loop0: 111.8 GiB, 120031543296 bytes, 234436608 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Created a new DOS disklabel with disk identifier 0x7a291e5f.
/dev/loop0p1: Created a new partition 1 of type 'W95 FAT32 (LBA)' and of size 41.8 MiB.
/dev/loop0p2: Created a new partition 2 of type 'Linux' and of size 111.8 GiB.
/dev/loop0p3: Done.

New situation:
Disklabel type: dos
Disk identifier: 0x7a291e5f

Device       Boot Start       End   Sectors   Size Id Type
/dev/loop0p1       8192     93802     85611  41.8M  c W95 FAT32 (LBA)
/dev/loop0p2      98304 234436607 234338304 111.8G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
Re-reading the partition table failed.: Invalid argument
The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or kpartx(8).
Syncing disks.
===> Reloading new partition table
===> Restoring backup into /store1/backup/Openhab/openHABianPi/openHABianPi-rsync-backup-20210726-200018/openHABianPi-backup.dd
--- RBK0009I: nas: raspiBackup.sh V0.6.6 (9df3c1e) started at Thu Jul 29 00:17:58 CEST 2021.
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
--- RBK0168I: :-D raspiBackup.sh beta version 0.6.6.1 is available. Any help to test this beta is appreciated. Just upgrade to the new beta version with option -U. Restore to the previous version with option -V
--- RBK0114I: Visit https://www.linux-tips-and-tricks.de/en/versionhistory/ to read about the changes in the new version.
!!! RBK0143W: Target /dev/loop0 does not match with backup. Partitioning forced.
!!! RBK0065W: Device /dev/loop0 will be repartitioned and all data will be lost.
--- RBK0067I: Current partitions on /dev/loop0:
Number  Start   End       Size      Type     File system  Flags
 1      4.19MB  48.0MB    43.8MB    primary               lba
 2      50.3MB  120032MB  119981MB  primary
--- RBK0066I: Device /dev/loop0 will be overwritten with the saved boot and root partition.
--- RBK0069I: Bootpartition /dev/loop0p1 will be formatted and will get the restored Boot partition.
--- RBK0070I: Rootpartition /dev/loop0p2 will be formatted and will get the restored Root partition.
--- RBK0038I: Are you sure? y/N
--- RBK0050I: Restoring backup from /store1/backup/Openhab/openHABianPi/openHABianPi-rsync-backup-20210726-200018.
!!! RBK0139W: Partition creation ignores errors.
--- RBK0053I: Restoring first partition (boot partition) to /dev/loop0p1.
--- RBK0055I: Restoring second partition (root partition) to /dev/loop0p2.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0010I: openHABianPi: raspiBackup.sh V0.6.6 (9df3c1e) stopped at Thu Jul 29 00:23:59 CEST 2021 with rc 0.
--- RBK0076I: Restore finished successfully.
--- Cleaning up
```

framps, 2021-07-31:
I wrote this helper script when USB boot mode was not available. This script is provided as is and is not part of the official shipment and thus there is no regression test executed - neither for sd card only nor usb boot.

As of now I don't see any reason why the created dd image doesn't boot. Do you see any messages during boot which ask for any partition which is not found? Can you provide a screenshot of the boot messages? Usually there are UUID/PARTUUID mismatches if the system doesn't boot.

Could you provide the debug log of raspiBackup ?

vguna, 2021-08-01:
Hey, thanks for taking a look at this.

The thing is, that I don't see anything when powering on the rpi with the restored image on the SSD. Black screen. To doublecheck I also used Etcher then to write a plain vanilla RaspbianOS to the SSD. With this it boots perfectly fine. So maybe something is wrong with the partitions or so, that it dies very early.

I just created a new backup - see the log below if you can find anything fishy.

```
20210801-110345 DBG 1721:      --> initializeDefaultConfigVariables 
20210801-110345 DBG 1855:      <-- initializeDefaultConfigVariables 
20210801-110345 DBG 1860:      --> copyDefaultConfigVariables 
20210801-110345 DBG 1960:          --> substituteNumberArguments 
20210801-110345 DBG 1987:          <-- substituteNumberArguments 
20210801-110345 DBG 1931:      <-- copyDefaultConfigVariables 
20210801-110345 DBG 2702:      --> readConfigParameters 
20210801-110345 DBG 2758:      <-- readConfigParameters 
20210801-110345 DBG 1860:      --> copyDefaultConfigVariables 
20210801-110345 DBG 1960:          --> substituteNumberArguments 
20210801-110345 DBG 1987:          <-- substituteNumberArguments 
20210801-110345 DBG 1931:      <-- copyDefaultConfigVariables 
20210801-110345 DBG 1642:      --> logOptions Standard option files
20210801-110345 DBG 1644:      --- Options: 
20210801-110345 DBG 1645:      --- APPEND_LOG=0
20210801-110345 DBG 1646:      --- APPEND_LOG_OPTION=-a
20210801-110345 DBG 1647:      --- BACKUPPATH=/backup
20210801-110345 DBG 1648:      --- BACKUPTYPE=rsync
20210801-110345 DBG 1649:      --- AFTER_STARTSERVICES=
20210801-110345 DBG 1650:      --- BEFORE_STOPSERVICES=
20210801-110345 DBG 1651:      --- CHECK_FOR_BAD_BLOCKS=0
20210801-110345 DBG 1652:      --- COLORING=CM
20210801-110345 DBG 1653:      --- CONFIG_FILE=
20210801-110345 DBG 1654:      --- DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20210801-110345 DBG 1655:      --- DD_BLOCKSIZE=1M
20210801-110345 DBG 1656:      --- DD_PARMS=
20210801-110345 DBG 1657:      --- DD_WARNING=0
20210801-110345 DBG 1658:      --- DEPLOYMENT_HOSTS=
20210801-110345 DBG 1659:      --- DYNAMIC_MOUNT=
20210801-110345 DBG 1660:      --- EMAIL=v@@@@e(16)
20210801-110345 DBG 1661:      --- EMAIL_COLORING=SUBJECT
20210801-110345 DBG 1662:      --- EMAIL_PARMS=
20210801-110345 DBG 1663:      --- EXCLUDE_LIST=
20210801-110345 DBG 1664:      --- EXTENSIONS=
20210801-110345 DBG 1665:      --- FAKE=
20210801-110345 DBG 1666:      --- HANDLE_DEPRECATED=
20210801-110345 DBG 1667:      --- IGNORE_ADDITIONAL_PARTITIONS=0
20210801-110345 DBG 1668:      --- KEEPBACKUPS=3
20210801-110345 DBG 1669:      --- KEEPBACKUPS_DD=0
20210801-110345 DBG 1670:      --- KEEPBACKUPS_DDZ=0
20210801-110345 DBG 1671:      --- KEEPBACKUPS_TAR=0
20210801-110346 DBG 1672:      --- KEEPBACKUPS_TGZ=0
20210801-110346 DBG 1673:      --- KEEPBACKUPS_RSYNC=0
20210801-110346 DBG 1674:      --- LANGUAGE=EN
20210801-110346 DBG 1675:      --- LINK_BOOTPARTITIONFILES=0
20210801-110346 DBG 1676:      --- LOG_LEVEL=1
20210801-110346 DBG 1677:      --- LOG_OUTPUT=2
20210801-110346 DBG 1678:      --- MAIL_ON_ERROR_ONLY=0
20210801-110346 DBG 1679:      --- MAIL_PROGRAM=mail
20210801-110346 DBG 1680:      --- MSG_LEVEL=0
20210801-110346 DBG 1681:      --- NOTIFY_START=0
20210801-110346 DBG 1682:      --- NOTIFY_UPDATE=1
20210801-110346 DBG 1683:      --- PARTITIONBASED_BACKUP=0
20210801-110346 DBG 1684:      --- PARTITIONS_TO_BACKUP=1 2
20210801-110346 DBG 1685:      --- RESIZE_ROOTFS=1
20210801-110346 DBG 1686:      --- RESTORE_DEVICE=
20210801-110346 DBG 1687:      --- ROOT_PARTITION=
20210801-110346 DBG 1688:      --- RSYNC_BACKUP_ADDITIONAL_OPTIONS=
20210801-110346 DBG 1689:      --- RSYNC_BACKUP_OPTIONS=-aHx --delete
20210801-110346 DBG 1690:      --- RSYNC_IGNORE_ERRORS=
20210801-110346 DBG 1691:      --- SENDER_EMAIL=
20210801-110346 DBG 1692:      --- SKIP_DEPRECATED=
20210801-110346 DBG 1693:      --- SKIPLOCALCHECK=0
[... 858 lines cut ...]
20210801-111112 DBG 3198:          --- coloring: 
20210801-111112 DBG 3027:              --> colorAnnotation 0
20210801-111112 DBG 3051:              <-- colorAnnotation 
20210801-111112 DBG 3221:          --- eMail: v@@@@e(16)
20210801-111112 DBG 3222:          --- eMail Program: mail
20210801-111112 DBG 3223:          --- Subject: @HOSTNAME@: Backup finished successfully.
20210801-111112 DBG 3223:          --- MIME-Version: 1.0
20210801-111112 DBG 3223:          --- Content-Type: text/html; charset=utf-8
20210801-111112 DBG 3224:          --- ColoringOption: 
20210801-111112 DBG 3225:          --- ContentType: 
20210801-111112 DBG 3225:          --- MIME-Version: 1.0
20210801-111112 DBG 3225:          --- Content-Type: text/html; charset=utf-8
20210801-111112 DBG 3226:          --- Parms: 
20210801-111112 DBG 3231:          --- mail  -s "@HOSTNAME@: Backup finished successfully.
20210801-111112 DBG 3231:          --- MIME-Version: 1.0
20210801-111112 DBG 3231:          --- Content-Type: text/html; charset=utf-8" v@@@@e(16)
20210801-111112 DBG 3234:          --- mail: RC: 0
20210801-111112 DBG 3277:          <-- sendEMail 
20210801-111112 DBG 3386:      --- Masquerading eMail
20210801-111112 DBG 3409:      --- Masquerading some mount options
20210801-111112 DBG 3427:      --- Masquerading home directory name
20210801-111112 DBG 3432:      --- Masquerading hostname
20210801-111112 DBG 3437:      --- Masquerading sensitive non local IPs
20210801-111113 DBG 3599:      <-- cleanup 
```

framps, 2021-08-02:
Thank you very much for the log.

Last night I tried to reproduce your issue:
1) I created a rsync backup of an USB boot only Buster10 on my Raspi3 runnign on an USB disk with raspiBackup 
2) I restored the image with raspiBackuprestore2Image into a dd file and jused the code from github with pishrink at the end.
3) I wrote the generated image with dd on another USB disk
4) The system booted without any issue

I did this just to verify the whole process works also with an USB boot only system. As I wrote I tested the process with an SD os system. Now I verfiied the process also works with an USB only system.

Actually I don't have a second Raspi3 handy so I wasn't able to use a different HW to generated the dd form the rsync backup. But I dont't think this makes a difference. 

I checked your log and didn't find anything special. Actually I used a plain RaspiOS and you used a system with OpeneHAB installed. But I know a lot of people use raspiBackup with OpenHAB and have no issue.

Whenever I got an issue the restored system doesn't boot and just shows a blank screen as soon as another SD card was used the issue was gone. If you restore a dd image to a broken or buggy SD card you get this kind of boot issue. You restored the dd image to an USB disk. You also installed a fresh RaspianOS on the USB disk which boots up fine and we can assume the USB disk is OK. Maybe you have an USB flash drive handy you can try to restore the image? 

Would you please provide the output of following command on the created dd image? Maybe I get some indication there what's wrong
```
 sudo fdisk -lu yourDDImage.dd
```

vguna, 2021-08-02:
Hey, thanks again for digging into it. Sadly, I do not own a 120GB sd card. But I can try to use the pishrink image. If I got it correctly, it should resize it then to the actual media size when booting up, correct? So I can give it a try with a 32GB card - that should easily fit the actual 8 GB system. But I guess then maybe some configs etc. won't fit since it's a mmcblkx dev and not sdx?

Meanwhile here is the output of the command from the NAS - where the image is remotely stored on.

```
root@nas:/store1/backup/Openhab/openHABianPi# fdisk -lu openHABianPi-rsync-backup-20210802-050029.img
Disk openHABianPi-rsync-backup-20210802-050029.img: 111.8 GiB, 120031543296 bytes, 234436608 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x7a291e5f

Device                                         Boot Start       End   Sectors   Size Id Type
openHABianPi-rsync-backup-20210802-050029.img1       8192     93802     85611  41.8M  c W95 FAT32 (LBA)
openHABianPi-rsync-backup-20210802-050029.img2      98304 234436607 234338304 111.8G 83 Linux
```

vguna, 2021-08-02:
I just wrote the pishrinked image via Etcher to the SD-Card -> same issue. Black screen when powering on 🤔

framps, 2021-08-03:
> But I guess then maybe some configs etc. won't fit since it's a mmcblkx dev and not sdx?

No config change is required.

> same issue.

:cry: 

You dd image looks good. Have you tried to restore the rsync backup and does it boot? Maybe there is an issue already with the rsync backup?

vguna, 2021-08-03:
Just tried the suggested rsync restore via raspiBackup to the SSD which also fails. See log below. But I have to admit, I restored from the NAS directly - although you warn about the potential issues with incompatible fdisk versions. But both systems are buster - so 🤞 that it's not the issue (?). Looking at the debug log, it complains with the following before it dies:

```
/dev/sdh2 is apparently in use by the system; will not make a filesystem here!
```

I really wiped the whole disk a couple of times with cfdisk and reconnected. It is not mounted or such. Wondering why it thinks it's in use.

```
root@nas:/# raspiBackup.sh -d /dev/sdh /store1/backup/Openhab/openHABianPi/openHABianPi-rsync-backup-20210802-050029
--- RBK0009I: nas: raspiBackup.sh V0.6.6 (9df3c1e) started at Tue Aug  3 21:39:22 CEST 2021.
!!! RBK0018W: Target /dev/sdh with 111.79 GiB is larger than backup source with 111.78 GiB. root partition will be expanded accordingly to use the whole space.
!!! RBK0065W: Device /dev/sdh will be repartitioned and all data will be lost.
--- RBK0146I: No partitiontable found on /dev/sdh.
--- RBK0066I: Device /dev/sdh will be overwritten with the saved boot and root partition.
--- RBK0069I: Bootpartition /dev/sdh1 will be formatted and will get the restored Boot partition.
--- RBK0070I: Rootpartition /dev/sdh2 will be formatted and will get the restored Root partition.
--- RBK0038I: Are you sure? y/N
y
--- RBK0050I: Restoring backup from /store1/backup/Openhab/openHABianPi/openHABianPi-rsync-backup-20210802-050029.
--- RBK0004I: Adjusting second partition from 111.74 GiB to 111.75 GiB.
--- RBK0053I: Restoring first partition (boot partition) to /dev/sdh1.
??? RBK0185E: Rootpartition creation failed with RC 1.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0010I: openHABianPi: raspiBackup.sh V0.6.6 (9df3c1e) stopped at Tue Aug  3 21:42:28 CEST 2021 with rc 117.
??? RBK0077E: Restore failed. Check previous error messages.
```

Here is also the [debug log](https://github.com/framps/raspiBackup/files/6926837/raspiBackup.log)

vguna, 2021-08-15:
Anything else I could test/check?

TotalCoolness, 2021-08-23:
Same issues. Nothing Raspibackup creates from a booted SSD works.  I tried dd & tar (cant use rsync)

Its not my hardware as I used imgutils and it can create a booting SSD perfectly fine.

framps, 2021-09-08:
@vguna I just noticed there is a different issue #358 (in German) which addresses exactly your issue. Please grab the latest raspiBackuprestore2Image.sh v1.7 from master and try again.

vguna, 2021-09-08:
Cool, let me give it a try, thanks!

vguna, 2021-09-08:
Just tried it with the same result.

But meanwhile I think the issue is a different one. When I look at my openhab, it does not seem to contain any `bootcode.bin` or `start.elf` files on `/boot` which I think are required to boot from a rpi3. The origin system is a rpi4 one.
I just put the files from https://github.com/raspberrypi/firmware/tree/master/boot there, but it does not help. The ACT led blinks 4 times - which means `start.elf` could not be loaded. But no clue what is the issue with it :/. I will keep digging - maybe I get somehow serial debugging working. Since all I see is a black screen.

framps, 2021-09-08:
:cry: 

> The origin system is a rpi4 one.

Just to make sure I get you right: You create a backup on a Raspi3 and compress it with raspiBackupRestore2Image on a Raspi4?ted

> Since all I see is a black screen.

You're the first one who reports a blank screen :cry: . 

Please note: When I tested raspiBackupRestore2Image initially and used the same HW and OS version. Boot worked because the migration was executed on the same system. 

Now because of #358 I tested to convert a Raspi3 tar backup into a dd backup on a Raspi4 (a different system) which now booted successfully with 1.7. So I expected your image to also boot successfully. Looks like you have some slightly different environment.

vguna, 2021-09-08:
The origin system is running on a rpi4. The backup is created on this but written to a NFS share on a NAS. The NAS creates an image from the backup using `raspiBackupRestore2Image` at a later time. The image is restored via Windows to a SSD (same model than the origin system, incl. same SATA2USB adapter). Then I try to boot from the SSD using a rpi3b directly. If I install a fresh raspbianOs using the rpi3b and the SSD it boots without any issue.

So in the end I guess It's more an issue of trying to run a rpi4 system on a rpi3 than raspiBackup itself. I just created a thread [here ](https://raspberrypi.stackexchange.com/questions/132114/run-a-raspbianos-system-for-rpi4-on-rpi3)to get some more insights. I will report back.

framps, 2021-09-08:
Ok. So it's the other way around and primarily the question whether a cloned Raspi4 image onto a Raspi3 (that happens when you restore a Raspi4 backup to a Raspi3) is suggested to boot. Maybe https://www.raspberrypi.org/forums/ is another place to ask this question? But please beware of multipostings people are sensitive of :wink:

vguna, 2021-10-04:
Finally it's booting. Sorry for the delay, but other things kept me busy. After a sunday of serial debugging of my rpi3 I finally found the issue.

It's not raspiBackup - as expected.

Since the image was originally running on a rpi4 it got 2 issues:
- `config.txt` contained `gpu_mem=16` which the rpi3 does not seem to like
- Missing files that the rpi3 needs in /boot like `kernel.img`, `start.elf` etc. Also if you use a new kernel from the raspberry firmware github page, make sure that you also copy all other files like overlays etc. Otherwise you are in trouble.

So making those modifications, let the rpi3 finally boot from the SSD - yay!

One thing left I may ask: pishrink made a 6GB file out of 120 GB drive. But after the image boots, it does not resize back to 120GB but stays at 6GB. Is this expected?

And sorry for all the trouble!

vguna, 2021-10-04:
OK it seems that could be an issue: https://github.com/Drewsif/PiShrink/blob/43f1a885cd147638a0d1961753f711f838915b88/pishrink.sh#L95

framps, 2021-10-04:
> Sorry for the delay, but other things kept me busy.

No problem. Glad you keep me updated.

> Since the image was originally running on a rpi4 it got 2 issues

As far as I see there was a big design change between Raspi3 and Raspi4 in RaspiOS and this causes the restored system to fail during boot.

Unfortunately there is no metadata created by raspiBackup during backup to pass any system information from the backup to the restore to check for HW incompatibilities.

> OK it seems that could be an issue:

Good catch :+1:  Looks like this is a pishrink feature :wink: