framps/raspiBackup #348: raspiBackupWrapper.sh implementation

MrR341-GER, 2021-07-31:
something more here to say.. first of all:
Thanks for you'r answer on youtube.

1. I deactivated "Regular Backup" with "sudo raspiBackupInstallUI.sh"
2. I got my 256GB USB-Stick and successfully testet "raspiBackupWrapper.sh" after I rewrote the content of it.. (set my own mounting point)

No logs, becouse nothing special to see here...

Now is the big question:
Will raspiBackupWrapper.sh hold to the regular raspiBackup config I edited in the UI?

If not:
Is there any way to implement raspiBackupWrapper.sh IN the raspiBackup.sh?
So it's fully handeled by the regular raspiBackup config including:
- deactivating and reactivating services
- special backup strategy (keeping backups as defined in UI)
- zipping (taring?) the image for better file size INCLUDING the shrinking of the .img (pishrink)
- and last but not least: (and you told to manually configure raspiBackupWrapper in crontab) that raspiBackupWrapper will automaticly included to crontab or however you managed the automatic execution as you defined with the UI

I just activated the packing in the UI and check again if this works... if so, I never should have any problems with missing space left, since the dd backup will generate a 128GB file first and then shrinking it... I have to manage it anyhow with 256GB in total..

Have a nice day :)

framps, 2021-07-31:
> Now is the big question:
> Will raspiBackupWrapper.sh hold to the regular raspiBackup config I edited in the UI?

Yes. raspiBackup will be called and will use the config file. I also added following comment in the code
```
# ===> configure all options in /usr/local/etc/raspiBackup.conf. Don't forget -a and -o options
```
If you want to overwrite some options use one of the various options documented [her](https://www.linux-tips-and-tricks.de/en/backup/#parameters).

> Is there any way to implement raspiBackupWrapper.sh IN the raspiBackup.sh?

raspiBackup just creates and restores a backup.  All code in the helper directory is sample code which should help to extend any user raspiBackup capabilities. 

> So it's fully handeled by the regular raspiBackup config

In the wrapper script raspiBackup is called and all stuff you mention is done (service activation/deactivation, .. backup strategy, ...). The crontab stuff has to be done manually because as I wrote before - raspiBackupWrapper is just sample code and you have to add the cron invocation.

> since the dd backup will generate a 128GB file first and then shrinking it.

Have you read [FAQ16](https://www.linux-tips-and-tricks.de/en/faq/#a16) which is an alternate way to reduce the dd image size?

MrR341-GER, 2021-07-31:
> Have you read FAQ16 which is an alternate way to reduce the dd image size?
I did, but I have only two partitions which I cant resize (boot and root)
> In the wrapper script raspiBackup is called and all stuff you mention is done (service activation/deactivation, .. backup strategy, > ...). The crontab stuff has to be done manually because as I wrote before - raspiBackupWrapper is just sample code and you > > have to add the cron invocation.
Okay thx anyway :)

I added some code in the wrapper:
`
function zipping() {
	if which tar 2>&1 1>/dev/null; then
		readVars
		sudo tar cJf - $BACKUP_TARGETFILE -P | pv -s $(du -sb $BACKUP_TARGETFILE | awk '{print $1}') | gzip > "$BACKUP_TARGETFILE.txz"
		rc=$?
		if (( $rc == 0 )); then
			echo "zipping succeeded :-)"					# do whatever has to be done in case of success
		else
			echo "zipping failed with rc $rc :-("			# do whatever has to be done in case of backup failure
			exit $rc
		fi
	else
		echo "tar not found :-("
	fi
}

function pishrink() {
	if which pishrink.sh 2>&1 1>/dev/null; then
		readVars
		sudo pishrink.sh $BACKUP_TARGETFILE
		rc=$?

		if (( $rc == 0 )); then
			echo "pishrink succeeded :-)"					# do whatever has to be done in case of success
			zipping
		else
			echo "pishrink failed with rc $rc :-("			# do whatever has to be done in case of backup failure
			exit $rc
		fi
	else
		echo "pishrink not found :-("
		exit 42
	fi
}
`
But there is one Problem left:
The zipping is taking AGES and I dont know how to resolve that problem yet... maybe there is any parameter to use more cores or something

Do you have any reccommendations?

framps, 2021-08-01:
> I did, but I have only two partitions which I cant resize (boot and root)

What's the issue to resize the second partition to for example 20G? The according FAQ16 your dd image will use only 20G :wink: 

> The zipping is taking AGES and I dont know how to resolve that problem yet..

I added code in pishrink which enables to use multiple cores (option -a).

MrR341-GER, 2021-08-01:
> > I did, but I have only two partitions which I cant resize (boot and root)
> 
> What's the issue to resize the second partition to for example 20G? The according FAQ16 your dd image will use only 20G 😉 Just use gparted to resize the partition.
> 
> > The zipping is taking AGES and I dont know how to resolve that problem yet..
> 
> I added code in pishrink which enables to use multiple cores (option -a).

The problem is, that I cant resize my partition... IDK why... not with gparted, not if I remove the SD-Card and try to resize it with an other OS/Distribution
In gparted I have a small lock sign (not a key sign), and in other OS/Distros its simply not possible to resize the partition...
Like there is a flag (there isn't.. at least not in gparted) which tells any partition tool, that this partition size is fixed and can't be changed.... All I read it's "becouse it's mounted" but this didn't to seem that this is the problem here becouse it's not working with other OS either...

Another automated "solution" to the size problem in my head was something like:
1. Shrink the partition to the smallest size possible (maybe let a few MB left for safety)
2. make the dd Backup
3. zip the dd Backup
4. inflate the partition again
wich would also meke the process even faster
but I don't really have any Idea how to manage that... in specially becouse I can't resize anything...

me too... I added something else in my zipping function in the wrapper:
```bash
sudo tar -c -I 'xz -9 -T0' -f $BACKUP_TARGETFILE$TAR_EXT $BACKUP_TARGETFILE
```

WAIIIITwaitwaitwaitwait...

Do I see that correctly that with pishrink -a the img gets compressed, too and I dont need to do an extra tar command? O.o
Btw. Where did you add it? :D I cant see any new commits or did you mean local?
after all I read I would do:
```bash
sudo pishrink -Za $BACKUP_TARGETFILE
```

framps, 2021-08-02:
> Do I see that correctly that with pishrink -a the img gets compressed, too and I dont need to do an extra tar command? O.o

yes

> Btw. Where did you add it? :D I cant see any new commits

It's > 2 year ago. See [here](https://github.com/Drewsif/PiShrink/commit/ab320089812913f2a5ff368dae02c411396b9321)

framps, 2021-09-08:
Closing because of inactivity

closed 2021-09-08 by framps