framps/raspiBackup #351: No access to /media/pi after restore

Marian1GitHub, 2021-08-02:
After restore on SD card with default settings I'm not able to access USB drives if I boot up the restored SD card.
I have no rights to `/media/pi`, my current workaround is to remove the `\media\pi` folder with `sudo rmdir pi`
and reboot, but I don't know if something else goes also wrong which I not noticed yet.

New Image:
 `ls -la /media` show up a + sign with some extended rights.
`drwxr-x---+  4 root root 4096 Aug  2 15:32 pi`
Backup storage:
 `ls -la /media` show also a + sign with some extended rights.
`drwxr-x---+  4 root root 4096 Aug  2 15:32 pi`
Restored SD card:
 `ls -la /media` the + sign is gone.
`drwxr-x---  4 root root 4096 Aug  2 15:32 pi`

New Image / Backup storage :
```
/media $ getfacl -e pi
# file: pi
# owner: root
# group: root
user::rwx
user:pi:r-x                     #effective:r-x
group::---                      #effective:---
mask::r-x
other::---
```

Restored SD card:
```
/media $ getfacl -e pi
# file: pi
# owner: root
# group: root
user::rwx
group::r-x
other::---
```

[image]

framps, 2021-08-02:
Looks like the ACL is not restored somehow :worried: 

> workaround is to remove the `\media\pi` folder with `sudo rmdir pi`
> and reboot

That's interesting. Do you have now the original ACLs again?

Would you please attach the debug log of a backup run?

Marian1GitHub, 2021-08-02:
Yes after delete of the folder I got all rights back.
I have searched for a solution and read somewhere that the system create the folder again if it is gone.

[raspiBackup.log](https://github.com/framps/raspiBackup/files/6919479/raspiBackup.log)

I have also tried a run without any mounted drive in /media/pi but the same happens.

framps, 2021-08-03:
According the log the ACL is already missing in the backup.

```
20210802-103335 DBG 3287:              --> cleanupBackupDirectory 
20210802-103335 DBG 3289:                  *** ls -la /mnt/pidrive2/backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20210802-103304
20210802-103335 DBG 3289:                      total 262300
20210802-103335 DBG 3289:                      drwxr-xr-x  18 root root      4096 Jul 31 17:59 .
20210802-103335 DBG 3289:                      drwxr-xr-x   7 root root      4096 Aug  2 10:33 ..
20210802-103335 DBG 3289:                      -rw-r--r--   1 root root 268435456 Aug  2 10:33 @HOSTNAME@-backup.img
20210802-103335 DBG 3289:                      -rw-r--r--   1 root root       512 Aug  2 10:33 @HOSTNAME@-backup.mbr
20210802-103335 DBG 3289:                      -rw-r--r--   1 root root       195 Aug  2 10:33 @HOSTNAME@-backup.sfdisk
20210802-103335 DBG 3289:                      lrwxrwxrwx   5 root root         7 Mai  7 16:39 bin -> usr/bin
20210802-103335 DBG 3289:                      drwxr-xr-x   2 root root      4096 Jan  1  1970 boot
20210802-103335 DBG 3289:                      drwxr-xr-x   2 root root      4096 Aug  1 14:02 dev
20210802-103335 DBG 3289:                      drwxr-xr-x 114 root root      4096 Jul 31 23:58 etc
20210802-103335 DBG 3289:                      drwxr-xr-x   3 root root      4096 Mai  7 16:42 home
20210802-103335 DBG 3289:                      lrwxrwxrwx   5 root root         7 Mai  7 16:39 lib -> usr/lib
20210802-103335 DBG 3289:                      drwx------   2 root root      4096 Mai  7 17:06 lost+found
20210802-103335 DBG 3289:                      drwxr-xr-x   3 root root      4096 Jul 31 18:03 media
20210802-103335 DBG 3289:                      drwxr-xr-x   3 root root      4096 Jul 31 12:14 mnt
20210802-103335 DBG 3289:                      drwxr-xr-x   4 root root      4096 Mai  7 16:49 opt
20210802-103335 DBG 3289:                      dr-xr-xr-x   2 root root      4096 Jan  1  1970 proc
20210802-103335 DBG 3289:                      -rw-r--r--   1 root root     71177 Aug  2 10:33 raspiBackup.log
20210802-103335 DBG 3289:                      -rw-r--r--   1 root root      1003 Aug  2 10:33 raspiBackup.msg
20210802-103335 DBG 3289:                      drwx------   5 root root      4096 Jul 31 22:27 root
20210802-103335 DBG 3289:                      drwxr-xr-x   2 root root      4096 Aug  2 00:26 run
20210802-103335 DBG 3289:                      lrwxrwxrwx   5 root root         8 Mai  7 16:39 sbin -> usr/sbin
20210802-103335 DBG 3289:                      drwxr-xr-x   2 root root      4096 Mai  7 16:39 srv
20210802-103335 DBG 3289:                      dr-xr-xr-x   2 root root      4096 Jan  1  1970 sys
20210802-103335 DBG 3289:                      drwxrwxrwt   2 root root      4096 Aug  2 10:33 tmp
20210802-103335 DBG 3289:                      drwxr-xr-x  11 root root      4096 Mai  7 16:48 usr
20210802-103335 DBG 3289:                      drwxr-xr-x  11 root root      4096 Mai  7 17:07 var
```
but you detected ACLs.

> Backup storage:
> `ls -la /media` show also a + sign with some extended rights.
> `drwxr-x---+ 4 root root 4096 Aug 2 15:32 pi`

Do you have any idea why you detected ACL on `/media` in the backup but the log doesn't show them up?

rsync is enabled to backup ACLs (`-A` in (`20210802-103305 DBG 1689:      --- RSYNC_BACKUP_OPTIONS=-aHAx --delete`) . So I don't have an idea why they are missing in the backup.

Actually I also wonder where the ACL comes from in `/media`. I also have Raspberries running Buster 10 and I don't have any ACLs  on `/media`.

Marian1GitHub, 2021-08-03:
On `/media` I have also no ALCs, but the subdirectory `/media/pi` they are applied

framps, 2021-08-03:
> Can you at least reproduce it with the current recommended RaspbianOS 32Bit, created via imager.

I need to know exactly how you mount the two partitions and what's on the two partitions. According the debug log it's mounted
```
20210802-103308 DBG 5056:              /dev/sda2 on /media/pi/rootfs type ext4 (rw,nosuid,nodev,relatime,uhelper=udisks2)
20210802-103308 DBG 5056:              /dev/sda1 on /media/pi/boot type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
```
and seems to be some RasbianOS stored on an USB disk or stick.

Marian1GitHub, 2021-08-03:
Here is my setup:

`/dev/mmcblk0p1 /boot`
`/dev/mmcblk0p2 /rootfs`
RaspianOS 32Bit installed via Imager directly on SD Card
Productive System to be restored

`/dev/sda1      /media/pi/boot`   (At time of backup creation automatic mounted in Productive System)
`/dev/sda2      /media/pi/rootfs`  (At time of backup creation automatic mounted in Productive System)
RaspianOS 32Bit installed via Imager directly on SSD + gparted + raspiBackup + Imager
Backup System to restore Productive System Backups from `/dev/sda3` to SD Card
Will boot if SD Card is not pluged in Pi

`/dev/sda3      /mnt/pidrive2`
Data and Backup ext4 partition created via gparted directly after install from Imager and
before first boot of Backup system because of automatic rootfs expansion from new install

The folloing steps I perform if I want to restore the Productive System to SD Card from the Backup:
• Boot Backup System (Boot from USB without SD Card)
• SD Card to be resorted will be connected as `/dev/sdb` via Card Reader
• `sudo umount /dev/sdb1`
• `sudo umount /dev/sdb2` (if exists)
• `sudo raspiBackup -d /dev/sdb /mnt/pidrive2/backup`
(Backup system has `/dev/sda3` also mounted in `/mnt/pidrive2` via `/etc/fstab`)

framps, 2021-08-03:
Thank you very much for the detailed answer. It took me some time to grasp your backup and restore strategy but I got it now :smiley: 

I think I found the root cause for your issue :smirk: : Please use the attached patched raspiBackup version (rename .txt to .sh) and restore your backup. Then check the ACL again on the restored system on /media/pi.

[raspiBackup.txt](https://github.com/framps/raspiBackup/files/6926627/raspiBackup.txt)

Marian1GitHub, 2021-08-03:
Great, now it seems to be working.

Thanks for your fast support framps !

framps, 2021-08-03:
I'm glad I found the rot cause. Actually this is a bug and I will roll in the fix into current raspiBackup beta 0.6.6.1.

> I'm only be surprised no one else has reported this issue before,

I don't have any automounted devices on my Raspberries because it's unusual on servers to use automount.  Looks like other users of raspiBackup also don't have automounted devices.

> this was also my first reported issue on GitHub

Congratulations :+1: Hope you now don't hesitate any more to create issues in github for all the open source you use :wink: If you're fine now please close the issue.

> Thanks for your fast support framps !

You're welcome

Marian1GitHub, 2021-08-03:
>   ... but my name is framp ...

oh sorry for this mistake framp it was not intended.

> I don't have any automounted devices on my Raspberries because it's unusual on servers to use automount. 

I'm currently upgrading my old pi2 mini server to a pi4, mainly because of the 1gbit Lan support, and I tried the new self backup strategy because my previous one dont perform on the new RaspianOS (I have found a workaround now, but I don't know why it was not working before) . Now I started to transfer the data from my old hdd USB drive to the new SSD via USB and I simply plugged it in and run into this trouble. Because USB 2.0 is still faster than 100mbit Lan.

closed 2021-08-03 by Marian1GitHub

framps, 2021-08-04:
> oh sorry

NP. I'm used to be called `framps` :laughing: