framps/raspiBackup #352: -L current not working anymore

christianTF, 2021-08-04:
Hi Framp!

Something seems have changed with the `-L current` parameter in the current version of raspiBackup.

Users with current version seems to get errors
```
touch: cannot touch '/home/loxberry/raspiBackup.log': No such file or directory
touch: cannot touch '/home/loxberry/raspiBackup.msg': No such file or directory
tee: /home/loxberry/raspiBackup.log: No such file or directory
tee: /home/loxberry/raspiBackup.log: No such file or directory
```
(see appended log of a user)

In the log, you see the line
`<INFO> Changing to directory /opt/loxberry/log/plugins/lbbackup`
that changes the current directory.
After starting raspiBackup with `-L current`, the script for an unknown reason seems to search for `/home/loxberry`, that does not exist on LoxBerry.

The loxberry user has defined the /opt/loxberry directory as home:
```bash
loxberry@loxberry-dev:/ $ export
declare -x DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1001/bus"
declare -x HOME="/opt/loxberry"
```
Also as user root, the home directory of `loxberry` can be queried:
```bash
root@loxberry-dev:~# eval echo ~loxberry
/opt/loxberry
```

Therefore, there seem to be two issues:

1. The script does not use the current directory anymore for logfiles with `-L current` parameter
2. The script does not correctly aquire the home directory of the user.

BTW, in the docu https://www.linux-tips-and-tricks.de/de/raspibackup/#parm_L the param -L is not explained anymore.
If I could directly give the log path as parameter, I wouldn't need the `-L current` at all. 

[20210731_150746_820_backup.log](https://github.com/framps/raspiBackup/files/6928850/20210731_150746_820_backup.log)

For the upcoming LoxBerry Backup release, I've changed the install behaviour to directly ship with the latest stable release (currently 0.6.6) of raspiBackup.sh, not using the installer. The 0.6.6 version doesn't seem to have the `-L current` issue.

Best regards!
Christian

framps, 2021-08-04:
0.6.6.1 Beta has the logging rewritten. I'm sure I tested `-L` but looks like there is still some issue.

>     2\. The script does not correctly aquire the home directory of the user.

Would you please explain in more detail ? Not sure I get your point.

> BTW, in the docu https://www.linux-tips-and-tricks.de/de/raspibackup/#parm_L the param -L is not explained anymore.

Thank you very much you point out this issue. I had some trouble on my website recently and the option vanished by accident. Will fix this.

> For the upcoming LoxBerry Backup release, I've changed the install behaviour to directly ship with the latest stable release (currently 0.6.6) of raspiBackup.sh, not using the installer.

That's a very good idea :+1: .

christianTF, 2021-08-04:
`touch: cannot touch '/home/loxberry/raspiBackup.log': No such file or directory`

Your script seems to try to access the `/home/loxberry` directory. 

You have hardcoded `/home` in your code:
https://github.com/framps/raspiBackup/blob/a33a2321f0b33cbf1fb2af1d95503b41b46090b1/raspiBackup.sh#L2705

But the actual home directory of users is configured in `/etc/passwd` file. 
[image]
(6th column on `:` split)

Therefore, the static `/home` path in the code may lead to problems.

The `eval echo ~username` is a method to aquire the home directory of the specified user.

framps, 2021-08-04:
> The `eval echo ~username` is a method to aquire the home directory of the specified user.

Got it. I frankly didn't know that's an easy way to retrieve the configured users home directory :+1:

framps, 2021-08-05:
I fixed the issue. Please grab the new beta code with `sudo raspiBackup.sh -U -S` and test the fix.

framps, 2021-08-24:
@christianTF Would you please verify the fix?

framps, 2021-09-08:
Closing the issue because of inactivity

closed 2021-09-08 by framps