framps/raspiBackup #353: Beta: Parameter -L shows error

WAdama, 2021-08-05:
When starting beta version 0.6.6.1 I got this error:
!!! RBK0059W: Invalid parameter '/mnt/cifs/Temp/Logs/Backup/<servername>/2021-08-05_083712-backup.log' for option -L detected.

raspiBackup is started via script, which creates all the necessary variables used for raspiBackup.
0.6.6.0 worked without problems in this case.

framps, 2021-08-05:
This most probably is be a duplicate of #352 . Please provide the debug log to check.

WAdama, 2021-08-05:
Hi framp,
I'm not entirely sure about that. I'm just getting this message direct after start but nothing happens of that described in #352

framps, 2021-08-05:
You should find the debug log in your home directory if you get this warning message.

WAdama, 2021-08-05:
This should be this one...
[raspiBackup.log](https://github.com/framps/raspiBackup/files/6938884/raspiBackup.log)

framps, 2021-08-05:
Thank you very much for the log. I fixed the issue. Please grab the new beta code with `sudo raspiBackup.s -U -S` and test the fix.

WAdama, 2021-08-05:
I'm afraid this is even worse.
I got no log not even a debug log.
Therefor extension output on screen. It seems to me as the script tries to get user from passwd. I notice user from my AD domain in the output.

framps, 2021-08-05:
> the script tries to get user from passwd.

Yes, that's what I had to change because of the other -L issue. I'm unable to test with AD because I have no AD environment :cry: . An alternate way to get the home directory of a user is `eval echo ~<user>` . Could you please execute on your system with the AD user previous command and show the result?

WAdama, 2021-08-05:
If I do a `eval echo ~root` I get /root or
If I do a `eval echo ~pi` I get /home/pi
I hope this is it what you ment...

framps, 2021-08-05:
> I hope this is it what you ment

Yes. Thank you very much. This sounds good. Could you just update the following line 139

`CALLING_HOME="$(getent passwd $CALLING_USER | cut -f6 -d:)"`
to
`CALLING_HOME="$(eval echo ~${CALLING_USER})"`

and test agin? Crossing fingers it will then work fo you ...

WAdama, 2021-08-05:
It works...

The funny thing: I tested the old line `getent passwd root | cut -f6 -d:` and it also gave /root back...
So why didn't it work in the script? What's the difference?

framps, 2021-08-05:
Not sure I get you :cry: 

> It works...
You modified line 139 to use eval echo ... and then raspiBackup finished successfully?

> I tested the old line `getent passwd root | cut -f6 -d:` and it also gave /root back...
Did you use this code in line 139 or just on the command line?

WAdama, 2021-08-05:
Sorry for the confusion...

The "It works" is targeted to your new line... With the eval command the backup worked again...

But it made me curious what the old line would give back for the user. And it also gave back the correct home path. So I'm a little puzzled.

framps, 2021-08-05:
What do you get for

```
getent passwd root | cut -f6 -d:
getent passwd pi | cut -f6 -d:
eval echo ~root
eval echo ~pi
```

?

WAdama, 2021-08-05:
For `getent passwd root | cut -f6 -d:` /root
For `getent passwd pi | cut -f6 -d:` /home/pi
For `eval echo ~root` /root
For `eval echo ~pi` /home/pi
It seems all identical...

framps, 2021-08-05:
Hm ... would you do me a favor?

Please use the original beta script with `CALLING_HOME="$(getent passwd $CALLING_USER | cut -f6 -d:)"` in line 139 and execute following commands:

1) script raspiBackup.lg
2) sudo bash -x raspiBackup.sh
3) exit

Finally attach the file raspibackup.lg in this issue. But please check for sensitive data in the file first.

WAdama, 2021-08-06:
Of course...

I have changed sensitive data. I hope it is to see what it was for your analysis...
[raspiBackup.zip](https://github.com/framps/raspiBackup/files/6944407/raspiBackup.zip)

framps, 2021-08-06:
>  I hope it is to see what it was for your analysis...

Yes :+1:  I downloaded the log and deleted it. Just for sake of safety :wink: 

The function i wrote to detect the calling user home directory (findUser) returns an empty string. Then `"$(getent passwd $CALLING_USER | cut -f6 -d:)"` returns the home directory defined in passwd for **all** lines and then CALLING_HOME which is used all over in the script is incorrect and creates the issue you see. 

`$(eval echo ~${CALLING_USER})` instead returns /root if CALLING_USER is empty and that's why the issue is gone with this code.

```
||94> 	findUser(): '[' supervi+ = root ']'
||105> 	findUser(): getent passwd supervi+
||105> 	findUser(): cut -d: -f1
|138> 	CALLING_USER=
||139> 	getent passwd
||139> 	cut -f6 -d:
|139> 	CALLING_HOME='/root
/usr/sbin
/bin
/dev
/bin
...
```

I have no idea but according the log there exists a user `supervi+` which is not defined in passwd but returned with `whoami`. Do you have any explanation where user `supervi+` comes from and where it's used?

The very detailed debug log shows there is an issue in findUser in your environment. So either I find out what's the root cause `supervi+` is and fix this or I have to step back and find another way to locate the home directory of the sudo user.

Let's see whether you have an idea where `supervi+` comes from.

WAdama, 2021-08-06:
This is not much a mystery...
This is a local user on my raspberry. Why it's shortened I can't imagine, the name is longer.
But the interesting thing is that the same user exists also in my domain. So if I do a `getent passwd | grep super` for example I get two users back.

WAdama, 2021-08-06:
Another observation:
If I do a `getent passwd supervi+` I get only the local user.
The domain user is written with a capital S so I have to do a `getent passwd Supervi+`

framps, 2021-08-06:
I now rewrote `findUser` and I think it should work now for you and all the others also. Please test the attached version. 
[raspiBackup.txt](https://github.com/framps/raspiBackup/files/6946374/raspiBackup.txt)

WAdama, 2021-08-06:
Just tested...

I'm afraid it's showing the same behaviour... :-(

WAdama, 2021-08-06:
Is it possible, you attached the wrong file? I can't determine a difference between the `finduser` functions...

framps, 2021-08-06:
Yes. Sorry about this :cry:  I uploaded the file again and verified the changes are in now :wink:

WAdama, 2021-08-06:
That's looking much better...

My Raspberry is just backing up...

framps, 2021-08-06:
Sounds promising. Do you follow [RFC 1149](https://de.wikipedia.org/wiki/Internet_Protocol_over_Avian_Carriers)? Did the backup still did not complete :laughing:  ?

WAdama, 2021-08-06:
The backup was complete, so everything seems ok now for this issue...

framps, 2021-08-06:
:+1: So this issue was solved with your help :+1: 

Thank you very much for your help to locate the root cause of the issue. I'll roll in the fix into the official beta tomorrow.

framps, 2021-08-06:
Rolled in the fix right now ...

framps, 2021-08-07:
> everything seems ok now for this issue...

Please close the issue if you're fine now.

WAdama, 2021-08-07:
I'm fine and closing this... :-)

closed 2021-08-07 by WAdama

framps, 2021-08-07:
Again thank you very much for your support to find the root cause for the issue :+1: