framps/raspiBackup #354: /bin /sbin /lib are symlinks to /usr/... instead of folders

Marian1GitHub, 2021-08-05:
Hello framp,

on the stored rsync backup and after restore the folders `/bin /sbin /lib` are now Symlinks to `/usr/...`
```
lrwxrwxrwx   1 root root     7 Mai  7 16:39 bin -> usr/bin
lrwxrwxrwx   1 root root     7 Mai  7 16:39 lib -> usr/lib
lrwxrwxrwx   1 root root     8 Mai  7 16:39 sbin -> usr/sbin
-rw-r--r--   1 root root 80912 Aug  4 11:57 raspiBackup.log
-rw-r--r--   1 root root  1425 Aug  4 11:57 raspiBackup.msg
```
and on an original installation they are only folders with also lesser content.
```
drwxr-xr-x   2 root root  4096 Nov  3  2020 bin
drwxr-xr-x  17 root root  4096 Mai 27  2020 lib
drwxr-xr-x   2 root root  4096 Nov  3  2020 sbin
```

Here are my file counts of this folders on a not restored RaspianOS

`ls -1 | wc -l`
`/bin` = 158
`/usr/bin` = 1228
`/sbin` = 175
`/usr/sbin` = 184
`/lib` = 19
`/usr/lib` = 129

I'm not familiar enough with Linux to decide if it is a problem or not, currently everything runs like expected. 
But at end of the day it's not equal to the origin.

Marian1GitHub, 2021-08-05:
Sorry, I had compared it  with a older RaspianOS Installation.
Now on my Backup System which was new installed and never restored this folders are also symlinks to `/usr/...` maybe the RaspianOS installation was changed in the meantime.

closed 2021-08-05 by Marian1GitHub

framps, 2021-08-05:
NP. Shit happens :laughing: