framps/raspiBackup #355: Kann man ein vorhandenes rsync Backup updaten?

stalakerob, 2021-08-10:
Ich würde gerne ein vorhandenes rsync Backup manuell updaten, geht das? 
Anwendungsfall:
- Ich habe ein komplettes raspiBackup rsync backup
- Jetzt ändere ich nur wenige Quell-Dateien (z.B. .py, .yaml, .sh Dateien)
- Kann ich diese geänderten Dateien manuell mit rsync (evtl auch cp) nachpflegen ohne das Backup unbrauchbar zu machen und welche rsync optionen müsste man verwenden?

framps, 2021-08-10:
Nur um sicher zu gehen ich verstehe Deine Frage richtig fasse ich noch mal zusammen: Du moechtest ein existierendes rsync Backup manuell aendern. D.h. Du moechtest ein paar Dateien die Du auf einem laufenden System hast in das Backup reinkopieren damit Du wenn das Backup restored wird die geaenderten Dateien beinhaltet? D.h. das ist quasi der schnelle Backup mit ein paar kleinen Modifikationen :wink: 

Das geht eigentlich recht einfach bei einem rsync Backup. Du renamest die alten Dateien im Backup von `<Dateiname>` in `<Dateiname>.old` damit Du sie fuer den Fall der Faelle noch zur Verfuegung hast und auch die alten Dateiattribute im Fall der Faelle nachsehen kannst (Nicht notwendig - aber ich empfehle es :wink: ) Dann kopierst Du mit `cp -a <Neuedatei> <Backupverzeichnis>` die Dateien vom laufenden System ins Backup an dieselbe Stelle wie die umbenannte Datei.

Sollten die Dateien nicht vom identischen laufenden System kommen wo die Attribute stimmen musst Du nachdem Du sie kopiert hast mit denselben Attributen versehen die die Originaldatei hat. Typische Befehle dazu sind `chown` und `chmod`. Fuer beide gibt es die Option `--reference=` mit der man sehr leicht Attribute clonen kann.

stalakerob, 2021-08-10:
Danke. Genau das war meine Frage.

closed 2021-08-10 by stalakerob