framps/raspiBackup #356: Error while backing up with 0.6.6.1 beta

hugalafutro, 2021-08-14:
Hi,
today I backed up 2 RPi3's and one of them sent me email from cron saying: `/usr/local/bin/raspiBackup.sh: line 2488: ((: !  : syntax error: operand expected (error token is "!  ")`

relevant portion of the log:
```
20210814-064419 DBG 3278:          --> cleanupBackupDirectory
20210814-064419 DBG 3280:                  *** ls -la /mnt/backup/@HOSTNAME@/@HOSTNAME@-tar-backup-20210814-063101
/usr/local/bin/raspiBackup.sh: line 2488: ((: !  : syntax error: operand expected (error token is "!  ")
20210814-064419 DBG 3280:                      total 3227255
20210814-064419 DBG 3280:                      drwxr-x--- 2 root root          6 Aug 14 06:31 .
20210814-064419 DBG 3280:                      drwxr-x--- 6 root root          6 Aug 14 06:31 ..
20210814-064419 DBG 3280:                      -rwxr-x--- 1 root root  268435968 Aug 14 06:31 @HOSTNAME@-backup.img
20210814-064419 DBG 3280:                      -rwxr-x--- 1 root root        512 Aug 14 06:31 @HOSTNAME@-backup.mbr
20210814-064419 DBG 3280:                      -rwxr-x--- 1 root root        181 Aug 14 06:31 @HOSTNAME@-backup.sfdisk
20210814-064419 DBG 3280:                      -rwxr-x--- 1 root root 6348769280 Aug 14 06:44 @HOSTNAME@-tar-backup-20210814-063101.tar
20210814-064419 DBG 3297:          <-- cleanupBackupDirectory
```

full log: https://paste.o5.ddns.net/?bda772943781a502#CWHjxy5d9eb6bozLjoBLf1hotHCakkK3CvBG4AZahBxv
rasbiBackup.conf: https://paste.o5.ddns.net/?1d79a4cec38dbf58#3y2ede8ogJvsKynNW6hpTm33wtJXktQKLpscvVmj5UXi

framps, 2021-08-14:
Thank you very much you reported the issue :+1: . i fixed it right now and if you grab the latest beta with `sudo raspu_backup.sh -U -S` you'll get the fixed version. Please verify the fix.

tocotronaut, 2021-08-14:
For me `sudo raspiBackup.sh -U -S` did the Update...

hugalafutro, 2021-08-14:
Same, probably just typo. fixed with  `sudo raspiBackup.sh -U -S` Thanks!

closed 2021-08-14 by hugalafutro