framps/raspiBackup #357: Cronjob Fehler - mv: failed to preserve ownership for...

Lusbueb69, 2021-08-18:
hallo Framp

Ich sichere meine vier RasPis mit "raspiBackupNfsWrapper" auf ein Netgear "ReadyNAS NV+ v2" und diese laufen auch fehlerlos durch jedoch erhalte ich gleich im Anschluss an die Sicherung vom Cronjob diese Fehlermeldung:

_Betreff:_ **Cron /usr/local/bin/raspiBackupNfsWrapper.sh > /dev/null # >> /var/log/raspiBackupJob.log 2>&1 #Backup Taeglich um 01:31 Uhr**

**mv: failed to preserve ownership for '/backup/RasPi/RasPi-tar-backup-20210818-013102/raspiBackup.log': Operation not permitted**

```
20210818-013102 DBG 1364:  --- raspiBackup.sh 0.6.6.1-beta, 2021-08-14/17:33:51 - a1880b2
20210818-013102 DBG 1367:  --- +================================================================================================================================================+
20210818-013102 DBG 1368:  --- | ===> A lot of sensitive information is masqueraded in this log file. Nevertheless please check the log carefully before you distribute it <=== |
20210818-013102 DBG 1369:  --- +================================================================================================================================================+
20210818-013102 DBG 1202:  --> trapWithArg TRAP cleanupStartup SIGINT SIGTERM EXIT
20210818-013102 DBG 1207:  <-- trapWithArg 
20210818-013102 DBG 1845:  --> initializeDefaultConfigVariables 
20210818-013102 DBG 1977:  <-- initializeDefaultConfigVariables 
20210818-013102 DBG 1982:  --> copyDefaultConfigVariables 
20210818-013102 DBG 2051:  <-- copyDefaultConfigVariables 
20210818-013102 DBG 2759:  --> readConfigParameters 
20210818-013102 DBG 2774:      --- Read config /usr/local/etc/raspiBackup.conf : 0.1.5
20210818-013102 DBG 2774:      --- VERSION_CONFIG="0.1.5"
20210818-013102 DBG 2774:      --- DEFAULT_BACKUPPATH="/backup"
20210818-013102 DBG 2774:      --- DEFAULT_BACKUPTYPE="tar"
20210818-013102 DBG 2774:      --- DEFAULT_ZIP_BACKUP="0"
20210818-013102 DBG 2774:      --- DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20210818-013102 DBG 2774:      --- DEFAULT_KEEPBACKUPS="30"
20210818-013102 DBG 2774:      --- DEFAULT_KEEPBACKUPS_DD=0
20210818-013102 DBG 2774:      --- DEFAULT_KEEPBACKUPS_DDZ=0
20210818-013102 DBG 2774:      --- DEFAULT_KEEPBACKUPS_TAR=0
20210818-013102 DBG 2774:      --- DEFAULT_KEEPBACKUPS_TGZ=0
20210818-013102 DBG 2774:      --- DEFAULT_KEEPBACKUPS_RSYNC=0
20210818-013102 DBG 2774:      --- DEFAULT_SMART_RECYCLE=1
20210818-013102 DBG 2774:      --- DEFAULT_SMART_RECYCLE_DRYRUN=0
20210818-013102 DBG 2774:      --- DEFAULT_SMART_RECYCLE_OPTIONS="7 4 12 3"
20210818-013102 DBG 2774:      --- DEFAULT_STOPSERVICES="systemctl stop lighttpd && systemctl stop pihole-FTL"
20210818-013102 DBG 2774:      --- DEFAULT_STARTSERVICES="systemctl start pihole-FTL && systemctl start lighttpd"
20210818-013102 DBG 2774:      --- DEFAULT_BEFORE_STOPSERVICES=""
20210818-013102 DBG 2774:      --- DEFAULT_AFTER_STARTSERVICES=""
20210818-013102 DBG 2774:      --- DEFAULT_EMAIL="p@@@@h(21)"
20210818-013102 DBG 2774:      --- DEFAULT_SENDER_EMAIL="R@@@@h(21)"
20210818-013102 DBG 2774:      --- DEFAULT_EMAIL_PARMS=""
20210818-013102 DBG 2774:      --- DEFAULT_MAIL_PROGRAM="mail"
20210818-013102 DBG 2774:      --- DEFAULT_TELEGRAM_TOKEN=""
20210818-013102 DBG 2774:      --- DEFAULT_TELEGRAM_CHATID=""
20210818-013102 DBG 2774:      --- DEFAULT_TELEGRAM_NOTIFICATIONS="F"
20210818-013102 DBG 2774:      --- DEFAULT_NOTIFY_START=0
20210818-013102 DBG 2774:      --- DEFAULT_MAIL_ON_ERROR_ONLY=0
20210818-013102 DBG 2774:      --- DEFAULT_MSG_LEVEL="1"
20210818-013102 DBG 2774:      --- DEFAULT_COLORING="CM"
20210818-013102 DBG 2774:      --- DEFAULT_EMAIL_COLORING="SUBJECT"
20210818-013102 DBG 2774:      --- DEFAULT_RESTORE_DEVICE=""
20210818-013102 DBG 2774:      --- DEFAULT_VERBOSE=0
20210818-013102 DBG 2774:      --- DEFAULT_SKIPLOCALCHECK=0
20210818-013102 DBG 2774:      --- DEFAULT_DD_BLOCKSIZE=1MB
20210818-013102 DBG 2774:      --- DEFAULT_DD_PARMS=""
20210818-013102 DBG 2774:      --- DEFAULT_EXCLUDE_LIST=""
20210818-013102 DBG 2774:      --- DEFAULT_NOTIFY_UPDATE=1
20210818-013102 DBG 2774:      --- DEFAULT_EXTENSIONS="temp"
20210818-013102 DBG 2774:      --- DEFAULT_LANGUAGE="DE"
20210818-013102 DBG 2774:      --- DEFAULT_DEPLOYMENT_HOSTS=""
20210818-013102 DBG 2774:      --- DEFAULT_YES_NO_RESTORE_DEVICE="loop"
20210818-013102 DBG 2774:      --- DEFAULT_RESIZE_ROOTFS=1
20210818-013102 DBG 2774:      --- DEFAULT_LINK_BOOTPARTITIONFILES=0
20210818-013102 DBG 2774:      --- DEFAULT_USE_HARDLINKS=1
20210818-013102 DBG 2774:      --- DEFAULT_TAR_BOOT_PARTITION_ENABLED=0
20210818-013102 DBG 2774:      --- DEFAULT_SKIP_DEPRECATED=""
20210818-013102 DBG 2774:      --- DEFAULT_USE_UUID=1
20210818-013102 DBG 2774:      --- DEFAULT_CHECK_FOR_BAD_BLOCKS=0
20210818-013102 DBG 2774:      --- DEFAULT_TIMESTAMPS=1
[... 938 lines cut ...]
20210818-013507 DBG 3023:          <-- colorAnnotation 
20210818-013507 DBG 3211:          --- eMail: p@@@@h(21)
20210818-013507 DBG 3212:          --- eMail Program: mail
20210818-013507 DBG 3213:          --- Subject: @HOSTNAME@: Backup erfolgreich beendet.
20210818-013507 DBG 3213:          --- MIME-Version: 1.0
20210818-013507 DBG 3213:          --- Content-Type: text/html; charset=utf-8
20210818-013507 DBG 3214:          --- ColoringOption: 
20210818-013507 DBG 3215:          --- ContentType: 
20210818-013507 DBG 3215:          --- MIME-Version: 1.0
20210818-013507 DBG 3215:          --- Content-Type: text/html; charset=utf-8
20210818-013507 DBG 3216:          --- Parms: 
20210818-013507 DBG 3221:          --- mail  -s "@HOSTNAME@: Backup erfolgreich beendet.
20210818-013507 DBG 3221:          --- MIME-Version: 1.0
20210818-013507 DBG 3221:          --- Content-Type: text/html; charset=utf-8" p@@@@h(21)
20210818-013509 DBG 3224:          --- mail: RC: 0
20210818-013509 DBG 3267:      <-- sendEMail 
20210818-013509 DBG 3337:      --- Masquerading eMail
20210818-013509 DBG 3345:      --- Masquerading sender eMail
20210818-013509 DBG 3360:      --- Masquerading some mount options
20210818-013509 DBG 3378:      --- Masquerading home directory name
20210818-013509 DBG 3383:      --- Masquerading hostname
20210818-013509 DBG 3388:      --- Masquerading sensitive non local IPs
20210818-013510 DBG 3587:  <-- cleanup 

```

framps, 2021-08-21:
Vielen Dank fuer das Log (Ich habe es in eine Datei ausgelagert. Da kann ich viel einfacher im Log suchen :wink: )

Der Fehler tritt offensichtlich ganz am Ende auf . Leider wird da nichts mehr gelogged da mit den Logfiles jongliert wird. Kannst Du bitte mal den Betacode ein ganz wenig wie folgt aendern?

Aus

```
# move temporary log file to it's destination
function logFinish() {
```

machst Du

```
# move temporary log file to it's destination
function logFinish() {
   set -x
```

und startest den Backup noch einmal. Dann solltest Du in /var/log/raspiBackupJob.log einen sehr detailierten Debugtrace bekommen der mir hilft die Ursache zu lokalisieren. Bitte das Log hier per drag and drop ablegen. Aber bitte vorher das Log nach moeglichen sensitiven Infos durchsehen und diese maskieren.

Lusbueb69, 2021-08-22:
Vielen Dank für deine Rückmeldung. Ich habe raspiBackup.sh entsprechend angepasst und hänge einerseits das raspiBackup.log und den Mail-Inhalt des Cron-Job bei.

[Cron-Mail.log](https://github.com/framps/raspiBackup/files/7027715/Cron-Mail.log)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/7027713/raspiBackup.log)

Die Log-Datei wird bei mir allerdings NICHT unter var/log/raspiBackup.log gespeichert sondern unter /home/pi/raspiBackup.log und unter /backup/<Pfad>/raspiBackup.log also auf dem NAS. Wenn der Job fertig ist habe ich nur noch das Log auf dem NAS.

framps, 2021-08-23:
Dank Deiner Aenderung habe ich jetzt die exakte Stelle gefunden wo der Fehler auftritt :+1: :

```
|1402> 	logFinish(): DEST_LOGFILE=/backup/RasPi/RasPi-tar-backup-20210822-013101/raspiBackup.log
|1403> 	logFinish(): mv /root/raspiBackup.log /backup/RasPi/RasPi-tar-backup-20210822-013101/raspiBackup.log
mv: failed to preserve ownership for '/backup/RasPi/RasPi-tar-backup-20210822-013101/raspiBackup.log': Operation not permitted
```
Daraus sieht man dass Du raspiBackup als root aufrufst (Das debuglog wird im Homeverzeichnis /root/raspiBackup.log erstellt). Am Ende wird das Log in das Backupverzeichnis kopiert  (/backup/RasPi/RasPi-tar-backup-20210822-013101/raspiBackup.log) und es kann die Ownership des OriginalLogs nicht gesetzt werden. Da Du das Log hier anhaengen kannst wird es schon kopiert - nur stimmt die Ownership nicht. 

Im Log sehe ich

```
20210822-013506 DBG 3276:                  *** ls -la /backup/@HOSTNAME@/@HOSTNAME@-tar-backup-20210822-013101
20210822-013506 DBG 3276:                      total 2337288
20210822-013506 DBG 3276:                      drwxr-xr-x  2 nobody nogroup       4096 Aug 22 01:31 .
20210822-013506 DBG 3276:                      drwxr-xr-x 24 nobody nogroup       4096 Aug 22 01:31 ..
20210822-013506 DBG 3276:                      -rw-r--r--  1 nobody nogroup   45190144 Aug 22 01:31 @HOSTNAME@-backup.img
20210822-013506 DBG 3276:                      -rw-r--r--  1 nobody nogroup        512 Aug 22 01:31 @HOSTNAME@-backup.mbr
20210822-013506 DBG 3276:                      -rw-r--r--  1 nobody nogroup        205 Aug 22 01:31 @HOSTNAME@-backup.sfdisk
20210822-013506 DBG 3276:                      -rw-r--r--  1 nobody nogroup 2348144640 Aug 22 01:35 @HOSTNAME@-tar-backup-20210822-013101.tar
```

D.h. es fehlt die Berechtigung auf dem NFS als Root zu schreiben da Du vermutlich nicht no_root_squash bei der Exportsdatei gesetzt hast. Die Logdatei wird geschrieben denn sonst koenntest Du sie ja nicht zeigen. Kannst Du mal die Ausgabe von 

`ls -la /backup/RasPi/RasPi-tar-backup-20210822-013101`

zeigen? Ich vermute raspiBackup.log hat auch nobody und nogroup. Warum beim Log eine Fehlermeldung geschrieben wird aber bei den anderen Backupdateien nicht verstehe ich noch nicht ganz. Dort haette ich sie auch erwartet.

Kannst Du mal 
`ls -la /backup/RasPi/RasPi-tar-backup-20210822-.....`
zeigen von einem Backup was mit 0.6.6 erstellt wurde zum Vergleich?

Tenor ist erst einmal: Das Backup ist OK. Die Fehlermeldung ist unschoen und sollte nicht erscheinen. Warum sie erscheint ist mir momentan noch nicht so klar. Ich hoffe Deine weiteren Infos helfen. Jedenfalls  wuerde bei Dir die Fehlermeldung verschwinden wenn Du no_root_squash benutzen wuerdest.

Lusbueb69, 2021-08-25:
wie gewünscht einmal die Ausgabe **20210822-013101 DBG 1364:  --- raspiBackup.sh 0.6.6.1-beta, 2021-08-14/17:33:51**
```
pi@RasPi4:~ $ ls -la /backup/RasPi/RasPi-tar-backup-20210822-013101
total 4947520
drwxr-xr-x  2 nobody nogroup       4096 Aug 22 02:34 .
drwxr-xr-x 25 nobody nogroup       4096 Aug 25 01:35 ..
-rw-r--r--  1 nobody nogroup   45190144 Aug 22 01:31 RasPi-backup.img
-rw-r--r--  1 nobody nogroup      82995 Aug 22 01:35 raspiBackup.log
-rw-r--r--  1 nobody nogroup        512 Aug 22 01:31 RasPi-backup.mbr
-rw-r--r--  1 nobody nogroup        205 Aug 22 01:31 RasPi-backup.sfdisk
-rw-r--r--  1 nobody nogroup 3168072192 Aug 22 02:34 RasPi-full-image--21-08-22_023415.img
-rw-r--r--  1 nobody nogroup 2348144640 Aug 22 01:35 RasPi-tar-backup-20210822-013101.tar
```

und die Ausgabe **20210809-013104 MSG 7666:  --- RBK0009I: @HOSTNAME@: raspiBackup.sh V0.6.5**, falls du eine 0.6.6 - Ausgabe möchtest, müsste ich erst eine erstellen.
```
pi@RasPi4:~ $ ls -la /backup/RasPi/RasPi-tar-backup-20210809-013102
total 2568540
drwxr-xr-x  2 nobody nogroup       4096 Aug  9 01:35 .
drwxr-xr-x 25 nobody nogroup       4096 Aug 25 01:35 ..
-rw-r--r--  1 nobody nogroup   45190144 Aug  9 01:31 RasPi-backup.img
-rw-r--r--  1 nobody nogroup      68370 Aug  9 01:35 raspiBackup.log
-rw-r--r--  1 nobody nogroup        512 Aug  9 01:31 RasPi-backup.mbr
-rw-r--r--  1 nobody nogroup       2980 Aug  9 01:35 raspiBackup.msg
-rw-r--r--  1 nobody nogroup        205 Aug  9 01:31 RasPi-backup.sfdisk
-rw-r--r--  1 nobody nogroup 2584872960 Aug  9 01:35 RasPi-tar-backup-20210809-013102.tar
```

Lusbueb69, 2021-08-30:
mit der Beta V0.6.6.1-beta (65352d6) läuft nun alles durch ohne zusätzliche Meldung des CronJob. Ich denke, dass dieses Thema nun geschlossen werden kann.

Vielen Dank für die Unterstützung und die Anpassung. :-)

closed 2021-08-30 by Lusbueb69