framps/raspiBackup #358: raspiBackupRestore2Image: Boot-Probleme nach Windows Restore (IMG), UUID falsch!(?)

Lusbueb69, 2021-08-18:
Hi Framp

Nachdem alle 4 Backups fertig sind, migriere ich mit einem RasPi alle 4 **TAR**-Sicherungen mit **raspiBackupRestore2Image** in eine **IMG**-Datei, damit ich diese unter Windows wieder auf eine SD-Karte kopieren kann um möglichst einfach ein defektes System wieder zu restoren.

Wenn ich dann jedoch ab dieser SD-Karte boote, bleibt das System irgendwann stehen. 

Habe dann herausgefunden, dass es mit der UUID der Partitionen zu tun hat und konnten das IMG nur zum Laufen bringen, wenn ich ein Restore der **TAR**-Datei unter Linux durchführe:
` sudo raspiBackup.sh -d /dev/sda /backup/RasPi/RasPi-tar-backup-20210523-013102/`

framps, 2021-08-21:
Die Aenderung in der Version 0.1.6 ist marginal und sollte das Ergebnis nicht aendern.

> Habe dann herausgefunden, dass es mit der UUID der Partitionen zu tun hat

In raspiBackupRestore2Image wird die UUID entsprechend angepasst und an der Codepathes wurde eigentlich nichts geaendert. Ich werde das Szenario bei mir mal nachstellen. Hast Du irgendwas an raspiBackupRestore2Image geaendert?

framps, 2021-08-23:
Ich habe jetzt Dein Szenarion nachgestellt mit der Betaversion von raspiBackup und raspiBackuprestore2Image. In letzterem musste ich noch folgendes aendern damit es mit der Beta laeuft:

```
echo "===> Restoring backup into $IMAGE_FILENAME"
raspiBackup.sh -1 -Y -d $LOOP "$BACKUP_DIRECTORY"
RC=$?
```
habe ich geaendert in

```
echo "===> Restoring backup into $IMAGE_FILENAME"
raspiBackup.sh -1 -Y -d $LOOP "$BACKUP_DIRECTORY"
RC=$?
RC=0
```

Damit erhalte ich ein gepishrinktes Image welches ich auf eine SD Karte restoren kann und das restorte Image auch bootet.

Lusbueb69, 2021-08-25:
Ich muss mir zuerst mal eine zusätzliche SD-Karte besorgen, deshalb melde ich mich diesbezüglich erst jetzt und **ja du hast recht, mit der 0.6.6.1 Beta lief dieser Job auch nicht mehr**, erst nachdem ich aus [https://github.com/framps/raspiBackup/issues/357](url) dies implementiert habe lief der Restore2Image durch...
```
# move temporary log file to it's destination
function logFinish() {
   set -x
```

Test und Resultat wird noch folgen :-)

Lusbueb69, 2021-08-29:
Ich bin gerade mal an einem RasPi am testen. Die SD-Karte beschreibe ich mit **balenaEtcher 1.5.200**. Nach dem Restore bootet der RasPi nicht mehr vollständig sondern bleibt stehen (siehe Bilder).

[image]
[image]

Am **raspiBackupRestore2Image** habe ich nichts gemacht, dafür im CronJob das Log wieder eingeschaltet, das sieht sehr spannend aus! Ich glaube dass mein Problem entsteht, weil ich mit einem am LAN angeschlossenen RasPi4 alle Images shrinke, dazu führe ich auf dem RasPi4 einen Cronjob aus **r2i.sh**. 

r2i.sh hat nur noch einen kleinen Fehler den ich erst jetzt bemerkt habe: Die Idee war, dass eine IMG-Datei nur geschrieben wird, wenn noch keine besteht, wie das Log jedoch zeigt, macht es immer eine IMG-Datei, muss ich dann noch korrigieren! 
[r2ish.txt](https://github.com/framps/raspiBackup/files/7072888/r2ish.txt)

Was mir im Log ausgefallen ist, dass bei jedem _RasPi_ beim migrieren so etwas passiert:
`Enter the new disk identifier: 
Disk identifier changed from 0xe0e05147 to 0x3914b778.`

Kann es sein, dass das Problem entsteht weil ich von dem **RasPi4** alle *.IMG erstelle anstelle von jedem einzelnen RasPi?

Liebe Grüsse
Peter
[r2i.log](https://github.com/framps/raspiBackup/files/7072908/r2i.log)

Lusbueb69, 2021-08-29:
Habe jetzt am RasPi4 per USB-Micro-SD Karten-Leser ein Linux Restore gemacht mit diesem Befehl:
`sudo raspiBackup.sh -d /dev/sda /backup/MM/MM-tar-backup-20210829-003102/`

Mein MagicMirror startet nun wieder *uff*

Anbei das Log: 
[raspiBackup.log](https://github.com/framps/raspiBackup/files/7073063/raspiBackup.log)
Kannst du damit etwas anfangen?

Liebe Grüsse
Peter

KleBoR, 2021-08-30:
Hi,

ich habe exakt das gleiche problem.
backup mittels wrapper gestartet - ebenfalls mittels wrapper raspi2image durchgeführt und mit pishrink verkleinert.

Läuft alles ohne fehler durch - jedoch ist das booten von dem zurückgeschriebenen image nich möglich.
Gibt es hierzu schon news?

p.s. im wrapper fehlt bei dem aufruf von raspi2image in zeile 129 "readVars"..

Gruss

framps, 2021-08-30:
Vielen Dank fuer die Screenshots. Bist Du sicher dass im zweiten Bild End ist und nichts mehr passiert? 

Bootprobleme sind leider sehr schwer zu debuggen :cry: Noch mal als Doublecheck: Mit 0.6.6 funktioniert das noch und mit der 0.6.6.1 Beta nicht mehr? Vielleicht finde ich ja auch die Ursache wenn ich den alten und neuen Codestand mal vergleiche.

Es gibt [diese Seite ](https://www.raspberrypi.org/forums/viewtopic.php?t=58151) wo diverse Tipps gegeben werden. Weiss nicht ob Du Dich da durchbeissen willst ...

> Was mir im Log ausgefallen ist, dass bei jedem _RasPi_ beim migrieren so etwas passiert:

Was meinst Du mit "migrieren"? Wo hast Du diese Meldung gesehen?

> Kannst du damit etwas anfangen?

Ja. Das bedeutet dass das Backup an sich OK ist und irgendwo im Pfad des Image generierens das Problem entsteht. Backup und Restore wird immer im Regressionlauf getestet. Die Generierung eines Images aus dem Backup ist als Helper as is von mir angeboten und wird deshalb nicht Regression getestet. Aber es sollte schon wieder mit der neuen Release funktionieren. Deshalb ist gut dass Du das getestet hast.

Wie gesagt funktioniert es bei mir.  Die Frage wo der Unterschied bei Dir liegt :thinking: 

Bestaetige bitte noch mal dass es mit 0.6.6 bei Dir funktioniert und mit 0.6.6.1 Beta nicht mehr. Dann werde ich versuchen durch Codevergeleich die Ursache zu finden.

> Gibt es hierzu schon News?

Leider nein. Ist es bei Dir auch so dass es noch mit der Releas 0.6.6 funktioniert hat?

Lusbueb69, 2021-08-30:
Nichts zu danken! Ich danke dir, dass du dir Zeit nimmst, macht echt Spass so :-)

> Bist Du sicher dass im zweiten Bild End ist und nichts mehr passiert?

Ja und Nein. Als ich das Problem Ende Mai 21 zum ersten Mal an meinem MagicMirror hatte, habe ich das Display ausgebaut da der Bildschirm quer eingebaut ist und leider die unterste(n) Zeilen nicht mehr anzeigt... Nun war es eben so: ie erste Meldung deutete darauf hin, dass keine Tastatur angeschlossen ist und dass man mit [ENTER] weiter kommen solle. Das war dann bedingt so... Auch mit Tastatur lief der Bootvorgang nicht durch, kam aber irgendwie weiter und ich hackte mich einfach durch soweit ich kam. Nach meiner Interpretation konnte er die Partition(en) nicht mounten weil irgend eine (UU)ID nicht mit ener "Vorgabe" übereinstimmte... Kam mir so vor, als ob er ein Medium mit der Bezeichnung ABCDEF mounten wollte aber das Medium meldete ich bin DEFGHI. Habe dann irgendwo im Netz gelesen, dass dies eine Sicherheitsrelevante Abfrage sei, die irgendwann mal in den letzten  Jahren unter Linux implementiert wurde... Ich bin kein Linux-Crack, ist aber gleich bescheuert wie Windows und UEFI-Boot! Ich sehne mich nach den guten, alten DOS-Disketten *lach*

> Mit 0.6.6 funktioniert das noch und mit der 0.6.6.1 Beta nicht mehr?

Falsch, Meine Restore-Versuche im Mai waren mit der V0.6.5. Ich habe jedoch gesehen, dass ich noch eine Sicherung von 2019 habe (*.IMG) die V0.6.4.3 erstellt wurde. Muss ich unbedingt testen!!!

> Was meinst Du mit "migrieren"? 

Unter migrieren meine ich die Konvertierung von *.TAR zu *.IMG inkl. PiShrink.

> Wo hast Du diese Meldung gesehen?

im [https://github.com/framps/raspiBackup/files/7072982/r2i.log](url) in den Zeilen **143**, **294**, **445**, **595** und **746**

> Wie gesagt funktioniert es bei mir. Die Frage wo der Unterschied bei Dir liegt

Es gibt mir ein sehr gutes Gefühl dass ich weiss dass das Backup in Ordnung ist und mittlerweile habe ich mir auch den Restore-Befehl unter Linux so notiert, dass ich ihn wieder finde *lach*

KleBoR, 2021-08-31:
Hi, zu den Versionen, ich habe 0.6.6 und 0.6.6.1B getestet und beide funktionieren (bei mir) leider nicht. @framps nutzt du auch den Wrapper resp. das Backup2image Script? 
könntest du evl deine Config posten dann würde ich die mal abgleichen. 

Danke & Gruss

framps, 2021-08-31:
> Es gibt mir ein sehr gutes Gefühl dass ich weiss dass das Backup in Ordnung ist und mittlerweile habe ich mir auch den Restore-Befehl unter Linux so notiert, dass ich ihn wieder finde (_lach_)

Das ist erst einmal das Wichtigste dass ein Restore ohne den Umweg ueber raspiBackup2Image funktioniert :+1: . Der Weg wird von mir für jedem Publish einer neuen Release regressiongetestet. 

raspiBackup2Image zusammen mit pishrink ist im Regressiontest nicht dabei da raspiBackup2Image ein Helperscript ist und alle Helperscripts sind as is angeboten und werden nicht wie raspiBackup von mir regressiongetestet, gepflegt und gewartet. Aber trotzdem versuche ich soweit wie moeglich dafuer zu sorgen dass es funktioniert.

Dumm ist dass es bei mir funktioniert. Meinen Test hatte ich mit der Beta gemacht. Ich werde jetzt noch mal die aktuelle 0.6.6. wie auch aktuelles raspiBackupRestore2Image und aktuelles pishrink nehmen und noch mal testen. Nutzt Du auch das aktuelle pishrink? Leider wird pishrink schlecht gewartet und ich koennte mir auch vorstellen dass dort irgendwo das Problem reinkommt.

framps, 2021-08-31:
>  @framps nutzt du auch den Wrapper resp. das Backup2image Script?

Ich nutze das aktuelle raspiBackup 
`raspiBackup         : Version: 0.6.6      Date: 2021-08-06 10:15:00  Sha: 262862c`   
 und aktuelle raspiBackupRestore2Image aus dem git.

Lusbueb69, 2021-09-01:
Endlich habe ich eine weitere SD-Karte zum testen und vorab die ersten Ergebnisse von 2 getesteten RasPi's. Zum besseren Verständnis eine Umgebung:

- 2x RasPi LAN, 3x RasPi WiFi
- Jeder RasPi sichert sich selber, zeitversetzt auf das NAS
- RasPi4 konvertiert und shrinkt nacheinander alle Images von TAR zu IMG (raspiBackupRestore2Image.sh) auf dem NAS da er der schnellste ist und die Dauer des Jobs per LAN-Anschluss relativ kurz ist.

**RasPi4 sichert sich auf das NAS danach konvertiert er sein eigenes tar-Image zu einem img-Image:**
```
RasPi4 (raspiBackup.sh 0.6.6.1-beta - 65352d6) -> NAS (RasPi4.tar)
RasPi4 (raspiBackupRestore2Image.sh v0.1.6) <-> NAS (RasPi4.tar->RasPi4.img)
balenaEtcher (1.5.120) <-> NAS (RasPi4.img auf SD-Karte) -> alles io, RasPi4 bootet
```
**MM sichert sich auf das NAS danach konvertiert RasPi4 das MM tar-Image zu einem img-Image:**
```
MM (raspiBackup.sh 0.6.6.1-beta - 65352d6) -> NAS (MM.tar)
RasPi4 (raspiBackupRestore2Image.sh v0.1.6) <-> NAS (MM.tar->MM.img)
balenaEtcher (1.5.120) <-> NAS (MM.img auf SD-Karte) -> MM bootet nicht!
```
**MM sichert sich auf das NAS danach weiss ich nicht mehr ob ob MM oder RasPi4 das tar-Image zu einem img-Image konvertiert hat**
```
MM (raspiBackup.sh V0.6.4.3 2d927a2) -> NAS (MM.tar)
? (raspiBackupRestore2Image.sh v?.?.?) <-> NAS (MM.tar->MM.img)
balenaEtcher (1.5.120) <-> NAS (MM.img auf SD-Karte) -> alles io, MM bootet
```

**Meine Einschätzung:**
Ich denke, dass das Problem besteht weil der RasPi4 die Konvertierung aller RasPi's durchführt. Wenn Jeder Raspi sich selber konvertieren würde, würde das Boot-Problem vielleicht nicht bestehen.

Ich mache noch weitere Tests und halte euch auf dem Laufenden.

Liebe Grüsse
Peter

framps, 2021-09-01:
> Wenn Jeder Raspi sich selber konvertieren würde, würde das Boot-Problem vielleicht nicht bestehen.

Ich habe mir den Code von raspiBackuprestore2Image noch mal danach genauer angesehen. Da pishrink die PARTUUID aendert wird sie zum Schluss noch mal korrigiert.

```
# pishrink destroyes PARTUUID with resizsefs, restore original PTUUID now

LOOP=$(losetup -f)
losetup -P $LOOP $IMAGE_FILENAME
printf "x\ni\n0x$PTUUID\nr\nw\nq\n" | fdisk $LOOP
```

Das heisst dass raspiBackupRestore2Image.sh nicht mit LABELs und UUIDs in /etc/fstab und /boot/cmdline.txt funktioniert. Ich werde in raspiBackupRestore2Image einen Check einbauen dass LABELs und UUIDs nicht unterstuetzt sind. 

Aber in Deinen Logs ist zu sehen dass Du PARTUUIDs in den Dateien hast. D.h. von der Seite ist alles OK.

Es koennte noch sein dass pishrink da irgendwie reinfunkt. 

@Lusbueb69 Koenntest Du vielleicht mal die Imagemiogration von MM auf der Raspi4 vornehmen lassen OHNE pishrink zu nutzen und testen ob der Restore bootet?

Dazu muesstest Du raspiBackupRestore2Image ein klein wenig aendern:

```
# restore backup into image

echo "===> Restoring backup into $IMAGE_FILENAME"
raspiBackup.sh -1 -Y -d $LOOP "$BACKUP_DIRECTORY"
RC=$?
```

musst Du aendern in

```
# restore backup into image

echo "===> Restoring backup into $IMAGE_FILENAME"
raspiBackup.sh -1 -Y -d $LOOP "$BACKUP_DIRECTORY"
RC=$?
echo "Return code of raspiBackup: $RC"
exit
```
Der RC muss  0 sein.

Was Du auch noch machen koenntest: Mal ueberpruefen ob die PARTUUIDs im Image OK sind beim Image was von der Raspi4 fuer MM migriert wurde. 
Das geht wie folgt:

```
1) kpartx mit `sudo apt-get install kpartx` installieren
2) sudo kpartx -av <imagefile>
3) sudo blkid
4) sudo mount /dev/mapper/loop0p1 /mnt
5) cat /mnt/boot/cmndline.txt
6) sudo umount /mnt
7) sudo mount /dev/mapper/loop0p2 /mnt
8) cat /mnt/etc/fstab
9) sudo kpartx -d <imagefile>
```

Dann sollte in (3), (4) und (7) immer dieselbe PARTUUID auftauchen.

framps, 2021-09-01:
> Hi, zu den Versionen, ich habe 0.6.6 und 0.6.6.1B getestet und beide funktionieren (bei mir) leider nicht.

@KleBoR Nutzt Du PARTUUIDs in /etc/fstab und /boot/cmdline.txt ?

KleBoR, 2021-09-01:
@framps ja in einem Image welches nicht bootet sind Partuuids. habe die Zuweisung schon abgeändert - dann bootet das Image weiter bis zu dem Fehler das der root acc locked ist wodurch sämtliche Services nicht starten. 
Ich habe vorhin keine Zeit mehr gehabt das defekte Image mit einem funktionablen zu testen - ja, ich habe unter all den "tausend" verschiedenen Pi Images ein funktionables entdeckt welches irgendwann mal mit dem Wrapper erstellt wurde.

Ich gehe der Partuuid nochmal genauer nach - mich wundert nur das Pishrink kein problem mit Partuuids hat sofern das Image mit Etcher oder win32diskimager erstellt wurde..

framps, 2021-09-01:
> habe die Zuweisung schon abgeändert

D.h. Du hast festgestellt dass die PARTUUIDs in /etc/fstab und /boot/cmdline.txt nicht stimmen? Wie hast Du sie geaendert?

>  mich wundert nur das Pishrink kein problem mit Partuuids hat sofern das Image mit Etcher oder win32diskimager erstellt wurde..

Das dd Image wird mit raspiBackupRestore2Image erstellt - nicht mit Etcher oder win32diskimager. Bislang gab es nie Probleme mit dem erstellten Image. irgendwas scheint sich geaendert zu haben. Was weiss ich noch nicht. Wir suchen gerade gemeinsam :wink: .

KleBoR, 2021-09-01:
> D.h. Du hast festgestellt dass die PARTUUIDs in /etc/fstab und /boot/cmdline.txt nicht stimmen? Wie hast Du sie geaendert?
ich habe die Partuuids rausgenommen und gegen mmcblk0p2 ersetzt sowie ebenfalls die ursprünglichen ausgetauscht. Beidesmal bootet der Pi weiter - jedoch eben nicht komplett.
> Das dd Image wird mit raspiBackupRestore2Image erstellt - nicht mit Etcher oder win32diskimager. Bislang gab es nie Probleme mit dem erstellten Image. irgendwas scheint sich geaendert zu haben. Was weiss ich noch nicht. 
mir ist bewusst das die images durch raspiBackupRestore2Image erstellt werden - ging mir nur um den Punkt das es an pishrink wohl nicht liegen kann da die Images von Etcher sowie win32disk.. ohne Probleme bootbar sind nach dem verkleinern.
> Wir suchen gerade gemeinsam 😉 .
Finden wir auch sicher heraus ;)

Lusbueb69, 2021-09-01:
@framps bei mir hat der Code noch **RC=0** drin stehen..
```
# restore backup into image

echo "===> Restoring backup into $IMAGE_FILENAME"
raspiBackup.sh -1 -Y -d $LOOP "$BACKUP_DIRECTORY"
RC=$?
RC=0

# The disk identifier is the Partition UUID (PTUUID displayed in blkid) and is stored just prior to the partition table in the MBR
```
Habe es jetzt nach deiner Vorgabe angepasst und werde nach dieser Nacht mal ein MM-Restore fahren (ohne pishrink) :-)

**kpartx** mit dem MM-Image von letzter Nacht, welches **nicht bootet** ergibt folgendes (leicht angepasste Befehlsfolge):
```pi@RasPi4:~ $ sudo blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="F661-303B" TYPE="vfat" PARTUUID="3914b778-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="8d008fde-f12a-47f7-8519-197ea707d3d4" TYPE="ext4" PARTUUID="3914b778-02"
/dev/mmcblk0: PTUUID="3914b778" PTTYPE="dos"

pi@RasPi4:~ $ sudo kpartx -av /backup/MM/MM-tar-backup-20210901-003102/MM-full-image--21-09-01_032325.img
add map loop0p1 (254:0): 0 88472 linear 7:0 8192
add map loop0p2 (254:1): 0 12055809 linear 7:0 98304

pi@RasPi4:~ $ sudo blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="F661-303B" TYPE="vfat" PARTUUID="3914b778-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="8d008fde-f12a-47f7-8519-197ea707d3d4" TYPE="ext4" PARTUUID="3914b778-02"
/dev/loop0: PTUUID="3914b778" PTTYPE="dos"
/dev/mmcblk0: PTUUID="3914b778" PTTYPE="dos"
/dev/mapper/loop0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="6228-7918" TYPE="vfat" PARTUUID="3914b778-01"
/dev/mapper/loop0p2: UUID="afbd33ad-d132-4e9c-aa7c-91f07c83983f" TYPE="ext4" PARTUUID="3914b778-02"

pi@RasPi4:~ $ sudo mount /dev/mapper/loop0p1 /mnt

pi@RasPi4:~ $ cat /mnt/cmdline.txt
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=0f464416-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait ipv6.disable=1

pi@RasPi4:~ $ sudo umount /mnt

pi@RasPi4:~ $ sudo mount /dev/mapper/loop0p2 /mnt

pi@RasPi4:~ $ cat /mnt/etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=0f464416-01  /boot           vfat    defaults          0       2
PARTUUID=0f464416-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

pi@RasPi4:~ $ sudo umount /mnt

pi@RasPi4:~ $ sudo kpartx -d /backup/MM/MM-tar-backup-20210901-003102/MM-full-image--21-09-01_032325.img
loop deleted : /dev/loop0

pi@RasPi4:~ $ sudo blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="F661-303B" TYPE="vfat" PARTUUID="3914b778-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="8d008fde-f12a-47f7-8519-197ea707d3d4" TYPE="ext4" PARTUUID="3914b778-02"
/dev/mmcblk0: PTUUID="3914b778" PTTYPE="dos"

pi@RasPi4:~ $
```

**kpartx** mit dem MM-Image von 2019, welches **bootet** ergibt folgendes:
```pi@RasPi4:~ $ sudo blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="F661-303B" TYPE="vfat" PARTUUID="3914b778-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="8d008fde-f12a-47f7-8519-197ea707d3d4" TYPE="ext4" PARTUUID="3914b778-02"
/dev/mmcblk0: PTUUID="3914b778" PTTYPE="dos"

pi@RasPi4:~ $ sudo kpartx -av /backup/MM/MM-tar-backup-20190927-010102/MM-full-image--19-09-27_044819.img
add map loop0p1 (254:0): 0 88472 linear 7:0 8192
add map loop0p2 (254:1): 0 11738281 linear 7:0 98304

pi@RasPi4:~ $ sudo blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="F661-303B" TYPE="vfat" PARTUUID="3914b778-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="8d008fde-f12a-47f7-8519-197ea707d3d4" TYPE="ext4" PARTUUID="3914b778-02"
/dev/loop0: PTUUID="bd98648d" PTTYPE="dos"
/dev/mmcblk0: PTUUID="3914b778" PTTYPE="dos"
/dev/mapper/loop0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="6228-7918" TYPE="vfat" PARTUUID="bd98648d-01"
/dev/mapper/loop0p2: UUID="dc5dec44-71b3-4736-a847-6ed442fda44c" TYPE="ext4" PARTUUID="bd98648d-02"

pi@RasPi4:~ $ sudo mount /dev/mapper/loop0p1 /mnt

pi@RasPi4:~ $ cat /mnt/cmdline.txt
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=bd98648d-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait ipv6.disable=1

pi@RasPi4:~ $ sudo umount /mnt

pi@RasPi4:~ $ sudo mount /dev/mapper/loop0p2 /mnt

pi@RasPi4:~ $ cat /mnt/etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=bd98648d-01  /boot           vfat    defaults          0       2
PARTUUID=bd98648d-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

pi@RasPi4:~ $ sudo umount /mnt

pi@RasPi4:~ $ sudo kpartx -d /backup/MM/MM-tar-backup-20190927-010102/MM-full-image--19-09-27_044819.img
loop deleted : /dev/loop0

pi@RasPi4:~ $ sudo blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="F661-303B" TYPE="vfat" PARTUUID="3914b778-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="8d008fde-f12a-47f7-8519-197ea707d3d4" TYPE="ext4" PARTUUID="3914b778-02"
/dev/mmcblk0: PTUUID="3914b778" PTTYPE="dos"

pi@RasPi4:~ $

```

Nimms mir nicht übel, aber vor lauter Bäumen sehe ich den Wald nicht mehr 8-)

KleBoR, 2021-09-02:
@framps ich habe die Images verglichen und sämtliche Punkte gecheckt. Was mir aufgefallen ist, das die Partuuid in der cmdline.txt sowie fstab gleich ist - sich jedoch von der blkid gelisteten unterscheidet! 
Diese editiert resultiert in einem augenscheinlich erfolgreichem Bootvorgang. Ich habe noch nicht alle Services des Images getestet - daher vorerst augenscheinlich.

framps, 2021-09-03:
Vielen Dank fuer Eure Tests und Hilfe die Ursache des Bootproblems zu lokalisieren :+1: . Irgendwo liegt der Hase in der UUID Ecke :cry: . Nun liegt es an mir die Ursache der ursache - sprich den Codefehler zu finden :wink: 

Ich halte Euch auf dem Laufenden.

framps, 2021-09-05:
@Lusbueb69 Eben habe ich Dein Scenario nachgestellt und das auf der remoten Raspi4 erstellte Image bootet nicht - das auf dem Backupsystem aber schon. Es hat aber nichts mit den UUIDs zu tun. Der Grund muss irgendwo anders liegen. Ich koennte mir vorstellen dass es irgendwie an der Raspi4 HW liegt. Muss das aber verifizieren und weiter suchen.

Kurzum:  Ich kann Dein Problem reproduzieren was schon mal sehr gut ist denn damit kann ich bei mir nach der Ursache suchen. Jetzt suche ich mal weiter ...

Lusbueb69, 2021-09-06:
vielen Dank @framps

framps, 2021-09-06:
@Lusbueb69 

> * RasPi4 (buster): bootet nicht.
> 
> * MM (stretch): bootet nicht.

Hm ... das ist natuerlich auch ein moeglicher Grund unterschiedliche Distros zu haben. Ich habe in meinem Test aber auf  einer Raspi3 mit Buster gesichert und auf einer Raspi4 mit Buster komprimiert. 

> Ich lass nun mal die Finger davon,

Jupp. Lass mich mal machen. Das wird aber vermutlich etwas dauern. Also bitte etwas Geduld.

Wenn das Image auf derselbeb Maschine gebaut wird wo das Backup erstellt wurde muss es aber funktionieren. irgendwo bist Du da vermutlich durcheinandergekommen. Das passiert leicht wenn man mit verschiedenen Kombinationen von Raspis und Images rumprobiert :wink:

framps, 2021-09-06:
Ich habe einen bloeden Fehler im Script gefunden :disappointed: und gefixed. Damit habe ich ein Image was auf einer Raspi3 mit Buster erstellt wurde auf einer Raspi4 mit Buster konvertiert und dann per dd auf eine SD Karte geschrieben. Es bootet erfolgreich.

Ich moechte jetzt also @Lusbueb69 und @KleBoR bitten sich den letzten Stand (Version 1.7) von raspiBackupRestore2Image aus dem git Repo zu holen und meinen Fix zu verifizieren :wink: 

@Lusbueb69 Ich habe kein Stretch mehr um exakt Dein Szenario nachzustellen. Jetzt hoffen wir mal dass es bei Dir trotzdem funktioniert.

KleBoR, 2021-09-06:
@framps vielen Dank für den schnellen Fix! Ich teste v1.7 und gebe bescheid. Gruss

Lusbueb69, 2021-09-07:
@framps auch von meiner Seite ein herzliches Dankeschön. Werde voraussichtlich morgen Abend testen und die Ergebinisse hier mitteilen.

Liebe Grüsse
Peter

Lusbueb69, 2021-09-08:
**HEUREKA!** du bist mein Held @framps !

Mein MM (RasPi  3B+, Stretch) startet einwandfrei nach der Migration von TAR zu IMG, bravo!

Vielen, vielen Dank, ich bin mega happy :)

framps, 2021-09-08:
Perfekt. Das freut mich :smile: War ein bloeder Fehler den man aber nur bemerkt wenn man raspiBackupRestore2Image auf einem anderen System wie Du ausfuehrt. Es ist immer wieder interessant auf welche Anwendungsszenarien Ihr raspiBackup Nutzer kommt :wink: . Aber Dein Szenario ist durchaus pfiffig und legitim. 

Jetzt hoffe ich von @KleBoR kommt auch weisser Rauch :crossed_fingers: 

PS: Alle Script aus dem Helperverzeichnis sind nicht Teil von dem raspiBackup Release und somit Beispiele fuer raspiBackup Erweiterungen und eigentlich nicht unterstuetzt da sie auch nicht im Regressionpackage enthalten sind. Aber es hat mich doch interessiert worin die Ursache liegt ... 

Auch noch mal vielen Dank fuer Deine Ausdauer und Hilfe die Ursache zu finden. Sowas ist leider zeitintensiv und bleibt meist an mir haengen :disappointed:

KleBoR, 2021-09-20:
Hi @framps ,
entschuldige die späte Rückmeldung - ich war im Urlaub.
Ich habe gestern Abend nochmal ein Backup mit v1.7 gemacht - leider selbes Ergebnis bei mir.
Habe das Image auch nicht Device übergreifend getestet sondern von einem RPI3 und im selbigen auch getestet.

Ich erstelle gerade nochmal eines ohne PiShrink um den Fehler zu finden - die Partuuids sind bei meinem Image nun bei fstab sowie cmdline anders als beim Original.

Gruss

closed 2021-11-21 by github-actions[bot]