framps/raspiBackup #359: Cant restore backup !

TotalCoolness, 2021-08-22:
I have a raspberry pi booting from NVME SSD via USB. 

I create the backup via .tar method and store onto a windows samba share. 

When ever I try to restore to test the backup it never boots via my spare NVME drive. It just attempts to boot and keep trying ( I also tried SD card.). No screen messages nothing. Just one flash of the SSDs light. Its like it doesnt like the MBR or something.

[raspiBackup.log](https://github.com/framps/raspiBackup/files/7027367/raspiBackup.log)

framps, 2021-08-24:
Would you please attach the debug log of the restore?

TotalCoolness, 2021-08-24:
Hi framps, 

Is it the raspiBackup.msg file ? Is that the log of the restore ?

Here:

[raspiBackup.txt](https://github.com/framps/raspiBackup/files/7039961/raspiBackup.txt)

framps, 2021-08-24:
raspiBackup creates raspiBackup.log in the current directory when a backup is restored.

TotalCoolness, 2021-08-24:
> 
> 
> raspiBackup creates raspiBackup.log in the current directory when a backup is restored.

See the first post for raspiBackup.log.

and my second post for the other file raspiBackup.msg .

framps, 2021-08-24:
The logfile on your first comment is a logfile of a backup run and doesn't help me. The message file tells me the restore failed. I thought that's some error on your side and the restore succeeded. If the restore indeed failed the restored image will not boot. 

If you don't provide the restore log there is no way for me to help you.

TotalCoolness, 2021-08-24:
Am I correct in saying you got two different log files that are both called raspiBackup.log ? But they have different content ?

One has Restore log info and one has Backup log info ? This is what was confusing me cos I think I deleted one of them cos I assumed it was duplicate since they both have the same name. 

Let me redo the restore..

TotalCoolness, 2021-08-24:
Ok here it is. 

[raspiBackup-stage2.log](https://github.com/framps/raspiBackup/files/7042213/raspiBackup-stage2.log)

framps, 2021-08-30:
Boot issues on a Raspberry are very hard to debug :cry: . 

Please attach a monitor and provide a screenshot when boot hangs. You may also read [this page](https://www.raspberrypi.org/forums/viewtopic.php?t=58151). Maybe you get some idea what's wrong and can give me some hints?

TotalCoolness, 2021-08-31:
OK, so I tried changing a few things last night.

I used the DD method and it created a 256GB file (the size of my NVME drive) which I wrote with Windows Raspberry Pi Imager and it worked !

When I tried this before it didnt boot. Last time I used SD cards and Raspibackup to also write the image (this time I used windows). 

So maybe some issue with the way Raspibackup creates the partitions ?? I dont know. It took a long time as it backed up 256GB and wrote it and then verified it. 

Tar method is better as the filesize is nice and small 12GB vs 256 GB but doesnt boot. I did find another util online that can create backups perfectly and uses rsync and makes images only 14GB. They boot fine.

Yes its hard to find the problem, maybe if you buy a nvme enclosure (with realtek chipset) and small nvme drive you could nail the issue. I dont think nvme is the issue as i had problems with SD cards too.

framps, 2021-08-31:
> When I tried this before it didnt boot. Last time I used SD cards and Raspibackup to also write the image (this time I used windows).

Hm ... you run a system from NVME only and restored the backup to an 256GB SD card? Actually there is no difference between the way the Imager and raspiBackup with dd restore a dd image. I never had somebody who restored to a 256GB SD card - but I had raspiBackup users which restored to a 8GB or 16GB SD card and had boot issues. Finally it turned out for all of them the restored system boots successfully when they used a new SD card. If the restore is done on a buggy SD card that's what can happen. 

Did you also restore the tar backup to this 256GB SD card? 

> Yes its hard to find the problem, maybe if you buy a nvme enclosure (with realtek chipset) and small nvme drive you could nail the issue. I dont think nvme is the issue as i had problems with SD cards too.

I wrote raspiBackup for my local environment which still uses SD cards only. Given the fact USB boot was enabled I added support for USB boot because I was able to by a Raspi3 from the donations I got for raspiBackup.  
Looks like there is something special in your environment. Frankly I have no problem to use a NVME in my tests but I don't need an NVME and thus only by one if the donations I get for raspiBackup can be used to populate my test equipment. 

> I did find another util online that can create backups perfectly and uses rsync and makes images only 14GB. They boot fine.

Then please use this utility and close the issue or donate and I can buy a NVME to test raspiBackup with a NVME.

TotalCoolness, 2021-08-31:
> 
> > When I tried this before it didnt boot. Last time I used SD cards and Raspibackup to also write the image (this time I used windows).
> 
> Hm ... you run a system from NVME only and restored the backup to an 256GB SD card? Actually there is no difference between the way the Imager and raspiBackup with dd restore a dd image. I never had somebody who restored to a 256GB SD card - but I had raspiBackup users which restored to a 8GB or 16GB SD card and had boot issues. Finally it turned out for all of them the restored system boots successfully when they used a new SD card. If the restore is done on a buggy SD card that's what can happen.
> 
> Did you also restore the tar backup to this 256GB SD card?
> 
> > Yes its hard to find the problem, maybe if you buy a nvme enclosure (with realtek chipset) and small nvme drive you could nail the issue. I dont think nvme is the issue as i had problems with SD cards too.
> 
> I wrote raspiBackup for my local environment which still uses SD cards only. Given the fact USB boot was enabled I added support for USB boot because I was able to by a Raspi3 from the donations I got for raspiBackup.
> Looks like there is something special in your environment. Frankly I have no problem to use a NVME in my tests but I don't need an NVME. I only support HW I don't use when I'm able to buy it with donations I get for raspiBackup and can populate my test equipment.
> 
> > I did find another util online that can create backups perfectly and uses rsync and makes images only 14GB. They boot fine.
> 
> Then please use this utility and close the issue or donate and I can buy a NVME to test raspiBackup with a NVME.

I boot from a 256GB USB NVME drive exclusively. I restored to another 256GB nvme drive via windows. tar method didnt work.
I also tried booting via an SD card then restored to a blank nvme and also restoring to another SD card. Again no boot. 
I also tried SD card to SD card.. both DD & tar method didnt work. Today is the only time it booted my NVME via DD method. 

Im confident my hardware is ok.  Theres nothing special about my setup I can assure you, its simple: A raspberry PI 4 4GB with a few extra standard software like Pi Hole and Docker.

Anyway all the best.

closed 2021-08-31 by TotalCoolness

framps, 2021-09-05:
Just for the records: NVMe uses a different device name than /dev/mmcblk which is used for SSD and /dev/sda which is used for for disks. See https://superuser.com/questions/1449499/why-does-linux-list-nvme-drives-as-dev-nvme0-instead-of-dev-sda. That's why raspiBackup doesn't support NVMe because of missing hardware to add NVMe support and tests.

TotalCoolness, 2021-09-05:
Yes thats an interesting read. I am connected via a USB to NVMe enclosure so it sees it as USB.

Actually I got a new theory as to why there's problems.  I originally had noobs on it and that has more than the standard partitions, but I got rid of noobs using this software usb-boot  ( https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=196778 ) so I think even though there are 2 partitions it did something differently to the partitions. 

Because using this script call "image-backup" ( https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=247568 ) it can back it up and they boot fine as its written by the same guy who made usb-boot.

framps, 2021-09-05:
> I am connected via a USB to NVMe enclosure so it sees it as USB.

You are right. In your log I see

```
20210822-125543 DBG 5056:              /dev/sda2 on / type ext4 (rw,noatime,stripe=4)
...
20210822-125543 DBG 5056:              /dev/sda1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
```
I'm not familiar with NVMes and when I found this link I didn't cross check your log :cry: 

> so I think even though there are 2 partitions it did something differently to the partitions.

Even there are are a lot of debug statements in raspiBackup it's very difficult for me to find the root cause of boot issues. So you may be right. The regression test of raspiBackup uses a plain RaspberryPiOS light on a simulated Raspberry. There are a lot of users out there who use different HW and SW and can use raspiBackup successfully. It's untested by me and thus an unsupported environment but the users like raspiBackup. Because of this I hesitate to add checks in raspiBackup for RaspiOS SW and Raspberry HW to reject any unsupported HW/SW combination. 

This doesn't help you and me in this issue - unfortunately :cry: 

Thank you very much for the links to the tools you use successfully. It's a nice resource for me to check whether there are better/other ways to create a backup.