framps/raspiBackup #364: FI NLS Support

teemue, 2021-09-13:
I'm just about to create an initial PR for raspiBackup.sh translation with some of the lines translated.

framps, 2021-09-16:
@teemue Thank you very much to the add Finnish language support to raspiBackup :+1: 

Do you plan to add Finnish support also in raspiBackupInstallUI.sh in another PR?

teemue, 2021-09-16:
Yes, I do. I'll PR it in a few days 👍

framps, 2021-09-16:
Great. Take your time. You may need some more time than with raspiBackup to translate all the messages - in particular the messages which describe the options. I initially had some trouble to find the right box sizes for Whiptail and finally came up with
```
# Whiptail box sizes

ROWS_MSGBOX=10
ROWS_ABOUT=16
ROWS_MENU=20
WINDOW_COLS=70
```
Please try to create FI text which is not longer than the DE and EN text. If it's not possible and the FI text is truncated in a box please let me know. I then revisit the Whiptail box sizes.

teemue, 2021-09-19:
Translation is almost done and before I PR, I test the script thoroughly.

I noticed that when the menu item "C3 Backup versions" of major options is selected, the "C5 Backup mode" is shown as a title - is this a bug ? :)

framps, 2021-09-20:
> - is this a bug ? :)

Yes. You test very thoroughly :+1: I fixed it right now in the beta branch. Up to you whether you want to checrry pick https://github.com/framps/raspiBackup/commit/3edda3211db251a1298e4bfe9e5969c825889222 :wink:

framps, 2021-09-24:
By mistake I already merged the NLS code in bet 0.6.6.1. Given this I plan to merge the beta_nls branch into beta 0.6.6.1 as soon as I thoroughly tested the NLS code in beta.

framps, 2021-09-25:
Done with https://github.com/framps/raspiBackup/pull/370. Great work @teemue . Thank you very much for your support :+1:

closed 2021-09-25 by framps

teemue, 2021-09-25:
Thanks for the great app  :-)