framps/raspiBackup #366: (Feature request) RSync generate new full backup every X backups

p1r473, 2021-09-14:
Hello,
I would like to use RSync but I would like to be able to generate a new full backup every X backups
This is because if anything goes wrong with one of the backups, subsequent backups may have issues. This is a problem with RSync backups because each backup relies on previous backup to work.
So, because of this, I wish to every X backups make a new RSync full backup.

Here is an example of Acronis for Windows. I use this feature and enjoy it.
[image]

* Provide the output of `sudo raspiBackup.sh --version` or `sudo raspiBackupInstallUI.sh --version`
sudo raspiBackup.sh --version
* Provide the output of `cat /etc/os-release`

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

framps, 2021-09-15:
>  **would like to be able to generate a new full backup every X backups**

I see your point. Until now nobody missed this option - maybe they were not aware about the issue you bring on the table with rsync and hardlinks. I think it's a useful new option :+1:  . I will call the new option `RSYNC_FULL_BACKUP_INTERVAL`

> would be nice to have an option where all of the weekly, monthly, and yearly are FULL backups.

If you set the new requested interval `RSYNC_FULL_BACKUP_INTERVAL` to 7 every weekly backup will become a full backup. And every other backup (monthly, yearly) will have at most 6 incremental backups but then a full backup. I see this as a possible low hanging fruit workaround for you :wink:

p1r473, 2021-09-15:
This is great news. Thank you.

closed 2021-11-13 by framps