framps/raspiBackup #369: Removing raspiBackup to start again with a fresh install

425629, 2021-09-20:
I am posting here to follow the directions given to me by framps to move from having this conversation on the extensions page of the raspiBackup web documentation.

I must've messed things up - perhaps by executing some operations logged on as root, rather than using sudo.  I've only configured raspiBackup to use rsync to back up to a USB-connected flash drive - which worked wonderfully.  I'm now wanting to learn how to use the extensions so I can trigger a python script as part of the execution.  I was trying to get there by first trying to get the default extensions to run - raspiBackup.0.6.5.1_mem_pre.sh, raspiBackup.0.6.5.1_disk_pre.sh, etc.  I know my versions are messed up so now I'm trying to remove the mess and start fresh.  Replicating the configuration to rsync to the USB stick is of course trivial.

Thanks for your time.

Dave Neill

* Read raspiBackup FAQ's
I searched the FAQ for "uninstall" and "remove" so I might get to a point where I can simply use the tools provided to install the latest version and the extensions that are appropriate to that version.

* Provide the output of `sudo raspiBackup.sh --version` 
Version: 0.6.6 CommitSHA: 262862c CommitDate: 2021-08-06 CommitTime: 10:15:00
* or `sudo raspiBackupInstallUI.sh --version`
/usr/local/bin/raspiBackupInstallUI.sh: illegal option -- -
raspiBackupInstallUI.sh 0.4.3.5, 2021-08-24/21:31:19 - 65352d6
raspiBackupInstallUI.sh ( -i [-e]? | -u | -U ) [-d]?
-d: enable debug mode
-e: unattended (re)install of raspiBackup extensions
-i: unattended (re)install of raspiBackup
-U: unattended update of raspiBackupInstallUI.sh
-u: unattended uninstall of raspiBackup

* Provide the output of `cat /etc/os-release`

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

If any action was not executed or any requested information was not provided the question will be rejected.

framps, 2021-09-24:
How to plugin in an extension is [described on this page](https://www.linux-tips-and-tricks.de/en/13-raspberry/443-raspibackup-extensions/).

I suggest to start over and install the extension samples to get familiar with the extensions.

Uninstall raspiBackup:
`sudo raspiBackupInstallUI -u`

Install raspiBackup:
`curl -sL https://raspibackup.linux-tips-and-tricks.de/install | sudo bash`
to install raspiBackup again and M2 -> I2

Then read sample exits in /usr/local/bin starting with `raspiBackup_`. If there are any further questions don't hesitate to ask.

framps, 2021-10-02:
@425629 Any update on this issue? :wink:

425629, 2021-10-02:
Wow!  Thanks for staying in touch.  I did a fresh install - latest stable release - I know you said go to the 06 beta but the stable seemed enough. I enabled the extensions and was able to see the default extensions on memory and disk and ? operate correctly. I haven't gotten to an attempt to engage my own extension. That's where I'm at.  Thank you so much for reaching out!

Sent from AT&T Yahoo Mail on Android

framps, 2021-10-03:
Great it works now. Please close the issue and open a new one if you face any new questions or issues.