framps/raspiBackup #371: Changing Cron date not saving

p1r473, 2021-09-22:
Hi, seems to be some issue changing the Cron date
I am changing my backup strategy from Sundays to Daily
I go through the steps, select daily, click OK, click finish. However, I am not prompted to change the Cron date. When I next load up raspiBackupInstallUI, it is still Sunday. So, **my changes did not save,** even though I selected finish

No debug log was generated as I havent ran the backup yet. Still configuring.

framps, 2021-09-24:
I was able to recreate the issue with
`raspiBackupInstallUI 0.4.3.5, 2021-08-24/21:31:19 - 65352d6`

framps, 2021-09-24:
I just fixed the issue in https://github.com/framps/raspiBackup/tree/master_371. Please verify the fix.

p1r473, 2021-09-24:
Any documentation how to upgrade to that branch to test? Thanks

framps, 2021-09-25:
Hm ... unfortunately not. Looks like I should add another page on my website ...

Execute following steps:

1) Logon to your Raspberry and download the fix with `wget https://raw.githubusercontent.com/framps/raspiBackup/master_371/installation/raspiBackupInstallUI.sh`
2) `chmod +x ./raspibackupInstallUI.sh`
3) Execute and test the fix with `sudo ./raspiBackupInstallUI.sh`

Please note the leading punctuation mark in the commands.

p1r473, 2021-09-25:
I confirm the fix works. I can now change the date.
HOWEVER the popup asking if I want to change the Cronjob does not pop up. (Was I supposed to get the Front popup?)
I do get the popup asking if I want to save my changes though

framps, 2021-09-25:
> HOWEVER the popup asking if I want to change the Cronjob does not pop up. (

I'm not able to reproduce this. I get the popup all the time. Would you please use `sudo ./raspiBackupInstallUI.sh -d` and recreate the issue? Please attach the debug file `raspiBackupInstallUI.log` which will exist in current directory.

p1r473, 2021-09-25:
I tested it again and it worked fine- possible my first test was not run correctly
Thanks!

closed 2021-09-25 by p1r473