framps/raspiBackup #373: Error on restore

juanca1944, 2021-10-01:
Please execute all following actions first and provide all requested information:

* Check whether my error messages were listed on the messages page and all suggested actions don't fixes my issue. [English](https://linux-tips-and-tricks.de/en/rmessages) - [German](https://linux-tips-and-tricks.de/de/fehlermeldungen)
* Attached raspiBackup debug log

If any action was not executed or any requested information was not provided the issue will be rejected.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/7268773/raspiBackup.log)

framps, 2021-10-01:
I checked the log and found the root cause for the issue: Partitioning of /dev/sdb fails.
```
Checking that no-one is using this disk right now ... OK

Disk /dev/sdb: 74.5 GiB, 80026361856 bytes, 156301488 sectors
Disk model: 0BB-22JHC0      
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x3236e643

Old situation:

Device     Boot Start       End   Sectors  Size Id Type
/dev/sdb1        2048 156301311 156299264 74.5G 83 Linux

>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Last LBA specified by script is out of range.
Last LBA specified by script is out of range.
Last LBA specified by script is out of range.
Failed to apply script headers, disk label not created.
Leaving.
```

Contents of the sfdisk input file:

```
20210930-175823 DBG 2954:                  *** cat /media/pi/a70c0951-ef5c-421a-9e19-6a75dc23942f/@HOSTNAME@/@HOSTNAME@-rsync-backup-20210930-045755/@HOSTNAME@-backup.sfdisk
20210930-175823 DBG 2954:                      label: gpt
20210930-175823 DBG 2954:                      label-id: 2E0854D9-D9E6-4C0E-9CE9-FAA9F55117E5
20210930-175823 DBG 2954:                      device: /dev/sda
20210930-175823 DBG 2954:                      unit: sectors
20210930-175823 DBG 2954:                      first-lba: 34
20210930-175823 DBG 2954:                      last-lba: 7813969886
20210930-175823 DBG 2954:                      
20210930-175823 DBG 2954:                      /dev/sda1 : start=        2048, size=      524288, type=EBD0A0A2-B9E5-4433-87C0-68B6B72699C7, uuid=3936FF3E-5D39-492C-8387-C8F27911CD0B
20210930-175823 DBG 2954:                      /dev/sda2 : start=      526336, size=    63488000, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=89233227-EFE8-4195-B546-4CE0D497BBA4
```

You use a gpt formatted disk. Right now raspiBackup supports a standard RaspberryOS which uses mbr but no gpt. Said this I unfortunately have to reject your issue because you have an unsupported  environment. I just created on behalf of you a [feature request to support gpt](https://github.com/framps/raspiBackup/issues/374).

I don't know whether you have to use gpt. If not you may switch to mbr which is supported with raspiBackup :wink: 

Please close the issue if you don't have any further comment or question about gpt support of raspiBackup.

juanca1944, 2021-10-01:
Hello Framps,
Thanks a lot. I am using GPT because my RaspiOS is installed in a 4 TB HDD.
Regards

framps, 2021-10-02:
> 4 TB HDD

This indeed requires gpt. So you cannot use raspiBackup - unfortunately. I create an issue but given the fact gpt usage with raspiBackup is rarely used there are no plans to add this support as of now.

juanca1944, 2021-10-02:
Halo Framps.
Well, I am going to think in something else.
Vielen Dank und alles gute.
Grüsse

closed 2021-10-02 by juanca1944