framps/raspiBackup #374: Support gpt

framps, 2021-10-01:
As of now the standard partitioning used by RaspberryOS is mbr. Therefore raspiBackup was developped to backup mbr disks/sds. There is no support for gpt.

ylexus, 2025-03-27:
Hi @framps , have you added the proper support for GPT now? I still use the workaround we discussed in the original issue I raised. Can I now upgrade to latest? I can't see anything related commits...

ylexus, 2025-03-27:
Can I please ask you to make at least the harmless fix, here's the PR: https://github.com/framps/raspiBackup/pull/862

This is a small step towards the full support of GPT, but it is at least something that fixes the problem in my circumstances and it is fully backwards compatible, so there is no harm.

framps, 2025-03-27:
When I did some initial investigations on the gpt support I also stumbled upon this line. gpt uses no longer numbers for partition types and therefore the test doesn't work any more. Unfortunately that's not the only place raspiBackup has to be updated in order to support gpt.

You PR accepts strings and numbers for the type and is identical from a semantic point of view with the actual code. 

Does this mean you can backup and restore a gpt based system with raspiBackup with this fix?

ylexus, 2025-03-27:
I can definitely backup a GPT partition with this fix. It's not identical to the actual code because it now accepts GPT GUID style partition types which are not numbers as in MBR. We discussed this here: https://github.com/framps/raspiBackup/issues/850#issuecomment-2722874172. 

I can definitely backup successfully with this change now and I can see it creates the .tgz file with correct contents and size. I have not tried restoring it yet - trying now, will report the results here.

framps, 2025-03-27:
> will report the results here.

Thx. Crossing fingers - even gpt is not officially supported ;-)

ylexus, 2025-04-21:
OK so the last-lba problem is this: https://github.com/util-linux/util-linux/issues/3186. It's been fixed in September last year and is in util-linux 2.41. Latest RaspiOS only comes with 2.38. Until then we can delete this header (which seems to be added by sfdisk for GPT partitions only), or we can just wait until 2.41+ is available.

framps, 2025-04-21:
Good catch 👍 Question now is how to handle this. There is already code which strips some line in the sfdisk file ([See here](https://github.com/framps/raspiBackup/blob/6c31e8c016380f608ba795c59e034a95a8e7a13b/raspiBackup.sh#L6396))

Would be great if you have a nice proposal to handle this. I don't want to have any version dependencies in raspiBackup. Otherwise we have to wait for 2.4.1.   

In addition the `--force` option is not used in sfdisk as of now and also has to be updated if we wait for 2.4.1.

ylexus, 2025-04-21:
I can see that you run `sfdisk -f` always, so which is the same as --force. 

I have added code that strips `first-lba:|last-lba:` headers from the sfdisk file, both are optional and not needed by the looks of it. However, I am still getting the following error from sfdisk. What is puzzling is why the restore code deals with partition 3 at all - the backup was made with "custom partitions" and with only partitions 1 and 2 selected, and the restore run says:

```
!!! RBK0305W: Restoring partitions "1 2" to /dev/sda
```

However, it still tries to create the 3rd partition. I am guessing the logic is that the restore script still creates all partitions, but just does not later restore the ones that are not needed - can you please confirm this is the intention? If so then it looks like there's some error in maths around the code that calculates the size of the last partition (when it is shrinked to match the smaller target drive).

The error is:
```
20250421-180633 DBG 4530:                      --- Updated sfdisk file
20250421-180633 DBG 4531:                          *** cat /tmp/1028739.sfdisk
20250421-180633 DBG 4531:                              label: gpt
20250421-180633 DBG 4531:                              label-id: B070687D-B700-440E-9D85-FBB8A9F48E47
20250421-180633 DBG 4531:                              device: /dev/nvme0n1
20250421-180633 DBG 4531:                              unit: sectors
20250421-180633 DBG 4531:                              sector-size: 512
20250421-180633 DBG 4531:                              
20250421-180633 DBG 4531:                              /dev/nvme0n1p1 : start=        8192, size=     1048576, type=EBD0A0A2-B9E5-4433-87C0-68B6B72699C7, uuid=A1D7A9AA-D2CD-4A61-A709-096793B25EEC, name="bootfs"
20250421-180633 DBG 4531:                              /dev/nvme0n1p2 : start=     1056768, size=   268435456, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=F768D69C-0A9D-4DB3-B08D-710F3FDA11A6, name="rootfs"
20250421-180633 DBG 4531:                              /dev/nvme0n1p3 : start=   269492224, size=  215474176, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=CE30B17F-4FB4-414C-BA13-37E26E214131, name="@HOSTNAME@"
20250421-180633 DBG 4534:                      --- Old: 3862806134784 (3.51 TiB) - New: 110322778112 102.74 GiB)
20250421-180633 DBG 4540:                  <-- createResizedSFDisk 3862806134784 110322778112 3
20250421-180633 MSG 6493:                  --- RBK0300I: Adjusting last partition from 3.51 TiB to 102.74 GiB 
Checking that no-one is using this disk right now ... OK

Disk /dev/sda: 231.25 GiB, 248302796800 bytes, 484966400 sectors
Disk model:                 
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: B164AFFC-D4ED-4DAD-9C97-6AADE6B88F7D

Old situation:

Device         Start       End   Sectors   Size Type
/dev/sda1       2048   1050623   1048576   512M Microsoft basic data
/dev/sda2    1050624 269486079 268435456   128G Linux filesystem
/dev/sda3  269486080 484964351 215478272 102.7G Linux filesystem

>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Created a new GPT disklabel (GUID: B070687D-B700-440E-9D85-FBB8A9F48E47).
/dev/sda1: Created a new partition 1 of type 'Microsoft basic data' and of size 512 MiB.
/dev/sda2: Created a new partition 2 of type 'Linux filesystem' and of size 128 GiB.
/dev/sda3: The last usable GPT sector is 484966366, but 484966399 is requested.
Failed to add #3 partition: Invalid argument
Leaving.
```

ylexus, 2025-04-21:
RE the PR - don't you first need to create the branch m_374? I am not super-proficient with GitHub, but I tried creating pull request from my fork's m_374 branch and it still only offers existing branches in your repo as target.

framps, 2025-04-22:
I'm sorry. Yes, you are right. I just created the branch.

> However, it still tries to create the 3rd partition. I am guessing the logic is that the restore script still creates all partitions, but just does not later restore the ones that are not needed - can you please confirm this is the intention?

Partitions are created per default and all of them are created. No partition is deleted from the sfdisk file. The last partition is truncated or expanded. 
If you use option -0 no partitioning is done - but the selected partitions are formatted. If you use option -00 on a partition oriented backup of type rsync the selected partitions are not formatted. The data is synced only. 

> I can see that you run `sfdisk -f` always, so which is the same as --force.

Ok. I missed this. I'm on vacation right now and just give quick support for the next couple of weeks. So in any case you have to wait until I'm back home and will be able to review your code ;-) But don't hesitate to ask questions. As long as I don't have to dig into the code I'll answer all your questions.

> If so then it looks like there's some error in maths around the code that calculates the size of the last partition (when it is shrinked to match the smaller target drive).

Are you sure?

framps, 2025-04-23:
> but it is related to the fact that GPT has 33 sectors reserved at the end of the disk for metadata.

Ok. So it's not a bug for systems which use mbr. Great you found the root cause 👍

framps, 2025-04-23:
Anybody who is interested in this PR is welcome to test the PR. Just download the updated raspiBackup.sh with `curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- m_374 raspiBackup.sh` and invoke the new code with `sudo ./raspiBackup.sh <options>`

framps, 2025-05-15:
@ylexus Just to let you know: I'm back home now and will review and test your PR the next couple of days 😉

framps, 2025-05-16:
@ylexus 

I created a normal backup and the restore and boot worked perfect :+1:

I then created a partition oriented backup and tried the restore. I got
```
sudo ~/raspiBackup.sh -d /dev/sdc .
--- RBK0009I: raspberrypi-bookworm64-lite-regression-gpt: raspiBackup.sh V0.7.0.2-m_374 - 2025-04-14 (6da3692) started at Fri May 16 19:53:42 CEST 2025
--- RBK0067I: Current partitions on /dev/sdc:
NAME    SIZE FSTYPE LABEL
sdc    59.5G        
|-sdc1  512M vfat   system-boot
`-sdc2   59G ext4   writable
!!! RBK0305W: Restoring partitions "1 2" to /dev/sdc
--- RBK0038I: Are you sure? y/N 
y
!!! RBK0052W: Creating partitions on /dev/sdc
--- RBK0038I: Are you sure? y/N 
y
--- RBK0323I: Target /dev/sdc with 59.47 GiB is larger than backup source with 7.49 GiB. Last partition will be expanded accordingly to use the whole available space
--- RBK0004I: Adjusting second partition from 6.99 GiB to 58.97 GiB
--- RBK0098I: Formatting partition /dev/sdc1 with fat32
??? RBK0153E: Unable to label partition: 'dosfslabel /dev/sdc1 Microsoft basic data' - RC: 1
??? RBK0001E: Unexpected program error occured. (6da3692), Linenumber: 2711, Error: Unkown exit error
??? RBK0148E: @@@@@@@@@@@@@@@@@@@@ Stacktrace @@@@@@@@@@@@@@@@@@@@
  File "/home/pi/raspiBackup.sh", line 10847, in main
    doit # no return
  File "/home/pi/raspiBackup.sh", line 7273, in doit
    		doitRestore
  File "/home/pi/raspiBackup.sh", line 9154, in doitRestore
    		restorePartitionBasedBackup
  File "/home/pi/raspiBackup.sh", line 8492, in restorePartitionBasedBackup
    				restorePartitionBasedPartition "$partitionBackupFile"
  File "/home/pi/raspiBackup.sh", line 8818, in restorePartitionBasedPartition
    			makeFilesystemAndLabel "$mappedRestorePartition" "$partitionFilesystem" "$partitionLabel"
  File "/home/pi/raspiBackup.sh", line 8766, in makeFilesystemAndLabel
    				exitError "$RC_LABEL_ERROR"
  File "/home/pi/raspiBackup.sh", line 2711, in exitError
    		assertionFailed $LINENO "Unkown exit error"
  File "/home/pi/raspiBackup.sh", line 2686, in assertionFailed
    	logStack
--- RBK0033I: Please wait until cleanup has finished
!!! RBK0256W: Restore finished with warnings. Check previous warning messages for details
??? RBK0077E: Restore failed. Check previous error messages
--- RBK0010I: raspberrypi-bookworm64-lite-regression-gpt: raspiBackup.sh V0.7.0.2-m_374 - 2025-04-14 (6da3692) stopped at Fri May 16 19:53:57 CEST 2025 with rc 101
--- RBK0026I: Debug logfile saved in /home/pi/raspiBackup.logr
```
Did you test the partition oriented backup? Do you have any idea what's the root cause of the restore failure?

ylexus, 2025-05-17:
@framps can you clarify what thee partition oriented backup is? I tested with backup mode = "Backup more than two partitions" with partitions 1 & 2 selected (out of 3). Is this the one? 
So the command `dosfslabel /dev/sdc1 "Microsoft basic data"` fails on the newly created fat32 partition `/dev/sdc1`. The actual error output of the command would be in the log file because it's executed as `$cmd "$partition" "$partitionLabel" &>>"$LOG_FILE"`, what do you see there?

It does not look related to gpt, but I can try to reproduce if you tell me how to re-creae what you are doing.

framps, 2025-05-17:
I'm sorry. I showed the restore command I used but not the backup creation command 😫 

I setup a standard lite Bookworm OS and converted it with mbr2gpt (a tool from RonR - see [here](https://forums.raspberrypi.com/viewtopic.php?t=196778) ) to use gpt instead of mbr. The system booted successfully and then I used the branch which includes your PR (m_374) with `sudo raspiBackup -P -t rsync` to create a backup which succeeded. Then I tried to restore the backup and got the failure listed above.

Sounds to me you used the important option `-P` to create a partition oriented backup. It may be mbr2gpt is the root cause for the issue. Can you give me instructions how I can easily convert a standard Bookworm lite system from mbr to gpt and I can try again?

ylexus, 2025-05-29:
I am sorry I have not forgotten but left this as a low priority thing to do for me. 
I remember I found a manual step by step procedure that copies existing RPi partitions to a USB drive and this is how I cloned my previous MBR 2TB NVMe to a new 4TB GPT NVMe. I can't find it anymore.
It looks like the usb-boot tool that you referenced can be used to copy the existing MBR installation onto a 2nd external USB drive, and you can select to create it as GPT - this may be more reliable than converting the partition type in-place.

Can I have an answer to this question in my previous comment?
>So the command dosfslabel /dev/sdc1 "Microsoft basic data" fails on the newly created fat32 partition /dev/sdc1. The actual error output of the command would be in the log file because it's executed as $cmd "$partition" "$partitionLabel" &>>"$LOG_FILE", what do you see there?

Which Raspberry Pi model did you use out of interest? I only have 5 and Zero, will be experimenting on a 5.

I will try to replicate your scenario, hope this week.

ylexus, 2025-05-31:
Not sure about the RPi type, just asked to get more info. After accidentally corrupting a working installation on one of my RPis (and, ironically, using raspiBackup to recover it!) I managed to reproduce the exact issue you are having. I will now look into why this is happening.

ylexus, 2025-05-31:
I found the issue, it's a slightly different format of the bulked output for GPT partitions. So the regex that tries to extract the partition label does not work as it gets the value from the PARTLABEL token and not the LABEL one. The fix is adding a single space - probably the shortest fix ever :-)

I have also found that the partition sizes are not extracted correctly for display in the console during backup. `df` displays all the values in KB, and the script assumes this is in bytes. I have added `-B 1` argument to `df` to fix that and also fixed the following awk script not to use the scientific notation for large numbers.

After the fix is made, the restore was successful. I restored the backup of an SD card to an external USB stick and then tried to boot from it, however got this:

[image]

This seems to be happening because the USB stick has different device names (`/dev/sdaX`), and the restored data somehow references `/dev/mmcblk01p1` - an SD card device. I suspect it has nothing to do with GPT, but I am not sure. Is this expected?

I have raised the PR: https://github.com/framps/raspiBackup/pull/891

ylexus, 2025-06-01:
The issue is fixed when on the restored card's `/etc/fstab` I replace
```
/dev/mmcblk0p1  /boot/firmware  vfat    defaults          0       2
```
with
```
PARTUUID=df06e23f-4069-4fb4-a5f2-65bd5c9ac229  /boot/firmware  vfat  defaults  0  2
```

Note that the original system uses the first variant (device name), unclear why as it's unreliable.

If I insert the SD card directly into RPi without any modifications, I get this, which I think simply indicates that the card is incompatible with the Raspberry Pi. I need another SD card but I don't have any. 

[image]

I think the fix works, but I am unable to do the clean verification as I don't have enough SD cards. As for booting via the USB card reader it looks it's a completely separate problem.

ylexus, 2025-06-01:
I have now figured out why, despite the genuine Raspberry Pi images NOT using device names in `/etc/fstab`,  my source SD card did. This is because `mbr2gpt` replaced it - this is the code it has:
```
sed -i "/^[[:space:]]*#/!s|^\S\+\(\s\+/boot\S*\s\+vfat\s\+.*\)$|/dev/mmcblk0p1\1|" "${MNTPATH}/etc/fstab"
```

i.e. it hardcodes `/dev/mmcblk0p1` as the device name for the mount. This is not good, and we can say that it's a problem of the `mbr2gpt` tool - have no idea why it uses device name instead of the partition UUID.

I have now restored the backup to a different SD card and it all worked - the result was a fully bootable SD card with the GPT partition. I pronounce the problem fully fixed. 

I did notice swap file getting corrupted after the restore, which is not a fatal error. It's easily fixable via the commands below. Do you know of any issues with the corrupt swap file after the restore? It looks like it's common, I suspect the shrinking or expanding the file system during restore can easily fragment the file and corrupt it.

```
sudo dphys-swapfile swapoff
sudo dphys-swapfile uninstall
sudo dphys-swapfile setup
sudo dphys-swapfile swapon
sudo systemctl restart dphys-swapfile
```

reraikes, 2025-06-01:
mbr2gpt sets the BOOT partition entry of the of the destination device's fstab to "/dev/mmcblk0p1" ONLY if you answer "Yes" to "Set SD card to boot <destination-device> (y/n)? ". This was used prior to the introduction of the Raspberry Pi 4/5 when USB booting was unsupported/unreliable. When this option is used, the SD card is rendered unbootable as a standalone device because its cmdline.txt "root=" is set to the PARTUUID of the destination device.

Only answer "Yes" to "Set SD card to boot <destination-device> (y/n)? " if you specifically want Hybrid booting (where the SD card boots the USB/NVMe device). If this is your use case, you can toggle SD card booting back and forth with sdc-boot.

framps, 2025-06-03:
@reraikes I just used gpt2mbr on a plain RaspianOS lite and got following output:
```
pi@raspberrypi-bookworm64-lite-regression:~ $ sudo ./mbr2gpt /dev/sdc
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)

Partition table type is MBR

Ok to convert /dev/sdc to GPT (y/n)? y

ROOT partition overlaps the Secondary partition table area

Ok to resize ROOT partition (y/n)? y

Ok to process /dev/sdc (y/n)? y

resize2fs 1.47.0 (5-Feb-2023)
Resizing the filesystem on /dev/sdc2 to 821943 (4k) blocks.
The filesystem on /dev/sdc2 is now 821943 (4k) blocks long.

resize2fs 1.47.0 (5-Feb-2023)
Resizing the filesystem on /dev/sdc2 to 1832704 (4k) blocks.
The filesystem on /dev/sdc2 is now 1832704 (4k) blocks long.
```
So I didn't get an query `Set SD card to boot destination-device (y/n)?`. 

But I also still have PARTUUIDs in /etc/fstab.

```
pi@raspberrypi-bookworm64-lite-regression:~ $ sudo fdisk -l /dev/sdc
Disk /dev/sdc: 7.5 GiB, 8053063680 bytes, 15728640 sectors
Disk model: MassStorageClass
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 730D952B-387D-4F1B-97E6-815A3FF91E05

Device       Start      End  Sectors  Size Type
/dev/sdc1    16384  1064959  1048576  512M Microsoft basic data
/dev/sdc2  1064960 15726591 14661632    7G Linux filesystem
pi@raspberrypi-bookworm64-lite-regression:~ $ cat /mnt/etc/fstab 
proc            /proc           proc    defaults          0       0
PARTUUID=0c4112d2-d9ee-4cc5-8b35-b8d6a1d96acd  /boot/firmware  vfat    defaults          0       2
PARTUUID=e902145c-4a50-4524-bc1f-66990c01b2d9  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
```

Anyhow - I will continue now with tests.

reraikes, 2025-06-03:
| Nobody wrote there is an issue with mbr2gpt

Based on the input from @ylexus (https://github.com/framps/raspiBackup/issues/374#issuecomment-2927008167), I gathered there was a suspicion that mbrgpt was making an incorrect change. 

Unless I hear differently, I'll assume mbrgpt is working as properly.

framps, 2025-06-03:
> I gathered there was a suspicion that mbrgpt was making an incorrect change.

Nobody expected the PARTUUID to be substituted by the device name by mbr2gpt. We learned now this is caused by the fact an SD card is available in /dev/mmcblk0 and then `SDBOOT=TRUE` is set in mbr2gpt which causes this change to support non USB boot capable Raspberries.

> Unless I hear differently, I'll assume mbrgpt is working as properly.

As you can read in my previous comment mbr2gpt words like a charm and supports us in the PR and there are no issues 😉

framps, 2025-06-04:
> Do you know of any issues with the corrupt swap file after the restore?

No. I didn't notice there is a problem to restore the swapfile - unfortunately. As you wrote it's not a severe issue. But nevertheless I created an issue to track this (https://github.com/framps/raspiBackup/issues/892)

framps, 2025-06-04:
I completed additional tests with option -P successfully 😃 

Backup and restore of multiple partitions (>2) and also use new option `-00` to speed up a rsync restore.

framps, 2025-06-05:
> But nevertheless I created an issue to track this

I fixed the swap issue recently. ...

@ylexus Thank you very much you mentioned this issue.

framps, 2025-06-05:
My tests are successful 👍 

Finally I will execute the regression suite to make sure everything still works as expected with mbr. 

Would be great to execute the regression suite also with gpt but given the regression suite is in transition to a new environment that's impossible right now. Given the updates are very local and specific to mbr/gpt handling and code review and my sample tests are successfully I'm confident this PR will add gpt support to raspiBackup.

@ylexus I'll execute the regression test for mbr the next couple of days and don't expect any issues. When the regression test succeeds your PR will be published with next upcoming  raspiBackup release. Thank you very much for your PR and help to add an additional feature to raspiBackup 👍

ylexus, 2025-06-06:
Thanks @framps - happy to contribute to the great project that I personally constantly use on multiple devices!

closed 2025-06-06 by framps

ylexus, 2025-06-06:
Thanks @framps for testing this! Can't wait to install the official version!

framps, 2025-06-08:
I usually collect multiple new feature until I publish a new release and this usually takes approx 6 -12 months.

But in the meantime you can grab raspiBackup.sh from the m_374 branch, comment out the warning not to use this version in production and copy raspiBackup.sh into /usr/local/bin.

You'll get notified if the next release was published and then update your m_375 version with the next official version 😉

framps, 2025-06-08:
> why not release more often,

I don't want to bother raspiBackup users with release updates because of new features whenever there is a new feature available because I suggest everybody to execute a regression test whenever a release update was downloaded which is time consuming. And for most of the features it's a new feature they are not interested at all. 
The gpt feature will definitely  be an upcoming feature requested by raspiBackup users but as of now it's just you who asked for it and fortunately helped to add the required code in a PR to support devices > 2GB. Unfortunately gpt is not supported as now by RaspbianOS and it's just you who asked for gpt support 😉

reraikes, 2025-06-08:
> Unfortunately gpt is not supported as now by RaspbianOS and it's just you who asked for gpt support

While Raspberry Pi OS images are not distributed with GPT partitioning, GPT support has been present as far back as Jessie (10 years ago).  Raspberry Pi OS images (32- and 64-bit) converted to GPT partitioning (by using mbr2gpt, for example) boot and run perfectly on all Raspberry Pi's, even a Raspberry Pi 1, even if the conversion is performed before the first boot.

framps, 2025-06-10:
I agree with you. gpt is supported for a long time already. But given the RaspbianOS is published with mbr most of the Raspberry users just use mbr. Only folks who use devices > 2TB detect mbr does not support these huge disks and will convert their system to gpt. 

Are there any other reasons I miss which are a pro for gpt other than the > 2TB device size?