framps/raspiBackup #386: Error in C5  Backup mode

sgrayban, 2021-11-07:
When setting up raspiBackup for the first time I ran into a issue in the Backup mode selection. When I try to go to it I get the following but this doesn't seem to happen with other setups I have .

`Programm error, unrecognized backup mode n`

Not sure how this happened

sgrayban, 2021-11-07:
never mind found the problem

framps, 2021-11-07:
Glad you found the root cause. Would you please give me some details? Maybe I can prevent this to happen with some code or doc update :wink:

framps, 2021-11-07:
Glad you found the root cause. Would you please give me some more details? Maybe I can update the code or update the doc in order to prevent this to happen in the future for other raspiBackup users  :wink:

sgrayban, 2021-11-07:
Somehow the config file saved a 'n' in the parameter DEFAULT_PARTITIONBASED_BACKUP

`DEFAULT_PARTITIONBASED_BACKUP="n"`

framps, 2021-11-07:
> Somehow the config file saved a 'n'

Then it's an issue. Would be glad you have the installer log handy but I assume you already started the installer again for other config updates and created a new log :cry: 

I will check the code carefully and hope I find the root cause.

sgrayban, 2021-11-07:
Sorry that log got over writien

framps, 2021-11-07:
> Sorry that log got over writien

That's what I expected :cry: . I checked the code carefully but I don't see any way how "n" could have been written by the installer. Either "0" or "1" is written. Are you sure you didn't modify the config file by hand?

If you remember how you got this issue please provide the debug log. Otherwise please close the issue and I will mark this issue as unreproducible.

sgrayban, 2021-11-07:
Never edited that file until today. I'll close the ticket

closed 2021-11-07 by sgrayban