framps/raspiBackup #387: setting up multi partition backup doesn't work

sgrayban, 2021-11-07:
In the GUI you can setup the backup mode.. either the standard 2 partitions or multi. When I choose multi and the next screen gives the partitions I select them and tab to ok to save but it doesn't and goes back to the same settings again like there is a issue when selecting them.

[image]
[image]

I manually made the change in the /usr/local/etc/raspiBackup.conf
[image]

Would this be ok or is there issue doing this?

VERSION_CONFIG="0.1.5"

sgrayban, 2021-11-07:
Actually it didn't even see all the partitions either

```
# fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 29.7 GiB, 31914983424 bytes, 62333952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xd78dc7ec

Device         Boot  Start      End  Sectors   Size Id Type
/dev/mmcblk0p1        2048   492187   490140 239.3M  e W95 FAT16 (LBA)
/dev/mmcblk0p2      492188 62333951 61841764  29.5G  5 Extended
/dev/mmcblk0p5      499712   565245    65534    32M 83 Linux
/dev/mmcblk0p6      565248   706559   141312    69M  e W95 FAT16 (LBA)
/dev/mmcblk0p7      712704 62333951 61621248  29.4G 83 Linux
```

It only saw 1 5 6 7 and skipped 2

So currently I have

```
# Options for partitionoriented mode
DEFAULT_PARTITIONBASED_BACKUP="1"
DEFAULT_PARTITIONS_TO_BACKUP="1 2 5 6 7"
```

framps, 2021-11-07:
> It only saw 1 5 6 7 and skipped 2

That's by intention. `/dev/mmcblk0p2` is an extended partition and it doesn't make sense to save this type of partition.

sgrayban, 2021-11-07:
Ok but it *should* work for the other partitions right? This raspberry image was created by berryboot so it adds its own recovery partitions.

framps, 2021-11-07:
> Ok but it _should_ work

Yes. raspiBackup is officially supported for RaspberryOS only and never tested with berryboot. But give it a try.

sgrayban, 2021-11-07:
Well let's see what happens with my backup tonight and I'll let you know if it failed or worked

framps, 2021-11-07:
Which installer version do you run?

`raspiBackupInstallUI.sh -h`

sgrayban, 2021-11-07:
`raspiBackupInstallUI.sh 0.4.3.6, 2021-11-04/15:20:10 - 7ec5f22`

sgrayban, 2021-11-07:
BTW no -h switch in mine

```
# raspiBackupInstallUI.sh -h
raspiBackupInstallUI.sh 0.4.3.6, 2021-11-04/15:20:10 - 7ec5f22
raspiBackupInstallUI.sh ( -i [-e]? | -u | -U ) [-d]? 
-d: enable debug mode
-e: unattended (re)install of raspiBackup extensions
-i: unattended (re)install of raspiBackup
-U: unattended update of raspiBackupInstallUI.sh
-u: unattended uninstall of raspiBackup

```

sgrayban, 2021-11-08:
So I started the backup using rsync for the multi partitions and its still running 12 hours later -- is this right?

framps, 2021-11-08:
The first rsync backup is a fulll backup and depending on the disk size and backup partition speed this takes time ...

sgrayban, 2021-11-08:
ok -- wish there was a bit more info in the backup logs such as what file is being backed up so we knida know that it is still working and not hung up because I have been tempted to stop the backup a few times.

```
--- RBK0092I: Saving partition mmcblk0p7 (31.6GB) ...
--- RBK0158I: Creating native rsync backup "/backup/weewx-pi-backups/weewx-pi.borgnet.us/weewx-pi.borgnet.us-rsync-backup-20211107-064952".
--- RBK0085I: Backup of type rsync started. Please be patient.
```

That's not a lot of info... at least a percentage would be nice.

sgrayban, 2021-11-08:
I started this backup early this morning

--- RBK0009I: weewx-pi.borgnet.us: raspiBackup.sh V0.6.6.1 (2ed7282) started at Sun 07 Nov 2021 06:49:54 AM PST.

And its now 11/8/21 12:12:40 AM PST

I did another pi and that only took about a hour to complete

framps, 2021-11-08:
> That's not a lot of info... at least a percentage would be nice.

Please check [this list of options]( https://www.linux-tips-and-tricks.de/en/backup/#alphabetical). Check option `-g`

framps, 2021-11-08:
> I did another pi and that only took about a hour to complete

Please check with top whether raspiBackup still consumes CPU. I'm sure it will 😃 .

sgrayban, 2021-11-08:
[image]
[image]

sgrayban, 2021-11-09:
So I tested out the `-g` option and was expecting to see this doing the backup using `raspiBackup.sh -g` but instead you only see this while tailing `raspiBackup.log` and even that is ambiguous....

`35,374,877  99%  350.50kB/s    0:01:38 (xfr#152, to-chk=132/286)`

I was hoping to actually see each file being backed up and the percentage of each.

sgrayban, 2021-11-09:
So I edited /usr/local/bin/raspiBackup.sh and changed the line

`cmd="rsync --info=progress2 $cmdParms"`

to

`cmd="rsync --progress $cmdParms"`

Which was awesome and exactly what I wanted to see in the backup logging. Maybe you should make this a rsync option??

framps, 2021-11-09:
> Maybe you should make this a rsync option??

Nobody asked for this feature until now :wink: . I created issue #391 on behalf of you.

There is a workaround for you: Just don't use option `-g` and define`DEFAULT_RSYNC_BACKUP_ADDITIONAL_OPTIONS="--progress"` in your config file.

Back to the topic of this issue. As far as I see you can close this issue because it doesn't make sense to backup extended partitions. Please open another issue if you have further questions/issues.

sgrayban, 2021-11-11:
Closed by accident as OP isn't addressed yet -- sorry

framps, 2021-11-18:
> So I started the backup using rsync for the multi partitions and its still running 12 hours later -- is this right?

Did you find the root cause for the issue and share your experience? I got just a comment from another raspiBackup user he also faces a long backup time and the backup still runs.

> Closed by accident as OP isn't addressed yet -- sorry

Not sure I get your point :tired_face:

sgrayban, 2021-11-18:
No I gave up on a rsync option and just used dd to create a daily image file of the whole sd card.

OP = original post

framps, 2021-11-18:
> No I gave up on a rsync option and just used dd to create a daily image file of the whole sd card.

The user I mentioned wrote he's going to create an issue.  Maybe you want to follow the issue to get an idea what's the root cause. I never had this kind of issue. 

> OP = original post

It's the issue whether a multi partition system can be saved with raspiBackup?  If you now use dd this will work.

sgrayban, 2021-11-18:
I just wrote a bash script that makes 1 img file and saved with a date format. Your script doesn't do this.

framps, 2021-11-18:
If you can write bash scripts you're not the target group of raspiBackup :smile: 

Anyhow raspiBackup can create backups with dd. Does your script also does versioning of the backups or offer to use a grandfather-father-son backup strategy :smiley:

closed 2021-12-14 by github-actions[bot]