framps/raspiBackup #393: raspiBackup terminates with RC23 if rsync type is used and an nfs backup space on Bullseye

framps, 2021-11-11:
Root cause is that /var/log/journal now has ACLs

bastelbert, 2021-11-12:
Errors on Raspberry Pi 4 with latest Bullseye and raspiBackup.sh 0.6.6.1:

`20211112-013013 DBG 1981:                      --- Redirect: & - Skips: 
rsync: [generator] set_acl: sys_acl_set_file(var/log/journal, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [generator] set_acl: sys_acl_set_file(var/log/journal/3e23e34d17484a28b9473f69f7d9d21c, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(var/log/journal/3e23e34d17484a28b9473f69f7d9d21c/.system.journal.8pTVZn, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(var/log/journal/3e23e34d17484a28b9473f69f7d9d21c/.system@0005cf904280e339-4b09e06bb02c79ac.journal~.j2Q8Jp, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(var/log/journal/3e23e34d17484a28b9473f69f7d9d21c/.system@0005d08251e08f15-891128351680d704.journal~.Wxd60n, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(var/log/journal/3e23e34d17484a28b9473f69f7d9d21c/.system@0005d083451f4562-ea8192744cbc848d.journal~.fU4w4o, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(var/log/journal/3e23e34d17484a28b9473f69f7d9d21c/.user-1000.journal.dIbQpp, ACL_TYPE_ACCESS): Operation not supported (95)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1333) [sender=3.2.3]`

framps, 2021-11-12:
@bastelbert Looks like you are an early bird :laughing: .

I'm still in the evaluation phase...

But there is a workaround:

If you are not interested in /var/log/journal (Only people which do Linux debugging will need the journal) just exclude `/var/log/journal` and `/run/log/journal`. See `man journald.conf` for details.

framps, 2021-11-13:
I just learned all previous RaspiOS images which can be downloaded from raspberrypi.com don't include /var/log/journal and therefore there was no issue with ACLs. But now the Bullseye image includes /var/log/journal (not sure whether by intention or by mistake) which causes the issue now.

framps, 2021-11-13:
Read [this article](https://lwn.net/Articles/812238/) which explains why Debian decided to enable persistent journal in Bullseye.

bastelbert, 2021-11-15:
> DEFAULT_EXCLUDE_LIST="--exclude=/var/log/journal"

Thanks for your great support, works now!

framps, 2021-11-19:
Fixed the issue in https://github.com/framps/raspiBackup/commit/5635776cad78088949148a4930e3af78134bff66.

@bastelbert Would you like to verify the fix? Please download https://raw.githubusercontent.com/framps/raspiBackup/bullseye/raspiBackup.sh into /home/pi .

Then change

`DEFAULT_EXCLUDE_LIST="--exclude=/var/log/journal"`

to 
`DEFAULT_EXCLUDE_LIST=""
``
and finally execute

`sudo bash ./raspiBackup.sh`

(note the leading dot :wink: )

bastelbert, 2021-11-19:
Cool, thank you! I can confirm that the fix works on my RasPi4.

fsc830, 2021-11-20:
Me too!
Fix works fine.
Regards

framps, 2021-11-20:
Thank you very much you verified the fix. I will publish the fix today.

framps, 2021-11-20:
Fixed in https://github.com/framps/raspiBackup/commit/5635776cad78088949148a4930e3af78134bff66

closed 2021-11-20 by framps

framps, 2021-12-02:
## Note how to upgrade raspiBackup to include the Bullseye fix

In order to update a local release 0.6.6.1 which includes the Bullseye fix execute `sudo raspiBackup.sh -U -S`. 
Everybody else who uses a release less than 0.6.6.1 should use `sudo raspiBackup.sh -U`to upgrade to release 0.6.6.1.

RonV42, 2022-09-12:
Hello I found a new journal directory that has this error when using Log2Ram on Raspian Bullseye.  It's creates the journal in this directory:

/var/hdd.log/journal

That generates this error:

`rsync: [receiver] set_acl: sys_acl_set_file(var/hdd.log/journal/e11f59cd445d4151ad173121a3ba17ab/.system@9a1286019d8c4496939bf1145d51b44e-0000000000004a7e-0005e7bb3401ce99.journal.mG3eOx, ACL_TYPE_ACCESS): Operation not supported (95)`

I have put an exclude statement to bypass the directory and it works.

DEFAULT_EXCLUDE_LIST="--exclude /var/hdd.log/journal"

framps, 2022-09-12:
Thank you very much you describe how to fix this issue which is the right way to do it :+1: . I hope others will search my issues as you did and find your fix. Log2Ram is not standard in Bullseye and thus I will not add this directory in future releases. Any issue on a non standard Raspbian OS should be fixed by a user of raspiBackup as you successfully did.

weitzner, 2022-11-04:
Hello, I am a new user and am having the same issue using raspiBackup 6.7 to backup Bullseye to a Synology using nfs4.1. I am under the impression that this should just work for `/var/log/journal`, but with or without adding `DEFAULT_EXCLUDE_LIST="--exclude /var/log/journal"` to ` /usr/local/etc/raspiBackup.conf` or adding `-u "\--exclude=/var/log/journal"` to the command line, the backup fails with:

`rsync: [generator] set_acl: sys_acl_set_file(mmcblk0p7/var/log/journal, ACL_TYPE_ACCESS): Operation not supported (95)`

The log incudes `--- RBK0151I: Using backuppath /backup with partition type nfs4.`, which makes me think the issue is not that the tool fails to detect the filesystem. Any ideas to try or information that would help get to the boom of this?

framps, 2022-11-04:
Hm ... I added code which excludes /var/log/journal (see https://github.com/framps/raspiBackup/commit/5635776cad78088949148a4930e3af78134bff66) :thinking: 

Would you please attach the debug log for further investigations?

weitzner, 2022-11-04:
Sure thing - I have this set up not to stop/start any services and to only backup the main partition (eventually I'd like to have a complete image from which I can restore if need be, but I was aiming to keep it simple while I am familiarizing myself with this tool). Possibly relevant - I am mounting the shared folder from the synology with the following entry in my fstab `<synology ip address>:/volume1/raspberrypi /backup nfs auto 0 0` 

Log file contents below

```
20221104-125346 DBG 1993:  --- Logfiles used: /tmp/raspiBackup.log and /tmp/raspiBackup.msg
20221104-125346 DBG 2004:  --- raspiBackup.sh 0.6.7, 2022-10-11/08:45:20 - 0d32ab9
20221104-125346 DBG 2007:  --- +================================================================================================================================================+
20221104-125346 DBG 2008:  --- | ===> A lot of sensitive information is masqueraded in this log file. Nevertheless please check the log carefully before you distribute it <=== |
20221104-125346 DBG 2009:  --- +================================================================================================================================================+
20221104-125346 DBG 1836:  --> trapWithArg cleanupStartup SIGINT SIGTERM EXIT
20221104-125346 DBG 1841:  <-- trapWithArg 
20221104-125346 DBG 2616:  --> initializeDefaultConfigVariables 
20221104-125346 DBG 2754:  <-- initializeDefaultConfigVariables 
20221104-125346 DBG 2759:  --> copyDefaultConfigVariables 
20221104-125346 DBG 2831:  <-- copyDefaultConfigVariables 
20221104-125346 DBG 3581:  --> readConfigParameters 
20221104-125346 DBG 3596:      --- Read config /usr/local/etc/raspiBackup.conf : 0.1.6
20221104-125346 DBG 3596:      --- VERSION_CONFIG="0.1.6"
20221104-125346 DBG 3596:      --- DEFAULT_BACKUPPATH="/backup"
20221104-125346 DBG 3596:      --- DEFAULT_BACKUPTYPE="rsync"
20221104-125346 DBG 3596:      --- DEFAULT_ZIP_BACKUP="0"
20221104-125346 DBG 3596:      --- DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20221104-125346 DBG 3596:      --- DEFAULT_KEEPBACKUPS="5"
20221104-125346 DBG 3596:      --- DEFAULT_KEEPBACKUPS_DD=0
20221104-125346 DBG 3596:      --- DEFAULT_KEEPBACKUPS_DDZ=0
20221104-125346 DBG 3596:      --- DEFAULT_KEEPBACKUPS_TAR=0
20221104-125346 DBG 3596:      --- DEFAULT_KEEPBACKUPS_TGZ=0
20221104-125346 DBG 3596:      --- DEFAULT_KEEPBACKUPS_RSYNC=0
20221104-125346 DBG 3596:      --- DEFAULT_SMART_RECYCLE="0"
20221104-125346 DBG 3596:      --- DEFAULT_SMART_RECYCLE_DRYRUN="1"
20221104-125346 DBG 3596:      --- DEFAULT_SMART_RECYCLE_OPTIONS="7 4 12 1"
20221104-125346 DBG 3596:      --- DEFAULT_STOPSERVICES=":"
20221104-125346 DBG 3596:      --- DEFAULT_STARTSERVICES=":"
20221104-125346 DBG 3596:      --- DEFAULT_BEFORE_STOPSERVICES=""
20221104-125346 DBG 3596:      --- DEFAULT_AFTER_STARTSERVICES=""
20221104-125346 DBG 3596:      --- DEFAULT_EMAIL=""
20221104-125346 DBG 3596:      --- DEFAULT_SENDER_EMAIL=""
20221104-125346 DBG 3596:      --- DEFAULT_EMAIL_PARMS=""
20221104-125346 DBG 3596:      --- DEFAULT_MAIL_PROGRAM="mail"
20221104-125346 DBG 3596:      --- DEFAULT_TELEGRAM_TOKEN=""
20221104-125346 DBG 3596:      --- DEFAULT_TELEGRAM_CHATID=""
20221104-125346 DBG 3596:      --- DEFAULT_TELEGRAM_NOTIFICATIONS="F"
20221104-125346 DBG 3596:      --- DEFAULT_NOTIFY_START=0
20221104-125346 DBG 3596:      --- DEFAULT_MAIL_ON_ERROR_ONLY=0
20221104-125346 DBG 3596:      --- DEFAULT_MSG_LEVEL="0"
20221104-125346 DBG 3596:      --- DEFAULT_COLORING="CM"
20221104-125346 DBG 3596:      --- DEFAULT_COLOR_CODES=("#FF8000 33" "#FF0000 31")
20221104-125346 DBG 3596:      --- DEFAULT_EMAIL_COLORING="SUBJECT"
20221104-125346 DBG 3596:      --- DEFAULT_RESTORE_DEVICE=""
20221104-125346 DBG 3596:      --- DEFAULT_VERBOSE=0
20221104-125346 DBG 3596:      --- DEFAULT_SKIPLOCALCHECK=0
20221104-125346 DBG 3596:      --- DEFAULT_DD_BLOCKSIZE=1M
20221104-125346 DBG 3596:      --- DEFAULT_DD_PARMS=""
20221104-125346 DBG 3596:      --- DEFAULT_EXCLUDE_LIST=""
20221104-125346 DBG 3596:      --- DEFAULT_NOTIFY_UPDATE=1
20221104-125346 DBG 3596:      --- DEFAULT_EXTENSIONS="mem temp disk"
20221104-125346 DBG 3596:      --- DEFAULT_RESTORE_EXTENSIONS=""
20221104-125346 DBG 3596:      --- DEFAULT_LANGUAGE="EN"
20221104-125346 DBG 3596:      --- DEFAULT_DEPLOYMENT_HOSTS=""
20221104-125346 DBG 3596:      --- DEFAULT_YES_NO_RESTORE_DEVICE=""
20221104-125346 DBG 3596:      --- DEFAULT_RESIZE_ROOTFS=1
20221104-125346 DBG 3596:      --- DEFAULT_LINK_BOOTPARTITIONFILES=0
20221104-125346 DBG 3596:      --- DEFAULT_TAR_BOOT_PARTITION_ENABLED=0
20221104-125346 DBG 3596:      --- DEFAULT_SKIP_DEPRECATED=""
20221104-125346 DBG 3596:      --- DEFAULT_USE_UUID=1
20221104-125346 DBG 3596:      --- DEFAULT_CHECK_FOR_BAD_BLOCKS=0
20221104-125346 DBG 3596:      --- DEFAULT_TIMESTAMPS=0
20221104-125346 DBG 3596:      --- DEFAULT_SYSTEMSTATUS=0
20221104-125346 DBG 3596:      --- DEFAULT_REBOOT_SYSTEM=0
[... 1294 lines cut ...]
20221104-132701 DBG 3849:                      <-- colorAnnotation 
--- RBK0010I: @HOSTNAME@: raspiBackup.sh V0.6.7 (0d32ab9) stopped at Fri 04 Nov 2022 01:27:01 PM MDT with rc 117.
20221104-132701 DBG 3970:                      --> sendEMail 
20221104-132701 DBG 4092:                      <-- sendEMail 
20221104-132701 DBG 4191:                      --- Masquerading some mount options
20221104-132701 DBG 4209:                      --- Masquerading home directory name
[... cut ...]
20221104-132704 DBG 2080:                              *** cat /tmp/raspiBackup.logf
20221104-132704 DBG 2080:                                  
20221104-132704 DBG 3825:                          --> colorAnnotation 1
20221104-132704 DBG 3849:                          <-- colorAnnotation 
20221104-132704 DBG 2095:                      <-- logFinish 
20221104-132704 DBG 4316:                      --> unLockMe 
20221104-132704 DBG 4318:                      <-- unLockMe 
20221104-132704 DBG 4426:                  <-- cleanup 
```

framps, 2022-11-04:
Thank you very much for the log. But please attach the log as a file with dragndrop and I can inspect it with my local editor :wink:

weitzner, 2022-11-04:
Roger that - updated the comment above

framps, 2022-11-04:
I detected in the log you use a NOOBs image and thus the partition oriented backup mode. 

I unfortunately missed to exclude the journal also in this mode and created a new version for you. Please download [this version](https://raw.githubusercontent.com/framps/raspiBackup/master_journal/raspiBackup.sh) and test. If it fails again please attach the debug log as a file :wink: 

Note: I strongly recommend not to use a NOOBs image any more. But nevertheless raspiBackup supports NOOBs with the partition oriented mode :smile:

weitzner, 2022-11-04:
Unfortunately that produces the same error

framps, 2022-11-04:
> If it fails again please attach the debug log as a file 

:wink:

weitzner, 2022-11-04:
Apologies! Attached. [raspiBackup.log](https://github.com/framps/raspiBackup/files/9942157/raspiBackup.log)

framps, 2022-11-05:
rsync doesn't like two slashes in an exclude statement :roll_eyes: . That's fixed now. Please try again :wink:

weitzner, 2022-11-05:
`--- RBK0017I: Backup finished successfully.` 🎉

framps, 2022-11-05:
I just published the fix. Use `sudo raspiBackup.sh -U -S` to grab the fix and replace the temp fix :wink: