framps/raspiBackup #394: Backup notification says I need to upgrade but already upgraded

SJMarty, 2021-11-12:
Hi Framp,

I have five Raspberry Pis running raspiBackup.  I've upgraded all of them to v0.6.6.1.  When the scheduled weekly backup runs, I'm getting this in each of the e-mail notifications...

```
--- RBK0128I: Using logfile /media/pi/nfs/pi-name-here/pi-name-here-rsync-backup-20211112-080001/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
--- RBK0080I: ;-) There is a new version 0.6.6.1 of raspiBackup available for download. You are running version 0.6.6 and now can use option -U to upgrade your local version.
--- RBK0114I: Visit https://www.linux-tips-and-tricks.de/en/versionhistory/ to read about the changes in the new version.
```

The "sudo raspiBackup -U" command produces the following output...

```
--- RBK0031I: Checking whether a new version of raspiBackup is available.
--- RBK0073I: /usr/local/bin/raspiBackup already current with version 0.6.6.1.
```

Not sure what I'm doing wrong.

framps, 2021-11-12:
Please provide the debug log :wink:

SJMarty, 2021-11-12:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/7528844/raspiBackup.log)

framps, 2021-11-12:
Please execute `sudo raspiBackup.sh -U` and show the debug log which was created in your home directory.

SJMarty, 2021-11-14:
I was executing `sudo raspiBackup -U` instead of `sudo raspiBackup.sh -U`.  After executing the latter, it upgraded successfully.

framps, 2021-11-14:
This shouldn't make a difference. Would you please show the result of
```
ls -la /usr/local/bin/raspiBackup*
```
?

SJMarty, 2021-11-14:
Sure...

```
-rwxr-xr-x 1 root root 307748 Nov 10 06:04 /usr/local/bin/raspiBackup
-rwxr-xr-x 1 root root 273696 Apr 10  2021 /usr/local/bin/raspiBackup.0.6.6.sh
lrwxrwxrwx 1 root root     38 Mar 31  2021 /usr/local/bin/raspiBackupInstallUI -> /usr/local/bin/raspiBackupInstallUI.sh
-rwxr-xr-x 1 root root 115675 Mar 31  2021 /usr/local/bin/raspiBackupInstallUI.sh
-rwxr-xr-x 1 root root 307748 Nov 12 12:57 /usr/local/bin/raspiBackup.sh
```

framps, 2021-11-14:
That's the format the files should be linked. 
```
lrwxrwxrwx 1 root root     29 Nov 14 18:07 raspiBackup -> /usr/local/bin/raspiBackup.sh
lrwxrwxrwx 1 root root     38 Nov 14 18:07 raspiBackupInstallUI -> /usr/local/bin/raspiBackupInstallUI.sh
-rwxr-xr-x 1 root root 162832 Nov 14 18:06 raspiBackupInstallUI.sh
-rwxr-xr-x 1 root root 338399 Nov 14 18:07 raspiBackup.sh
```
For some unknown reasons your raspiBackup is not linked to raspiBackup.sh and I guess they are not identical. Please delete /usr/local/bin/raspiBackup and link /usr/local/raspiBackup.sh to /usr/local/bin/raspiBackup to correct this issue.

SJMarty, 2021-11-14:
I'm not sure how to do this.  Would you mind providing the exact commands I need to execute so I don't mess anything up?  The listing of that folder's contents looks the same on all five of my Pis.

framps, 2021-11-14:
Sure.
```
sudo rm /usr/local/bin/raspiBackup
sudo ln -s /usr/local/bin/raspiBackup.sh /usr/local/bin/raspiBackup
ls -la /usr/local/bin/raspi*
```
Last command should show you a similar result as it's on my system.

> The listing of that folder's contents looks the same on all five of my Pis.
Looks like you started with raspiBackup release 0.6.6. Did you? Maybe there is some issue in the installer with this file linkage. I will try to reproduce your issue.

SJMarty, 2021-11-14:
Here is my result after...
```

lrwxrwxrwx 1 root root     29 Nov 14 12:38 /usr/local/bin/raspiBackup -> /usr/local/bin/raspiBackup.sh
-rwxr-xr-x 1 root root 273696 Apr 10  2021 /usr/local/bin/raspiBackup.0.6.6.sh
lrwxrwxrwx 1 root root     38 Jan 30  2021 /usr/local/bin/raspiBackupInstallUI -> /usr/local/bin/raspiBackupInstallUI.sh
-rwxr-xr-x 1 root root 101809 Jan 30  2021 /usr/local/bin/raspiBackupInstallUI.sh
-rwxr-xr-x 1 root root 307748 Nov 13 07:29 /usr/local/bin/raspiBackup.sh
```

I have an extra file that you don't show in your listing...

`-rwxr-xr-x 1 root root 273696 Apr 10  2021 /usr/local/bin/raspiBackup.0.6.6.sh`

Yes, I'm pretty sure I started with v0.6.6.

framps, 2021-11-14:
Great. Looks good now :+1: 

> I have an extra file that you don't show in your listing...

I know. This file was generated as a backup when you upgraded from 0.6.6 to 0.6.6.1. That way you can revert the upgrade if you want. Just try option `-V`. That's why I concluded you started with 0.6.6 :wink:

framps, 2021-11-15:
I tried to reproduce your issue but I failed. That's what I did:

1) Installed raspiBackup 0.6.6 with the installer. Checked the symlinks which are OK
2) Upgraded raspiBackup and the installer with option -U in the commandline. Checked the symlinks which are OK
3) Deleted raspiBackup and installer
4) Installed raspiBackup 0.6.6 with the installer. Checked the symlinks which are OK
5) Upgraded raspiBackup and the installer with the installer. Checked the symlinks which are OK

Do you have any idea how you managed to delete the symlink `/usr/local/bin/raspiBackup` to `/usr/local/raspiBackup.sh`?

SJMarty, 2021-11-15:
I documented all of the steps I took to initially install raspiBackup and I don't see anything that would have deleted the symlink.  Whatever happened, it happened to all five of my Pis.  I followed your steps and cleaned them all up.

Thanks for your quick response and solving my issue.

framps, 2021-11-15:
> I documented all of the steps I took to initially install raspiBackup

Would be great If you share the steps and I can reproduce the issue. Given the fact all your five Raspis have the same issue I think there is some bug I should fix.

SJMarty, 2021-11-15:
Here is the documentation I put together for myself...

The msmtp and msmtp-mta packages must be installed to allow e-mail notifications:
 
`sudo apt-get install msmtp msmtp-mta`
 
A default configuration file for msmtp must be created:
 
`sudo nano /etc/msmtprc`
 
Add the following entries to the new file:
 
```
defaults
auth            on
tls             on
tls_starttls    on
tls_trust_file  /etc/ssl/certs/ca-certificates.crt
logfile         /etc/msmtp.log

account         default
from            REMOVED@REMOVED
host            REMOVED
port            587
user            REMOVED
password        REMOVED
```
 
To mount an NFS share to use for a backup destination:
 
`sudo mkdir /media/pi/nfs`
 
Set the NFS share to automatically mount:
 
`sudo nano /etc/fstab`
 
Add the following line:
 
`REMOVED.local:/volume1/NFS    /media/pi/nfs    nfs    defaults    0    0`
 
Reboot and ensure that the NFS share auto-mounts.
 
The current installer for raspiBackup can be downloaded, installed, and started as follows:
 
`curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/installation/install.sh | sudo bash`
 
The installation/configuration will launch automatically.  To invoke the install/uninstall/configuration interface manually:
 
`sudo raspiBackupInstallUI`
 
M1 – Language [**English**]
M2 – Install Components | I1 – Install raspiBackup using a default configuration
 
<Back>
 
M3 – Configure Major Options
 
C1 – Language [**English**]
C2 – Backup path for Backups [**/media/pi/nfs**]
C3 – Backup Versions [**Keep a maximum number of backups | 3**]
C4 – Backup Type [**Backup with rsync and use hardlinks if possible**]
C5 – Backup Mode [**Backup the two standard partitions**]
C6 – Services to Stop and Start

	1. cups-browsed (all)
	2. cups (all)
	3. webmin (all)
	4. telegraf (all)
	5. grafana-server (REMOVED)
	6. Influxdb (REMOVED)
	7. channels-dvr (REMOVED)
	8. plexmediaserver (REMOVED)

C7 – Message Verbosity [**Display All Messages**]
C8 – E-Mail Notification [**REMOVED@ REMOVED / msmtp**]
C9 – Regular Backup [**Not Enabled**]
 
<Back>
 
Do you want to save the updated raspiBackup configuration now? [**Yes**]
 
<Finish>
 
<OK>
 
<OK>
 
The configuration file for raspiBackup is located at `/usr/local/etc/raspiBackup.conf`.
 
If using the rsync backup type, the following lines should be added to the end of the file:
  
```
# Modify rsync backups to not save ACLs (when backing up to NFS shares)
DEFAULT_RSYNC_BACKUP_OPTIONS="--delete -aHx"
```
 
The default sender of e-mail notifications will be root@HOSTNAME where HOSTNAME is the device name of the Raspberry Pi.  To change this, the following existing entry in the configuration file should be modified:
 
`sudo nano /usr/local/etc/raspiBackup.conf`
 
```
# Sender emailadress used with ssmtp
DEFAULT_SENDER_EMAIL="desired_from_address@domain"
```
 
To update raspiBackup to the latest version:
 
`sudo raspiBackup.sh -U -S`
 
To run raspiBackup in “fake” mode for testing purposes (doesn’t actually run a backup):
 
`sudo raspiBackup.sh -F`
 
To run raspiBackup and show progress:
 
`sudo raspiBackup.sh -g`

framps, 2021-11-15:
Great :+1: Thank you very much.

framps, 2021-11-15:
I executed most of your detailed documented steps (skipped the non raspiBackup related steps like msmtp) but failed :cry: I'm not able to reproduce your issue :cry: 

I'll mark this issue as unreproducable but keep an eye on it whether other users face the same issue and will reopen the issue. If your're fine with this decision please close the issue.

Thank you very much for your support to locate the root cause for the issue.

SJMarty, 2021-11-15:
Thanks for quickly providing the fix!

closed 2021-11-15 by SJMarty