framps/raspiBackup #399: Temperature extenstion unable to call vcgencmd with incorrect path

FawkesMeteor, 2021-11-23:
Hello,

When I trying to enable temperature extensions in the backup process, the temperature extensions seems cannot find the vcgencmd command with following error:
**/usr/local/bin/raspiBackup_temp_pre.sh: line 37: /opt/vc/bin/vcgencmd: No such file or directory**

The location of vcgencmd called by the shell script is wrong, the correct path should be /usr/bin/vcgencmd. I tried to call the vcgencmd in a test shell script without any problems.

Detailed debug log:

```
20211123-123314 DBG 1765:              --> callExtensions pre
20211123-123314 DBG 1792:                  --- Calling raspiBackup_mem_pre.sh 0
20211123-123314 DBG 2005:                  --> executeShellCommand . raspiBackup_mem_pre.sh 0
20211123-123314 DBG 2012:                  <-- executeShellCommand 0
20211123-123314 DBG 1795:                  --- Extension RC: 0
20211123-123314 DBG 1792:                  --- Calling raspiBackup_temp_pre.sh 0
20211123-123314 DBG 2005:                  --> executeShellCommand . raspiBackup_temp_pre.sh 0
/usr/local/bin/raspiBackup_temp_pre.sh: line 37: /opt/vc/bin/vcgencmd: No such file or directory
20211123-123314 DBG 2012:                  <-- executeShellCommand 0
20211123-123314 DBG 1795:                  --- Extension RC: 0
20211123-123314 DBG 1792:                  --- Calling raspiBackup_disk_pre.sh 0
20211123-123314 DBG 2005:                  --> executeShellCommand . raspiBackup_disk_pre.sh 0
20211123-123314 DBG 2012:                  <-- executeShellCommand 0
20211123-123314 DBG 1795:                  --- Extension RC: 0
20211123-123314 DBG 1805:              <-- callExtensions
```
```
20211123-125702 DBG 1765:          --> callExtensions post
20211123-125702 DBG 1792:              --- Calling raspiBackup_mem_post.sh 0
20211123-125702 DBG 2005:              --> executeShellCommand . raspiBackup_mem_post.sh 0
20211123-125703 DBG 3298:                  --> colorAnnotation 1
20211123-125703 DBG 3322:                  <-- colorAnnotation
--- RBK1001I: Memory usage - Pre backup - Used: 81 MB Free: 7491 MB - Post backup - Used: 90 MB Free: 6313 MB
20211123-125703 DBG 2012:              <-- executeShellCommand 0
20211123-125703 DBG 1795:              --- Extension RC: 0
20211123-125703 DBG 1792:              --- Calling raspiBackup_temp_post.sh 0
20211123-125703 DBG 2005:              --> executeShellCommand . raspiBackup_temp_post.sh 0
/usr/local/bin/raspiBackup_temp_pre.sh: line 37: /opt/vc/bin/vcgencmd: No such file or directory
20211123-125703 DBG 3298:                  --> colorAnnotation 1
20211123-125703 DBG 3322:                  <-- colorAnnotation
--- RBK1000I: CPU temperature pre and post backup:  -
20211123-125703 DBG 2012:              <-- executeShellCommand 0
20211123-125703 DBG 1795:              --- Extension RC: 0
20211123-125703 DBG 1792:              --- Calling raspiBackup_disk_post.sh 0
20211123-125703 DBG 2005:              --> executeShellCommand . raspiBackup_disk_post.sh 0
20211123-125703 DBG 3298:                  --> colorAnnotation 1
20211123-125703 DBG 3322:                  <-- colorAnnotation
--- RBK1001I: Disk usage pre backup: Used: 34.37 GiB Free: 2.51 TiB
20211123-125703 DBG 3298:                  --> colorAnnotation 1
20211123-125703 DBG 3322:                  <-- colorAnnotation
--- RBK1002I: Disk usage post backup: Used: 34.86 GiB Free: 2.51 TiB
20211123-125703 DBG 3298:                  --> colorAnnotation 1
20211123-125703 DBG 3322:                  <-- colorAnnotation
--- RBK1003I: Free change: -508.00 MiB (0.00 %%)
20211123-125703 DBG 2012:              <-- executeShellCommand 0
20211123-125703 DBG 1795:              --- Extension RC: 0
20211123-125703 DBG 1805:          <-- callExtensions

```
other information:

```
FM@Rpi:~ $ which vcgencmd
/usr/bin/vcgencmd
FM@Rpi:~ $ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
FM@Rpi:~ $ apt-cache policy vcgencmd
N: Unable to locate package vcgencmd
FM@Rpi:~ $ apt-cache policy libraspberrypi-bin
libraspberrypi-bin:
  Installed: 2+git20210928~141951+6e8f786-1~buster
  Candidate: 2+git20210928~141951+6e8f786-1~buster
  Version table:
 *** 2+git20210928~141951+6e8f786-1~buster 500
        500 http://archive.raspberrypi.org/debian buster/main arm64 Packages
        100 /var/lib/dpkg/status
FM@Rpi:~ $ vcgencmd measure_temp
temp=39.9'C
```

framps, 2021-11-23:
This is a bug which pops up in Bullseye. In all previous RaspiOS releases the extension worked fine. 

It's already fixed (see https://github.com/framps/raspiBackup/commit/5d123d8470fb6a8fde1cea4eff78d0faa211b1dd#diff-a7624ff686fa29777f0b6b695a70610d4fcd1bdf0de64a143529f16d392ef2b8). Please reinstall the sample extensions to grab the fix, verify and close the issue :wink:

framps, 2021-12-11:
Verified in #410

closed 2021-12-11 by framps