framps/raspiBackup #401: restore sucessfull but Raspi don't start - No working init found

TomW80, 2021-11-29:
Hello,

I made a backup with version 0.6.6 of my openhab system (openhabian 2.5.12) on a Raspi 4. The backup was finished successfully.
Now I wanted to restore the backup. The restore also finished successfully. But when I start the Raspi, I get the following error:

----[ end Kernel panic - not syncing: No working init found. Try passing the init= option to guidance. ]---

What is the problem here?

Attached are the two logs and a screenshot of the error

Tom

[image]
[raspiBackup_restore.log](https://github.com/framps/raspiBackup/files/7621751/raspiBackup_restore.log)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/7621752/raspiBackup.log)

framps, 2021-11-30:
Usually this happens if

1) PARTUUIDs don't match in /boot/cmdline.txt and /etc/fstab with the existing PARTUUID on the restored device. I checked the logs and everything is OK.
2) Filesystem on the restored system is corrupt. Please mount the restored SD card on your Mint system and run fsck on /dev/mmcblk0p2. I'll create an issue and will add the fsck call in next release of raspiBackup when the restore finished.
3) SD card used for restore is corrupt. Restore the backup on another SD card.

I noticed in the debug log you use raspiBackup release 0.6.6 (c8d3058) which is about 7 months old. I suggest to upgrade to latest release. That way you also will run the same release on your Mint system. I don't think that's the root cause for your issue but nevertheless I suggest to use same release for backup and restore.

TomW80, 2021-11-30:
2. see attached
3. I have no other one. I think the SD card ist not corrupt, but I will test with H2testw from ct

I also will make a test with the new Version 0.6.6.1
[image]
[image]
.

framps, 2021-12-01:
Filesystem is OK. :+1: So it looks like it's the SD card. 

> I have no other one. I think the SD card ist not corrupt, but I will test with H2testw from ct

I strongly recommend to have at least 1-2 new spare SD cards in the shelf :wink: I never used h2testw from c't to check a buggy SD card. Would be nice if you report your experiences and results with this tool.

TomW80, 2021-12-08:
I have now tested the SD card with h2test, no errors were found. Furthermore, I tested the restore with the same version as the backup version. Unfortunately, the error remains

framps, 2021-12-08:
:cry: I'm sorry - but I frankly don't know how any more what may the root cause of your issue. I know of various users of OH which successfully backup and restore their systems.

You use 0.6.6 whereas 0.6.6.1 is out already. But I don't think this makes a difference. Please try backupype dd whether it works.

TomW80, 2021-12-14:
Hello framps,
I have now tested the backup and restore with dd. That worked.
I have another raspi, with rsync-backup. I will test if I can restore this backup.

TomW80, 2021-12-23:
Hello framps,

I have now also tested my second rsync backup from another raspi with pihole. But it's the same problem here too.

Now I can only switch to dd backup or do you have another solution?

But it is strange why both rsync backups can't be restored for me and other users don't seem to have any problems. Or does no one test the recovery as recommended by you?

framps, 2021-12-24:
> Or does no one test the recovery as recommended by you?

I ask everybody to test the restore. And I know it works for other OH users. 

You may mount the restored image which fails during boot somewhere on a Linux system and provide /var/log/syslog. Maybe I find some indication what's going wrong. But please be careful. There may be sensual information in the log.

TomW80, 2021-12-29:
[Here ](https://abigailsh.private-ddns.de/index.php/s/AAn7SW6SyxadZXz)you can find the syslog.

framps, 2021-12-29:
There is nothing I can find in the log. Just to double check: That's the log of the restored system which fails during start?

TomW80, 2021-12-30:
Yes, I was surprised myself that the boot process was not logged. I had restored the backup, then started the Raspi. Then I checked the syslog on the SD card, but it hadn't changed.
What can I do to ensure that the boot process is logged?

framps, 2021-12-30:
> What can I do to ensure that the boot process is logged?

I unfortunately  don't know 😞

framps, 2021-12-30:
I just found [this page](https://raspberrypi.stackexchange.com/questions/79973/how-to-see-debug-logs-from-raspberry-pi-3-during-boot-process). May be it helps.

TomW80, 2022-01-01:
Ok, I have now installed bootlogd. I will test it with the next backup.

TomW80, 2022-01-13:
Sorry for my late reply, but my time is short at the moment.
I have now logged the boot process via the serial interface. Once the restore from a rsync and dd backup and once from a restored image with hdClone.

You can find the logs [here](https://abigailsh.private-ddns.de/index.php/s/PRLjH2HDDqS2W8s) 

I hope this helps with troubleshooting.

framps, 2022-01-15:
Thank you very much for the logs. Great you attached also logs with a successful boot. I found
```
[    3.338380] Starting init: /etc/init exists but couldn't execute it (error -13)
```
what causes the boot failure.

I have to check the logs more carefully now ... Please be patient.

framps, 2022-01-16:
According the logs the UUIDs are OK. I detected in the restore log
```
20211126-200935 DBG 1860:          *** cat /etc/os-release
20211126-200935 DBG 1860:              NAME="Linux Mint"
20211126-200935 DBG 1860:              VERSION="20.1 (Ulyssa)"
20211126-200935 DBG 1860:              ID=linuxmint
20211126-200935 DBG 1860:              ID_LIKE=ubuntu
20211126-200935 DBG 1860:              PRETTY_NAME="Linux Mint 20.1"
20211126-200935 DBG 1860:              VERSION_ID="20.1"
20211126-200935 DBG 1860:              HOME_URL="https://www.linuxmint.com/"
20211126-200935 DBG 1860:              SUPPORT_URL="https://forums.linuxmint.com/"
20211126-200935 DBG 1860:              BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
20211126-200935 DBG 1860:              PRIVACY_POLICY_URL="https://www.linuxmint.com/"
20211126-200935 DBG 1860:              VERSION_CODENAME=ulyssa
20211126-200935 DBG 1860:              UBUNTU_CODENAME=focal
```

Usually you should be able to restore a backup on any Linux system but I already had issues where there were version mismatches between the tools used to create the backup and the tools used to restore the backup.

The backup was created with
```20211126-021505 DBG 1597:          *** cat /etc/os-release
20211126-021505 DBG 1597:              PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
20211126-021505 DBG 1597:              NAME="Raspbian GNU/Linux"
20211126-021505 DBG 1597:              VERSION_ID="10"
20211126-021505 DBG 1597:              VERSION="10 (buster)"
20211126-021505 DBG 1597:              VERSION_CODENAME=buster
20211126-021505 DBG 1597:              ID=raspbian
20211126-021505 DBG 1597:              ID_LIKE=debian
20211126-021505 DBG 1597:              HOME_URL="http://www.raspbian.org/"
20211126-021505 DBG 1597:              SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
20211126-021505 DBG 1597:              BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
I frankly didn't find any indication in the logs why the boot of the restored system fails :cry: 

Please restore the backup on a system running the same OS which was used to create the backup. Crossing fingers the boot will succeed :crossed_fingers:

TomW80, 2022-01-19:
I have now restored the backup on a raspi4 with the same OS as it was created, but the problem still exists. :(
Under the link above you can find the boot log of the serial interface and the restore log of raspiBackup

TomW80, 2022-01-19:
Are we perhaps looking on the wrong side the problem? Couldn't it also be that the backup is created incorrectly?

framps, 2022-01-22:
> Couldn't it also be that the backup is created incorrectly?

May be. But I think it's on the restore side. But you can check the backup (see [here](https://www.linux-tips-and-tricks.de/en/backup/#directory) for the files in the rsync backup). And you should check the .img file contains the contents of /boot with losetup (See [here](https://superuser.com/questions/117136/how-can-i-mount-a-partition-from-dd-created-image-of-a-block-device-e-g-hdd-u) how to mount the image).

Now let's check the restored image. Please plug in the device in your Raspberry and execute following commands and show the replies:

`lsusb`

You should get an output like
```
NAME        MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINT
sda           8:48   0 119.8G  0 disk  
└─sda1        8:49   0   256M  0 part  
└─sda2      179:2    0     8G  0 part  
mmcblk0     179:0    0 119.8G  0 disk  
├─mmcblk0p1 179:1    0   256M  0 part  /boot
└─mmcblk0p2 179:2    0     8G  0 part  /
```

If it's not /dev/sda use the correct /dev/sdx in following commands:

Next

```
blkid
<OUTPUT>

sudo mount /dev/sda1 /mnt
sudo mount /dev/sda2 /media
cat /mnt/cmdline.txt
cat /media/etc/fstab
mount
<OUTPUT>
df -h
<OUTPUT>
ls -lRt /mnt
<OUTPUT>
ls -lrt /media
<OUTPUT>
sudo umount /mnt
sudo umount /media
```

TomW80, 2022-01-24:
I've put the log-file the information on the link above.

framps, 2022-01-25:
Sorry. I just find the debug logs in the link in your comment https://github.com/framps/raspiBackup/issues/401#issuecomment-1012616699 but I need the output of all commands I wrote down in previous comment.

TomW80, 2022-01-25:
Its at the end of the log file "putty_Raspi_Serial_20220124_232341.log"

framps, 2022-01-26:
That's the interesting sequence in the log:

```
penhabian@openhabTest:~$ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    1 59,5G  0 disk 
├─sda1        8:1    1  256M  0 part 
└─sda2        8:2    1 59,2G  0 part 
mmcblk0     179:0    0 59,5G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 59,2G  0 part /
openhabian@openhabTest:~$ blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="F661-303B" TYPE="vfat" PARTUUID="5cb814e6-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="8d008fde-f12a-47f7-8519-197ea707d3d4" TYPE="ext4" PARTUUID="5cb814e6-02"
openhabian@openhabTest:~$ sudo mount /dev/sa[Kda1 /mnt
[sudo] Passwort für openhabian: 
openhabian@openhabTest:~$ sudo mount /dev/sda1 /mnt[1P /mnt2 /mnt[C[C[C[C[C[K[Kedia
openhabian@openhabTest:~$ cat /mnt/cmdline.txt
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=5cb814e6-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
openhabian@openhabTest:~$ cat /media/etc/fstab
cat: /media/etc/fstab: Keine Berechtigung
openhabian@openhabTest:~$ cat /media/etc/fstab[1@s[1@u[1@d[1@o[1@ 
proc            /proc           proc    defaults          0       0
PARTUUID=5cb814e6-01  /boot           vfat    defaults          0       2
PARTUUID=5cb814e6-02  /               ext4    defaults,noatime  0       1
```
`5cb814e6` is the partuuid of the SD card. The backup should have the partuuid of /dev/sda in /boot/cmdline.txt and /etc/fstab which has the sd card uuid. I unfortunately cannot derive /dev/sda uuid. I expected /dev/sda1 and /dev/sda2 also to be listed in command `blkid`. Do you have any idea why it's not listed?

I will go back and check the restore log for the uuid changes which should have happen on /dev/sda.

framps, 2022-01-26:
In the restore log I found
```
0220119-211224 DBG 7088:                  /dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="F661-303B" TYPE="vfat" PARTUUID="5cb814e6-01"
20220119-211224 DBG 7088:                  /dev/mmcblk0p2: LABEL="rootfs" UUID="8d008fde-f12a-47f7-8519-197ea707d3d4" TYPE="ext4" PARTUUID="5cb814e6-02"
20220119-211224 DBG 7088:                  /dev/sda1: LABEL_FATBOOT="boot" LABEL="boot" UUID="F661-303B" TYPE="vfat" PARTUUID="5cb814e6-01"
20220119-211224 DBG 7088:                  /dev/sda2: UUID="ffa62655-6409-40b0-bcc4-d83f4f516330" TYPE="ext4" PARTUUID="5cb814e6-02"
```
which points out your sd card and /dev/sda have the same partuuid. In general this is not an issue as long as you don't have both devices plugged inon the system when you boot. But if you have the sd card plugged in when you boot the restored image this may be the root cause of your boot failure.

What you can do:
1) Either remove the sd card when you boot from /dev/sda
2) Use option `--updateUUIDs` when you restore the backup to create a different partuuid on /dev/sda

TomW80, 2022-01-27:
I do not boot from /dev/sda, but from the SD card.
Now I did a new restore with the --updatUUIDs option. Then I ran the commands from above to show the output.
After that I restart the Raspi with the restored sd card. But the same problems :(

You can find on the link above the new three files:
putty_Raspi_Serial_20220128_001817_bootlog_restore_rsync_fehler.log
putty_Raspi_Serial_20220128_000810.log
IMG_20220128_000112_1280x720.jpg

framps, 2022-01-29:
I unfortunately don't find anything :disappointed: I created [this script](https://github.com/framps/raspiBackup/blob/master/scripts/checkUUIDsOnSDCard.sh) which checks the UUIDs on your restored SD card image. You may test whether it reports an error. But I frankly don't think so.

I'm frankly disappointed to be unable to find the root cause. But I don't see any chance to debug the issue without access to the restored image even there are a lot of debug messages available. Boot issues are very difficult to debug for me.

Said this - I'm sorry - looks like you're not able to use raspiBackup - forever unkown reason :cry:

TomW80, 2022-02-07:
I tested again today and wanted to look at the partitions with my Linux Mint after restoring. I noticed that after restoring the rsync backup, I cannot view the second partition. I get the message "You do not have the necessary access rights to view the contents of xyz".
The problem did not occur when restoring the dd backup, where I was able to view the second partition.

Can this be the problem?

I will also see if I can make an image available. However, it may take a while.

Is it possible to switch off the automatic closing of the post?

framps, 2022-02-07:
> I will also see if I can make an image available. However, it may take a while.

I didn't ask for an image :wink: Please note my support is for free and I already spent a lot of time in this issue.  I'm not willing to spend any time on this issue any more. 

Said this - I'm sorry - you're not able to use raspiBackup for your environment and have to look for another backup tool.

closed 2022-02-07 by framps

TomW80, 2022-02-10:
Hello framps

I can understand that you don't want to search forever for a bug that only a user of raspibackup has. However, I was disappointed by the answer at the first moment.

I have not given up and found the error now.
I make the backup to a NFS share on my Synology. When restoring, I had not restored directly from the NAS but downloaded the entire directory of the rsync backup via the web interface and then restored. I suspect that the file attributes were lost. Because after the restore I could not access the second partition with my Linux Mint.
Yesterday I did the restore directly from the NAS and see that it now boots normally. :)

I do not know if you can check the data attributes to exclude this problem.

Anyway, it works now.

Thanks for your efforts and help.
Thanks also for this great tool.

TomW80

framps, 2022-02-10:
> However, I was disappointed by the answer at the first moment.

I see your point - but since a couple of months I get much more issues than before. I try to work on all of them but I cannot spend unlimited time on every issue and have to stop my activity at some point in time. Boot issues are very difficult to debug - unfortunately. 

> I have not given up and found the error now.

:+1: Thank you very much you share what you found out. Next time somebody has boot issues I will ask him whether he restores from the backup location or copied the backup somewhere elase first :wink: 

> Yesterday I did the restore directly from the NAS and see that it now boots normally. :)

That's what happens for all other users of raspiBackup :smiley: 

> I do not know if you can check the data attributes to exclude this problem.

I check whether the backup storage supports Linux file attributes when a backup is created. I had a lot of issues because the restore failed and the raspiBackup users used NTFS on the backup location :rage: rsync doesn't detect NTFS doesn't support all Linux fileattributes. Now every NTFS backupspace will be rejected with an error message.

I frankly never expected somebody to restore from a copy of the backup data. There may be ways to guard against this faux pas and I usually add checking code whenever I detect this user error may happen often. But I think it's very unusual to use a copy of the backupdata for restore.  You used the wrong tool/options to copy the backupdata. Take this as a Linux gotcha and I'm sure next time you copy bulk data you will make sure the attributes are copied :wink: 

> Anyway, it works now.

Glad you finally found the root cause and now can use raspiBackup as your backup tool. Enjoy :smiley: